rauc/rauc #582: Mounting fails

ArashJavan, 2020-04-27:
Hi, i just follow the example in doc. I build a my linux suing buildroot fpr stm32-dk2 eval board.
Following instructions in this [article](https://bootlin.com/blog/building-a-linux-system-for-the-stm32mp1-remote-firmware-updates/comment-page-1/#comment-2210763). But instead of SWupdate i want to use RAUC :). So after buildroot is finished is just copy the "rootfs.ex4"-File and follow the example article in  RAUC and create an rauc-bundle. On the device i get following error:

```
# rauc install /root/update-2020.04-01.raucb
installing
  0% Installing
  0% Determining slot states
 20% Determining slot states done.
 20% Checking bundle
 20% Verifying signature
 40% Verifying signature done.
 40% Checking bundle done.
100% Installing failed.
LastError: Failed mounting bundle: failed to run mount: Child process exited with code 32
Installing `/root/update-2020.04-01.raucb` failed
```

What am i doing wrong?

Thanks

jluebbe, 2020-04-27:
Did you enable squashfs and loopback block devices in your kernel? Please check the Integration chapter in the docs, especially https://rauc.readthedocs.io/en/latest/integration.html#kernel-configuration.

obbardc, 2020-05-15:
pretty sure this is a duplicate of #562

according to the hassos guys, this patch should fix it: https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/drivers/block/loop.c?id=7e81f99afd91c937f0e66dc135e26c1c4f78b003

can you check this patch is in your kernel or better yet run linux-next ?

ejoerns, 2020-10-24:
@ArashJavan did the patch mentioned resolve your issue, too? Did you have SquashFS support enabled as a module?

PR #625 (2020-10-27): docs: include Kernel loop device patch
There is a bug in Kernel version v5.0 until v5.7 when loop device
support is loaded as a module, in a small percentage of cases
mounting the bundle fails. Include information about the patch.

Resolves: #562
Resolves: #582
commit 666fafd0 (2020-10-27): docs: include Kernel loop device patch

There is a bug in Kernel version v5.0 until v5.7 when loop device
support is loaded as a module, in a small percentage of cases
mounting the bundle fails. Include information about the patch.

Resolves: #562
Resolves: #582
closed 2020-10-27 by jluebbe