rauc/rauc #586: Target side hostpage for update

lp35, 2020-05-11:
Hi,

We are currently using SwUpdate for updating our board. After reading the whole documentation, I think the design of Rauc better fits to our needs, and more flexible in some ways.

One feature is missing however, the ability to have a webpage served by the target in order to remote update. I have understood that RAUC is providing a DBus interface and nothing else (#266), but I think it would be very convenient for people to have this as an optionnal "plugin" for RAUC, even a simple one (see https://sbabic.github.io/swupdate/swupdate.html).

Can this be considered as a new feature?

Thanks

ejoerns, 2020-05-11:
@lp35 Thanks for evaluation RAUC for your project :)

About the simple web interface: Yes, I do see this is a piece of convenience that can be useful for people that do not need any specific upload setup but want a simple and quick way to provide a standard web interface.
If we had this, this would not be a 'plugin' of RAUC as in swupdate, because we are a little less monolithic, but it could be a project in the contrib/ folder or a separate git project. We have similar mechanism and interfaces already, like the [CGI](https://github.com/rauc/rauc/tree/master/contrib/cgi) as part of rauc repo or [rauc-hawkbit](https://github.com/rauc/rauc-hawkbit)/[rauc-hawkbit-updater](https://github.com/rauc/rauc-hawkbit-updater) as projects under the RAUC organization. We won't integrate a full web server I think, as there are good and lightweight ones outside (like http) but the web page together with some convenience to set it up could be provided.

> Can this be considered as a new feature?

I would say it can. RAUC is an open source project, thus is you plan to develop such a feature, you are free to contribute or discuss your development here and we could help / guide you.
If your plan is to let someone develop this for you, you can ask consulting companies (like the one I work for) to prepare this feature for you. A third way is to wait until either a community member of any of our customers intended to implement this, but as far as I know there currently is nothing planned in this direction.

Best regards, Enrico

ejoerns, 2020-06-16:
@lp35 Does this answer your question? Did this give you some impression on how we could proceed here?

Cicicok, 2020-09-28:
Can you please elaborate on this or provide link to the lightweight web server?

> We won't integrate a full web server I think, as there are good and lightweight ones outside (like http) but the web page together with some convenience to set it up could be provided.

I'm used to deploy webpages using apache or nginx and I don't think this dependency should be used.

ejoerns, 2020-09-28:
@Cicicok I actually meant "like lighttpd" ;) see https://www.lighttpd.net/

Cicicok, 2020-09-28:
@ejoerns thank you for the link. When I'm understanding this correctly the backend of the `Target side webpage for update` will be like the [CGI](https://github.com/rauc/rauc/tree/master/contrib/cgi) and frontend regular html webpage?