rauc/rauc #588: Setting bootname on both parent and child slots can cause confusing non-deterministic errors: invalid slot status 7

danc86, 2020-05-22:
After upgrading glib and some other packages (not rauc) we started seeing this confusing error message from rauc when running `rauc status` or any installation commands:

```
(rauc:103246): rauc-ERROR **: 06:45:31.099: invalid slot status 7
```

But the error was not occurring all the time. Eventually I narrowed down the root cause to a misconfiguration in our product's /etc/rauc/system.conf. Here is the relevant parts:

```
[slot.rootfs.0]
device=/dev/rootfs-slot-0
type=raw
bootname=slot_0

[slot.kernel.0]
device=/boot/slot_0.bin
type=raw
parent=rootfs.0

[slot.kernelsig.0]
device=/boot/slot_0.bin.sig
type=raw
parent=rootfs.0

[slot.blsconf.0]
device=/boot/loader/entries/slot_0.conf
type=raw
parent=rootfs.0
bootname=slot_0

[slot.cmdlinesig.0]
device=/boot/slot_0.bin.cmdline.sig
type=raw
parent=rootfs.0

[slot.rootfs.1]
device=/dev/rootfs-slot-1
type=raw
bootname=slot_1

[slot.kernel.1]
device=/boot/slot_1.bin
type=raw
parent=rootfs.1

[slot.kernelsig.1]
device=/boot/slot_1.bin.sig
type=raw
parent=rootfs.1

[slot.blsconf.1]
device=/boot/loader/entries/slot_1.conf
type=raw
parent=rootfs.1
bootname=slot_1

[slot.cmdlinesig.1]
device=/boot/slot_1.bin.cmdline.sig
type=raw
parent=rootfs.1
```

Notice that the rootfs.0 slot has `bootname=slot_0`, but its child slot blsconf.0 also has `bootname=slot_0`. According to the rauc docs about [grouping slots](https://rauc.readthedocs.io/en/latest/integration.html#grouping-slots) this configuration is clearly wrong. But actually rauc doesn't notice the misconfiguration, it happily accepts this configuration file.

And it was actually working fine in our product for several years. The reason is the `determine_slot_states()` function iterates the slots in non-deterministic hash table order. If the hash table happens to put the parent slot first, `determine_slot_states()` correctly picks the parent slot as the booted slot and everything works.

But some change in glib, or just some bad luck, means that now the hash table iteration order is different and the child slot was being picked as the booted slot incorrectly. Then this block in `determine_slot_states()` was constructing an invalid enum value for ->state:

```
        /* Determine active group members */
        for (GList *l = slotlist; l != NULL; l = l->next) {
                RaucSlot *s = (RaucSlot*) g_hash_table_lookup(r_context()->config->slots, l->data);

                if (s->parent) {
                        if (s->parent->state & ST_ACTIVE) {
                                s->state |= ST_ACTIVE;
                        } else {
                                s->state |= ST_INACTIVE;
                        }
                } else {
                        if (s->state == ST_UNKNOWN)
                                s->state |= ST_INACTIVE;
                }
        }
```

`s->state` is `ST_BOOTED` but `s->parent->state` is `ST_INACTIVE`. 5 | 2 = 7.

I would like to contribute a fix for handling this misconfiguration, but I'm not sure what would be the best solution. Here are some possibilities I thought of:

* if a slot has both `parent=` and `bootname=` make the config parser reject it with an error
* if a slot has both `parent=` and `bootname=` make the config parser warn and ignore the bootname (I am going to use this as a temporary solution in our product but it does not seem ideal)
* if two slots have the same `bootname=` make the config parser reject it with an error
* avoid using |= when constructing s->state so that an invalid enum value cannot be produced
* make the logic for s->state more robust, for example if a child is booted mark its parent as active, or just add some g_warn_if_fail() to catch nonsensical states

jluebbe, 2020-05-25:
Thanks for the in-depth analysis! As the system.conf is shipped together with the rauc binary and are tested together after any rauc update, it seems unlikely that any new error message during config parsing would only appear in the field.

So I think we should:
- [ ] reject slots with both `parent=` and `bootname=`
- [ ] reject duplicate `bootname`s
- [ ] add more sanity checks to the slot state logic

What do you think, @ejoerns?

danc86, 2020-05-27:
I figured out why our product had this invalid configuration in the first place.

We have a slot-post-install hook for the child slot which wants to refer to `$RAUC_SLOT_BOOTNAME`. If we fix our configuration to remove the bootname from the child slot that env var will no longer be set, and there is no nice way to figure out the bootname in a child slot hook in that case.

I think it makes sense to just populate that env var for all slots in a group (parent and children) based on the bootname set on the parent hook. So I will send a separate PR for that.

ejoerns, 2020-05-28:
> So I think we should:
> 
> * [ ]  reject slots with both `parent=` and `bootname=`
> * [ ]  reject duplicate `bootname`s
> * [ ]  add more sanity checks to the slot state logic
> 
> What do you think, @ejoerns?

I'd agree with this, yes. I had some of this points on my mental list already, but never enough time or focus to actually fix this. Thus thanks to @danc86 for working on it!

PR #591 (2020-07-14): prevent issues around duplicate bootnames
The first two commits ensure the config file is rejected if it does not meet the existing assumptions about how slot groups are constructed (bootname must be set only on the parent slot in a group).

The third commit just tidies up the confusing use of |= which led to the invalid enum values.

Fixes #588.
commit 0cf720b2 (2020-07-14): src/install: avoid constructing enum values with |=

This refactors determine_slot_states() so that the ->state field for
each slot is assigned exactly once, and it is always assigned a valid
enum value.

Fixes #588.
closed 2020-07-14 by jluebbe