rauc/rauc #594: Installation Error: Error writing to file descriptor: No space left on device

amod-amatya, 2020-05-29:
I'm integrating the rauc recipes for a raspberrypi using meta-rauc, but the upgrade process fails.
I am trying to manually upgrade the image with 
> rauc install update-bundle.raucb

```
  0% Installing
  0% Determining slot states
 20% Determining slot states done.
 20% Checking bundle
 20% Verifying signature
 40% Verifying signature done.
 40% Checking bundle done.
 40% Loading manifest file
 60% Loading manifest file done.
 60% Determining target install group
 80% Determining target install group done.
 80% Updating slots
 80% Checking slot rootfs.1
 90% Checking slot rootfs.1 done.
 90% Copying image to rootfs.1
100% Copying image to rootfs.1 failed.
100% Updating slots failed.
100% Installing failed.
LastError: Installation error: Failed updating slot rootfs.1: Failed splicing data: Error writing to file descriptor: No space left on device
```
Please let me know what further info I need to provide.

ejoerns, 2020-05-29:
@amod-amatya Maybe the created partition image is larger than 1GB? This could be the case even if the actual content is much less.

Also the output of the rauc service would help. If you use systemd you should be able to get it with

    journalctl -u rauc

amod-amatya, 2020-05-29:
@ejoerns what I did next was, for testing purpose just increased the rootfs partition size to around 4gb, but then a new error popped up.

My logs from journalctl

```
-- Logs begin at Fri 2020-05-22 09:43:23 UTC, end at Fri 2020-05-29 15:04:55 UTC. --
May 22 09:43:27 raspberrypi-cm3 systemd[1]: Starting Rauc Update Service...
May 22 09:43:27 raspberrypi-cm3 systemd[1]: Started Rauc Update Service.
May 22 09:43:27 raspberrypi-cm3 rauc[179]: rauc-Message: 09:43:27.535: rauc mark: marked slot rootfs.0 as
 good
May 29 14:57:53 raspberrypi-cm3 rauc[179]: input bundle: /home/root/update-bundle-rpi-test-raspberrypi-cm
3-20200529130809.raucb
May 29 14:57:53 raspberrypi-cm3 rauc[179]: rauc-Message: 14:57:53.728: Active slot bootname: A
May 29 14:57:53 raspberrypi-cm3 rauc[179]: rauc-Message: 14:57:53.729: installing /home/root/update-bundl
e-rpi-test-raspberrypi-cm3-20200529130809.raucb: started
May 29 14:57:53 raspberrypi-cm3 rauc[179]: rauc-Message: 14:57:53.733: installing /home/root/update-bundl
e-rpi-test-raspberrypi-cm3-20200529130809.raucb: Checking and mounting bundle...
May 29 14:57:53 raspberrypi-cm3 rauc[179]: rauc-Message: 14:57:53.734: Reading bundle: /home/root/update-
bundle-rpi-test-raspberrypi-cm3-20200529130809.raucb
May 29 14:57:53 raspberrypi-cm3 rauc[179]: rauc-Message: 14:57:53.739: Verifying bundle...
May 29 14:57:54 raspberrypi-cm3 rauc[179]: rauc-Message: 14:57:54.072: Mounting bundle '/home/root/update
-bundle-rpi-test-raspberrypi-cm3-20200529130809.raucb' to '/run/rauc/bundle'
May 29 14:57:54 raspberrypi-cm3 rauc[179]: rauc-Message: 14:57:54.146: Marking target slot rootfs.1 as no
n-bootable...
May 29 14:57:54 raspberrypi-cm3 rauc[179]: rauc-Message: 14:57:54.231: installing /home/root/update-bundl
e-rpi-test-raspberrypi-cm3-20200529130809.raucb: Updating slots...
May 29 15:00:51 raspberrypi-cm3 rauc[179]: rauc-Message: 15:00:51.291: Checking image type for slot type:
 ext4
May 29 15:00:51 raspberrypi-cm3 rauc[179]: rauc-Message: 15:00:51.295: Image detected as type: *.ext4
May 29 15:00:51 raspberrypi-cm3 rauc[179]: rauc-Message: 15:00:51.296: installing /home/root/update-bundl
e-rpi-test-raspberrypi-cm3-20200529130809.raucb: Checking slot rootfs.1
May 29 15:00:51 raspberrypi-cm3 rauc[179]: rauc-Message: 15:00:51.345: Slot needs to be updated with /run
/rauc/bundle/update-image-raspberrypi-cm3.ext4
May 29 15:00:51 raspberrypi-cm3 rauc[179]: rauc-Message: 15:00:51.346: installing /home/root/update-bundl
e-rpi-test-raspberrypi-cm3-20200529130809.raucb: Updating slot rootfs.1
May 29 15:00:51 raspberrypi-cm3 rauc[179]: rauc-Message: 15:00:51.347: Updating /dev/mmcblk0p3 with /run/
rauc/bundle/update-image-raspberrypi-cm3.ext4
May 29 15:00:51 raspberrypi-cm3 rauc[179]: rauc-Message: 15:00:51.348: opening slot device /dev/mmcblk0p3

May 29 15:00:51 raspberrypi-cm3 rauc[179]: rauc-Message: 15:00:51.361: writing data to device /dev/mmcblk
0p3
May 29 15:04:48 raspberrypi-cm3 rauc[179]: (rauc:179): rauc-WARNING **: 15:04:48.840: Installation error:
 Failed updating slot rootfs.1: Written size (2147483647) != image size (4152360960)
May 29 15:04:48 raspberrypi-cm3 rauc[179]: rauc-Message: 15:04:48.843: installing /home/root/update-bundl
e-rpi-test-raspberrypi-cm3-20200529130809.raucb: Installation error: Failed updating slot rootfs.1: Writt
en size (2147483647) != image size (4152360960)
May 29 15:04:48 raspberrypi-cm3 rauc[179]: rauc-Message: 15:04:48.844: installing /home/root/update-bundl
e-rpi-test-raspberrypi-cm3-20200529130809.raucb: finished
May 29 15:04:48 raspberrypi-cm3 rauc[179]: rauc-Message: 15:04:48.845: installing `/home/root/update-bund
le-rpi-test-raspberrypi-cm3-20200529130809.raucb` failed: 1
```

ejoerns, 2020-05-29:
@amod-amatya RAUC still has some issues with 32 bit limitations. Maybe you run into that.
But most of them were fixed with RAUC v1.3. Thus consider switching to this.

However, it appears you also increased the image size, not only the partition size?

Maybe an output of `rauc info` on the bundle would also reveal some details

amod-amatya, 2020-05-29:
@ejoerns I got your point, when I updated the system with core-image-minimal, the update succeeds as the image size was approx 1gb where as my custom image was around 2gb. Also, I happened to increase the image size as well.
Isn't this supposed to increase the free space,
 In my local.conf, I have  
```WKS_FILE = "pi-custom.wks"
IMAGE_ROOTFS_SIZE = "1048576"
IMAGE_ROOTFS_EXTRA_SPACE = "2100000"
IMAGE_OVERHEAD_FACTOR = "1"
IMAGE_FSTYPES = "ext4 wic"
```
Isnt this suppose to add only extra space to the rootfs partition but my whole wic image seems to increase.

amod-amatya, 2020-06-04:
Ok, this issue can be resolved, Thanks for the support.
PS. I solved the issue by adding --size 1024 in wks file. This only increases the partition size keeping the image size same.

closed 2020-06-04 by amod-amatya