rauc/rauc #596: Set offset for bootloader image in the eMMC bootpartitions

ctenruh-phytec, 2020-06-03:
Hello,

i was reading the mailing list about [U-Boot Offset for RAW Image on Slot Update](https://www.mail-archive.com/search?l=rauc@pengutronix.de&q=subject:%22%5C%5BRAUC%5C%5D+U%5C-Boot+Offset+for+RAW+Image+on+Slot+Update%22&o=newest&f=1). I want to write the bootloader with an offset in the boot partitions with RAUC. Is there any progress in that area or do you have an idea how we could implement it without using hooks?

jluebbe, 2020-06-03:
With hooks, you can't modify *how* the image itself is written. So the straightforward solution is to add the padding before creating the bundle. I currently don't see any benefit in having the additional complexity, when it can be done at build time. Perhaps some convenience variable for an offset in the rauc.bbclass would be the better solution.

ctenruh-phytec, 2020-06-18:
I added the offset in the bundle recipe. Thanks.

ejoerns, 2020-06-18:
@ctenruh-phytec Ok. If that's anything we could have as a generic mechanism in the bundle class, feel free to open a PR.

closed 2020-11-04 by ejoerns