rauc/rauc #598: Failed to run casync extract

mschwan-phytec, 2020-06-16:
Hi,

I am currently experimenting with casync support. The bundle can be converted to a casync one with `rauc convert` without any issues. But the installation of the bundle fails:
```
root@host:~# rauc install --debug casync-bundle.raucb
rauc-Message: 15:03:31.334: Debug log domains: 'all'

(rauc:4087): rauc-DEBUG: 15:03:31.337: install started
(rauc:4087): rauc-DEBUG: 15:03:31.337: input bundle: /root/casync-bundle.raucb
(rauc:4087): rauc-DEBUG: 15:03:31.349: Trying to contact rauc service
installing
  0% Installing
  0% Determining slot states
 20% Determining slot states done.
 20% Checking bundle
 20% Verifying signature
 40% Verifying signature done.
 40% Checking bundle done.
 40% Loading manifest file
 60% Loading manifest file done.
 60% Determining target install group
 80% Determining target install group done.
 80% Updating slots
 80% Checking slot rootfs.0
 85% Checking slot rootfs.0 done.
 85% Copying image to rootfs.0
[ 2314.897908] EXT4-fs (mmcblk2p2): mounted filesystem with ordered data mode. Opts: (null)
 90% Copying image to rootfs.0 done.
[ 2353.312953] EXT4-fs (mmcblk2p2): mounted filesystem with ordered data mode. Opts: (null)
 90% Checking slot boot.0
 95% Checking slot boot.0 done.
 95% Copying image to boot.0
100% Copying image to boot.0 failed.
100% Updating slots failed.
100% Installing failed.
LastError: Installation error: Failed updating slot boot.0: failed to run casync extract: Child process exited with code 1
Installing `/root/casync-bundle.raucb` failed
```
Now, I would like to find out why casync failed exactly. Exit code 1 is not very descriptive. Unfortunately, setting `G_MESSAGES_DEBUG=all` (nor `--debug`) will print debug messages inside the corresponding functions, like `casync_extract()`, in `src/update_handler.c`. I also did not find any handler that would redirect the Glib output to a different stream. Any idea how debug messages can be enabled and other ways to find out more of the actual problem?

OnkelUlla, 2020-06-16:
Did you take a look into the system's journal as all messages from RAUC's service unit will end up there?

ejoerns, 2020-06-16:
@mschwan-phytec changing the debug level as well as inspecting log (as @OnkelUlla already mentioned) mainly affects the rauc service. The CLI is just a small D-Bus 'frontend' for RAUC.

Thus

    journalctl -u rauc

is what you probably want to inspect.

Also refer https://rauc.readthedocs.io/en/latest/using.html?highlight=debugging#debugging-rauc

mschwan-phytec, 2020-06-16:
Doh! Yes, of course I should check these. :see_no_evil: 

```
Mar 12 18:59:00 machine systemd[1]: Starting Rauc Update Service...
Mar 12 18:59:00 machine systemd[1]: Started Rauc Update Service.
Mar 12 18:59:37 machine rauc[4313]: input bundle: /root/casync-bundle.raucb
Mar 12 18:59:37 machine rauc[4313]: Active slot bootname: system1
Mar 12 18:59:37 machine rauc[4313]: installing /root/casync-bundle.raucb: started
Mar 12 18:59:37 machine rauc[4313]: installing /root/casync-bundle.raucb: Checking and mounting bundle...
Mar 12 18:59:37 machine rauc[4313]: Reading bundle: /root/casync-bundle.raucb
Mar 12 18:59:37 machine rauc[4313]: Verifying bundle... 
Mar 12 18:59:37 machine rauc[4313]: Mounting bundle '/root/casync-bundle.raucb' to '/run/rauc/bundle'
Mar 12 18:59:37 machine rauc[4313]: Starting pre install handler: /usr/bin/rauc_downgrade_barrier.sh
Mar 12 18:59:37 machine rauc[4313]: Marking target slot rootfs.0 as non-bootable...
Mar 12 18:59:37 machine rauc[4313]: Checking image type for slot type: ext4
Mar 12 18:59:37 machine rauc[4313]: Image detected as type: *.caidx
Mar 12 18:59:37 machine rauc[4313]: installing /root/casync-bundle.raucb: Updating slots...
Mar 12 18:59:37 machine rauc[4313]: installing /root/casync-bundle.raucb: Checking slot rootfs.0
Mar 12 18:59:37 machine rauc[4313]: mounting slot /dev/mmcblk2p2
Mar 12 18:59:37 machine rauc[4313]: Slot needs to be updated with /run/rauc/bundle/image-machine.tar.gz.caidx
Mar 12 18:59:37 machine rauc[4313]: Updating /dev/mmcblk2p2 with /run/rauc/bundle/image-machine.tar.gz.caidx
Mar 12 18:59:37 machine rauc[4313]: Formatting ext4 slot /dev/mmcblk2p2
Mar 12 18:59:37 machine rauc[4313]: installing /root/casync-bundle.raucb: Updating slot rootfs.0
Mar 12 18:59:37 machine rauc[4313]: mke2fs 1.44.5 (15-Dec-2018)
Mar 12 18:59:37 machine rauc[4313]: 64-bit filesystem support is not enabled.  The larger fields afforded by this feature enable full-strength checksumming.  Pass -O 64bit to rectify.
Mar 12 18:59:45 machine rauc[4313]: Discarding device blocks:    4096/1916416                528384/1916416               1052672/1916416               1576960/1916416                                             done
Mar 12 18:59:45 machine rauc[4313]: Creating filesystem with 1916416 4k blocks and 479552 inodes
Mar 12 18:59:45 machine rauc[4313]: Filesystem UUID: 72cd7a5f-14e0-426d-af62-6ef32bc20e05
Mar 12 18:59:45 machine rauc[4313]: Superblock backups stored on blocks:
Mar 12 18:59:45 machine rauc[4313]: 	32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632
Mar 12 18:59:45 machine rauc[4313]: Allocating group tables:  0/59               done
Mar 12 18:59:45 machine rauc[4313]: Writing inode tables:  0/59               done
Mar 12 18:59:46 machine rauc[4313]: Creating journal (16384 blocks): done
Mar 12 18:59:47 machine rauc[4313]: Writing superblocks and filesystem accounting information:  0/59               done
Mar 12 18:59:47 machine rauc[4313]: Mounting ext4 slot /dev/mmcblk2p2
Mar 12 18:59:47 machine rauc[4313]: Extracting /run/rauc/bundle/image-machine.tar.gz.caidx to /run/rauc/rootfs.0
Mar 12 19:00:23 machine rauc[4313]: Unmounting ext4 slot /dev/mmcblk2p2
Mar 12 19:00:24 machine rauc[4313]: installing /root/casync-bundle.raucb: Updating slot rootfs.0 status
Mar 12 19:00:24 machine rauc[4313]: Updating slot file /run/rauc/rootfs.0/slot.raucs
Mar 12 19:00:24 machine rauc[4313]: Checking image type for slot type: vfat
Mar 12 19:00:24 machine rauc[4313]: installing /root/casync-bundle.raucb: Updating slot rootfs.0 done
Mar 12 19:00:24 machine rauc[4313]: Image detected as type: *.caidx
Mar 12 19:00:24 machine rauc[4313]: installing /root/casync-bundle.raucb: Checking slot boot.0
Mar 12 19:00:24 machine rauc[4313]: mounting slot /dev/mmcblk2p1
Mar 12 19:00:24 machine rauc[4313]: Failed to load status file /run/rauc/boot.0/slot.raucs: No such file or directory
Mar 12 19:00:24 machine rauc[4313]: Slot needs to be updated with /run/rauc/bundle/boot.tar.gz.img.caidx
Mar 12 19:00:24 machine rauc[4313]: Updating /dev/mmcblk2p1 with /run/rauc/bundle/boot.tar.gz.img.caidx
Mar 12 19:00:24 machine rauc[4313]: installing /root/casync-bundle.raucb: Updating slot boot.0
Mar 12 19:00:24 machine rauc[4313]: Formatting vfat slot /dev/mmcblk2p1
Mar 12 19:00:24 machine rauc[4313]: mkfs.fat: warning - lowercase labels might not work properly with DOS or Windows
Mar 12 19:00:24 machine rauc[4313]: mkfs.fat 4.1 (2017-01-24)
Mar 12 19:00:24 machine rauc[4313]: Mounting vfat slot /dev/mmcblk2p1
Mar 12 19:00:25 machine rauc[4313]: Extracting /run/rauc/bundle/boot.tar.gz.img.caidx to /run/rauc/boot.0
Mar 12 19:00:26 machine rauc[4313]: Failed to run synchronizer: Operation not permitted
Mar 12 19:00:26 machine rauc[4313]: Unmounting vfat slot /dev/mmcblk2p1
Mar 12 19:00:27 machine rauc[4313]: Installation error: Failed updating slot boot.0: failed to run casync extract: Child process exited with code 1
Mar 12 19:00:27 machine rauc[4313]: installing /root/casync-bundle.raucb: Installation error: Failed updating slot boot.0: failed to run casync extract: Child process exited with code 1
Mar 12 19:00:27 machine rauc[4313]: installing /root/casync-bundle.raucb: finished
Mar 12 19:00:27 machine rauc[4313]: installing `/root/casync-bundle.raucb` failed: 1
```
Seems like casync fails in `ca_sync_step()` with a permission error... will investigate this further.

ejoerns, 2020-06-18:
@mschwan-phytec Did you test against the latest casync version from master?

But non-descriptive error messages like the one you see still seem to be an open issue: https://github.com/systemd/casync/issues/44

mschwan-phytec, 2020-06-23:
I tracked this down to the following lines in casync:

https://github.com/systemd/casync/blob/4ad9bcb94bc83ff36cfc65515107ea06a88c2dfc/src/cadecoder.c#L3646-L3656
```c
                if (st.st_uid != uid || st.st_gid != gid) {

                        if (child->fd >= 0)
                                r = fchown(child->fd, uid, gid);
                        else {
                                assert(dir_fd >= 0);

                                r = fchownat(dir_fd, name, uid, gid, AT_SYMLINK_NOFOLLOW);
                        }
                        if (r < 0)
                                return -errno;
```

The corresponding file is the kernel image `Image` being written to the "boot" slot. Here casync uses `uid=19695` and `gid=10513`. But since RAUC is run as root (`uid=0` and `gid=0`), casync cannot write the image and exits with `-EPERM`. Writing the root filesystem with the same permissions works just fine. This is the situation of the slots after trying to install a casync bundle:
```
root@machine:~# ls -la boot
drwxr-xr-x    2 root     root         16384 Jan  1  1970 ./
drwx------    5 root     root          4096 Mar 12 14:41 ../
-rwxr-xr-x    1 root     root      30960128 Mar 12 14:29 .#Image6023ccc7f7472450*
root@machine:~# ls -la root
drwxr-xr-x   18 19695    10513         4096 Mar 12 14:29 ./
drwx------    5 root     root          4096 Mar 12 14:41 ../
drwxr-xr-x    2 19695    10513         4096 Jun 23  2020 bin/
drwxr-xr-x    2 19695    10513         4096 Jun 23  2020 boot/
drwxr-xr-x    2 19695    10513         4096 May  8  2020 dev/
drwxr-xr-x   30 19695    10513         4096 Jun 23  2020 etc/
drwxr-xr-x    2 19695    10513         4096 May  8  2020 home/
drwxr-xr-x    9 19695    10513         4096 Jun 23  2020 lib/
drwxr-xr-x    2 19695    10513         4096 May  8  2020 media/
drwxr-xr-x    2 19695    10513         4096 May  8  2020 mnt/
dr-xr-xr-x    2 19695    10513         4096 May  8  2020 proc/
drwx------    2 19695    10513         4096 May  8  2020 root/
drwxr-xr-x    2 19695    10513         4096 Jun 23  2020 run/
drwxr-xr-x    2 19695    10513         4096 Jun 23  2020 sbin/
dr-xr-xr-x    2 19695    10513         4096 May  8  2020 sys/
drwxr-xr-x    2 19695    10513         4096 May  8  2020 tmp/
drwxr-xr-x   10 19695    10513         4096 Jun 23  2020 usr/
drwxr-xr-x    8 19695    10513         4096 Jun 23  2020 var/
```
The boot slot with kernel image and device tree is formatted with vfat and the rootfs slot with ext4. I suspect that since vfat does not support permissions on a per-file basis and permission are set during mounting, that this the reason casync cannot write to this partition.

I am wondering though, why casync writes with different uid and gid? Would it not make sense for it to write with the same permissions as RAUC does (RAUC calls casync as a subprocess)?

mschwan-phytec, 2020-06-23:
Not sure if it is the correct way of solving this, but adding the `permissions` archive feature when extracting does the trick. Additionally, the `symlinks` feature must be activated for symlinks to be extracted.
```diff
diff --git a/src/update_handler.c b/src/update_handler.c
index 249e895..a8b7839 100644
--- a/src/update_handler.c
+++ b/src/update_handler.c
@@ -179,6 +179,8 @@ static gboolean casync_extract(RaucImage *image, gchar *dest, const gchar *seed,
                g_ptr_array_add(args, g_strdup(store));
        }
        g_ptr_array_add(args, g_strdup("--seed-output=no"));
+       g_ptr_array_add(args, g_strdup("--with=permissions"));
+       g_ptr_array_add(args, g_strdup("--with=symlinks"));
        g_ptr_array_add(args, g_strdup(image->filename));
        g_ptr_array_add(args, g_strdup(dest));
        g_ptr_array_add(args, NULL);
```

jluebbe, 2020-07-13:
> The corresponding file is the kernel image `Image` being written to the "boot" slot. Here casync uses `uid=19695` and `gid=10513`.

The your boot slot uses vfat, so only uid 0 / gid 0 is possible:
```
$ dd if=/dev/zero of=vfat bs=1M count=128
128+0 records in
128+0 records out
134217728 bytes (134 MB, 128 MiB) copied, 0,0683507 s, 2,0 GB/s
$ mkfs.vfat vfat 
mkfs.fat 4.1 (2017-01-24)
$ mkdir mp
$ mount vfat mp
$ cd mp/
$ touch dummy
$ ll
total 0
-rwxr-xr-x 1 root root 0 Jul 13 15:21 dummy
$ chown 0:0 dummy 
$ chown 1:0 dummy 
chown: changing ownership of 'dummy': Operation not permitted
```

This is expected, as vfat has no concept of user IDs.

> But since RAUC is run as root (`uid=0` and `gid=0`), casync cannot write the image and exits with `-EPERM`. Writing the root filesystem with the same permissions works just fine. 

No. The issue is that RAUC (casync) is trying to extract a file with non-zero uid/gid to the boot vfat filesystem.

> This is the situation of the slots after trying to install a casync bundle:
>
> ```
> root@machine:~# ls -la boot
> drwxr-xr-x    2 root     root         16384 Jan  1  1970 ./
> drwx------    5 root [...]

Your rootfs doesn't belog to root either, but as ext4 supports IDs, it can be extracted.

> The boot slot with kernel image and device tree is formatted with vfat and the rootfs slot with ext4. I suspect that since vfat does not support permissions on a per-file basis and permission are [...]

`rauc convert` preserves the permissions in the tar archive from the original bundle by calling `tar` and `casync make` under `fakeroot`. My guess is that your tar archives already contain the wrong uid/gid (maybe from your build user?).

mschwan-phytec, 2020-07-16:
Thanks @jluebbe for the explanation! You are correct that the files have the user and group ID from my build user.

If I understand you correctly, the regular bundle must have the correct file permissions (root) already in its content. However, I am not sure how this can be achieved. The simplest way I could think of, is to change the user and group ID of the files to be bundled before executing `rauc bundle` in `do_bundle()`. I tried different things like `fakeroot do_bundle()` or even plain `chown`. But when converting the bundle, casync exits with code 15 (ENOTBLK? does not seem right).

jluebbe, 2020-07-16:
Hmm. The correct permissions should already be produced by OE/Yocto and this works file with the normal bundle code, and also when running `rauc convert` separately from the image build. Could you check if your normal bundles are correct and can be converted correctly outside of bitbake?

mschwan-phytec, 2020-07-16:
Creating a regular bundle with fakeroot works. Converting the bundle to a casync bundle outside of bitbake works. But the casync bundle does not install on the target. It fails at the boot slot with a permission error the same way it did before. Strangely enough, the root filesystem does have the correct ownership now:
```
root@machine:~# ls -l root
drwxr-xr-x    2 root     root          4096 Jul 16 07:40 bin/
drwxr-xr-x    2 root     root          4096 Jul 16 07:39 boot/
drwxr-xr-x    2 root     root          4096 Jun 30 01:42 dev/
drwxr-xr-x   31 root     root          4096 Jul 16 07:40 etc/
...
```

mschwan-phytec, 2020-09-25:
I think I nailed down the root-cause of this issue. There are two things that need fixing:

1. In our own layer (not meta-rauc), we have an extra task that packs the kernel image and device tree in a tar.gz archive. This archive is then extracted to a separate boot partition. The problem here is that the archive is created by the build user and thus the kernel image and device tree have according ownership (build user instead of root). Of course, this then generates the issue described above where files cannot be extracted to the vfat boot partition because of incorrect UID and GID.
2. Something is still not right about the `rauc convert` command in the `bundle.bbclass`. Normally, casync requires `fakeroot` to be installed. As there is no (native) fakeroot package in any Yocto layer, I initially worked around this by using `pseudo` (created a link named `fakeroot` pointing to `pseudo`). This builds without error messages, but it seems ownership is not preserved correctly. Building outside of bitbake with a real `fakeroot` (but same rauc and casync binary) preserves the correct ownership of the root filesystem (that is mostly owned by root).

Fixing 1. should be no problem. But 2. is a little tricky. Should we add a recipe for building a real `fakeroot` binary? Or did I miss something in `bundle.bbclass` that would allow us using `pseudo`, as I first intended to?

mschwan-phytec, 2020-09-28:
> 2. Something is still not right about the `rauc convert` command in the `bundle.bbclass`. Normally, casync requires `fakeroot` to be installed. As there is no (native) fakeroot package in any Yocto [...]

Seems like one has to enable `pseudo` with `PSEUDO_DISABLED=0` when using `rauc convert` in the `bundle.bbclass`. Looking at the corresponding task `run.do_bundle` in the working directory of the built bundle, `PSEUDO_DISABLED` is exported with `1` by default.

```diff
diff --git a/classes/bundle.bbclass b/classes/bundle.bbclass
index 43ba6c5..357c260 100644
--- a/classes/bundle.bbclass
+++ b/classes/bundle.bbclass
@@ -262,7 +262,7 @@ do_bundle() {
                if ! [ -x "$(command -v fakeroot)" ]; then
                        ln -sf ${STAGING_DIR_NATIVE}${bindir}/pseudo ${STAGING_DIR_NATIVE}${bindir}/fakeroot
                fi
-               PSEUDO_PREFIX=${STAGING_DIR_NATIVE}/usr ${STAGING_DIR_NATIVE}${bindir}/rauc convert \
+               PSEUDO_PREFIX=${STAGING_DIR_NATIVE}/usr PSEUDO_DISABLED=0 ${STAGING_DIR_NATIVE}${bindir}/rauc convert \
                        --debug \
                        --cert=${RAUC_CERT_FILE} \
                        --key=${RAUC_KEY_FILE} \
```
@jluebbe @ejoerns Is this an appropriate fix for you?

closed 2021-04-15 by ejoerns