rauc/rauc #606: casync chunk store URL handling improvements

kevinior, 2020-06-30:
For our current project we're looking at using RAUC with casync. Since we might have to use bundle and chunk store URLs with a query I had a look at the RAUC code to see how that's handled.

It seems like the [code that works out the chunk store path](https://github.com/rauc/rauc/blob/5dba83af16e4a0567a3f9bfce5e12fc90c7e8b46/src/bundle.c#L746) always treats the paths as if they're local files, even if they're URLs. So if you have a bundle URL like `https://something.com/update.raucb?key=12345` then the chunk store will be `https://something.com/update.raucb?key=12345.castr`, which is unlikely to work.

Unfortunately we might well need to use URLs with a query part because of stupid backend systems, so we'll need to find a solution to the problem.

It looks like a workaround might be to handle the URL parsing in our own code and modify the `[casync]storepath` setting in the system configuration file. However, proper URL parsing in RAUC would be a much cleaner solution.

ejoerns, 2020-07-01:
@kevinior Setting `storepath=https://something.com/store/key=0815` will not be sufficient for your use case?

If `storepath` is set explicitly, there is no concatenation happening as you can see in the code snippet you referenced

kevinior, 2020-07-01:
Thanks for the quick response.

The problem with setting `storepath` is that we'd have to rewrite the RAUC system configuration file every time we did an update. A better option would be if we could override `storepath` from the command line: `rauc install --storepath=https://something.com/store?key=0815`. And that I should be able to patch myself.

Although casync seems to discard queries from URLs, so we'd have to fix the backend anyway to work with URLs like your example (key in path).

In general I'm a bit allergic to messing around with URLs without using a proper parser, I've been burned too many times.

ejoerns, 2020-07-08:
@kevinior since we have the new D-Bus installation API allowing to pass new arguments on installation to the service, having an argument to pass it the storepath per-call either form any D-Bus client or from the rauc client itself (via `--storepath`) should be an option, yes.

RAUC itself tries to interpret URLs as little as possible (so we just use the scheme parser from glib) and forward it to casync or libcurl. The attempt to append `.castr` is just a convenience oversimplification for specific use cases. I'd tend to say that most will require to configure this, anyway.

For the casync way of handling URLs that's probably something one needs to fix there, yes.

kevinior, 2020-07-08:
OK, thanks for the information. I'll try to submit a pull request for overriding `storepath` from the command line.

ejoerns, 2020-07-08:
If you don't mind I'd like to keep this open for reference and would ask you to put a "Fixes #606" in your commit that will both allow linking to this Issue as well as automatically closing it on merge.

kevinior, 2020-09-11:
@ejoerns I finally got round to creating a pull request for this. codecov seems unhappy with it but I don't know what the project's policy about test coverage is. Should all newly-added code have 100% coverage?

Developing tests that need to run in QEMU is a bit tricky since I'm forced to develop in Windows Subsystem for Linux (v1, not even the decent version) and running the tests in QEMU on top of WSL takes approximately forever.

ejoerns, 2020-09-11:
I'd like to have a view on this from a security perspective before, as changing the store location could be a potential attack option if casync does not verify the data downloaded.
There is no very strict policy on coverage, but changes should be covered by tests if possible with manageable effort.