rauc/rauc #607: Is it possible to disable CURLOPT_SSL_VERIFYHOST in rauc?

getsandyrw, 2020-06-30:
Hello All,

We have requirement where our device will update its software using "rauc install" command by connecting to server over HTTPS.

Since this is done in local network (PC directly connected to device over Ethernet), there is no domain name for server. The download is done using the IP Address of the server directly. The Server SSL certificate contains subject name.

In this case, "rauc install" is failing with error Subject name mismatch. (Transfer failed: SSL: certificate subject name (x.y.z) does not match target host name '169.254.X.X')

Is it possible to disable the subject name verification in rauc through configuration?  (E.g. disabling curl CURLOPT_SSL_VERIFYHOST option). or is there any alternate option available to resolve above problem at rauc level?  

Please share your valuable inputs / feedback.

Thank you very much in advance.

Sandy

ejoerns, 2020-07-01:
Hi Sandy, if this a local network under your control I assume this to be a trusted connection anyway. Wouldn't it then be sufficient to use http for this?
Another option for a simple setup could be to set a hostname in /etc/hosts if the remote IP is well-known.

DIsabling the subject name verification is currently not possible without changing the code and normally a reasonable setting.

Also note that the download capability of RAUC is primarily meant to be used for downloading small casync-enabled bundles and thus has a very limited default download size that needs to be adapted (`max-bundle-download-size=...`) for your use case.

Regards, Enrico

getsandyrw, 2020-07-01:
Hello Enrico,

Thank you very much for your reply.

Although this is local network, due to regulatory requirement,  we wanted to have the update done over secure connection.

We have already tried /etc/hosts solution, but every-time server's IP address will be different. So manual update is required in /etc/hosts, which is not feasible.

We also tried using curl setting file (.curlrc) in home directory with "insecure" option. But rauc is not using this option/settings while downloading the file.

We have tried with rauc code changes (CURLOPT_SSL_VERIFYHOST) and it is working fine.  I think for time being, we will go ahead with code changes in rauc.

We have updated the default download size as per our requirement.

Regards
Sandy

jluebbe, 2023-10-24:
Setting [CURLOPT_SSL_VERIFYHOST](https://curl.se/libcurl/c/CURLOPT_SSL_VERIFYHOST.html) to 0 would accept any valid certificate, making MitM attacks very easy. Accordingly we don't intend to support this, as you could also just use plain HTTP.

Instead, if you know which host name the certificate contains, the proper approach would be to use [CURLOPT_RESOLVE](https://curl.se/libcurl/c/CURLOPT_RESOLVE.html) to tell curl about the host to IP mapping. In this case, the certificate is checked against the host name in the URL. This option could be passed on `rauc install` similar to the `--tls-cert=`/`--http-header=` options.

I'm going to close this issue, so please open a new one or a PR for the `CURLOPT_RESOLVE` approach if it's still relevant to you.