rauc/rauc #620: Install rootfs update when booted from external drive - how to?

hdmi87, 2020-07-21:
I use Phytec phyCORE i.MX6 SOM with my custom Yocto build. I use barebox bootloader with two rootfs slots (`rootfs.0` and `rootfs.1`) on SOM's eMMC. Here is a part of `/etc/rauc/system.conf` file:

    [slot.rootfs.0]
    device=/dev/mmcblk3p1
    type=ext4
    bootname=system0

    [slot.rootfs.1]
    device=/dev/mmcblk3p2
    type=ext4
    bootname=system1

On a new SOM I boot from external microSD and run a bash script that creates partitions on eMMC, installs bootloader and unpack `.tar.gz` image archive into both root partitions. After this primary installation, the system is booted from eMMC and can be updated using RAUC (this forks fine).

I want to use RAUC also for primary system installation when SOM is booted from eMMC. The reason is because when RAUC installs the rootfs image, it generates `slot.raucs` file that contains useful information about installed bundle version, date of installation etc. 

The problem is that RAUC cannot determine booted slot (because system is not booted neither from `rootfs.0` nor from `rootfs.1`):

    $ rauc install /update-bundle.raucb
    rauc-Message: 10:38:15.239: installing /update-bundle.raucb: installing
    rauc-Message: 10:38:15.240: installing /update-bundle.raucb:   0% Installing
    rauc-Message: 10:38:15.241: installing /update-bundle.raucb:   0% Determining slot states
    rauc-Message: 10:38:15.241: installing /update-bundle.raucb:  20% Determining slot states failed.
    rauc-Message: 10:38:15.247: installing /update-bundle.raucb: 100% Installing failed.
    rauc-Message: 10:38:15.249: installing /update-bundle.raucb: LastError: Did not find booted slot

    $ rauc status
    Compatible:  phycore-imx6dl-emmc
    Variant:
    Booted from: (null) (/dev/mmcblk0p2)
    Activated:   rootfs.1 (system1)
    slot states:
      rootfs.0: class=rootfs, device=/dev/mmcblk3p1, type=ext4, bootname=system0
          state=inactive, description=, parent=(none), mountpoint=(none)
          boot status=good
      rootfs.1: class=rootfs, device=/dev/mmcblk3p2, type=ext4, bootname=system1
          state=inactive, description=, parent=(none), mountpoint=(none)
          boot status=good

I tried to call `rauc install` with `--override-boot-slot` parameter , but this changes nothing.

Is it possible to use RAUC to update/install the system that is not currently booted?

I tried to use `rauc write-slot`, it works, but does not write `slot.raucs` file into target slot.

obbardc, 2020-07-21:
Can you try `--override-boot-slot` with a global rauc status file
which contains dummy information?

hdmi87, 2020-07-21:
> Can you try `--override-boot-slot` with a global rauc status file which contains dummy information?
Can you explain your idea more particularly? Currently I do not use global RAUC status files, only per slot status files.

Now I tried to add microSD root partition into `/etc/rauc/system.conf`:

    #System MicroSD
    [slot.microsd.0]
    device=/dev/mmcblk0p2
    type=ext4

Now RAUC can determine which slot is booted:

    Booted from: microsd.0 (/dev/mmcblk0p2)
    Activated:   rootfs.0 (system0)

and it installs update bundle successfully, but only into one slot (`rootfs.0`). If I run it again, it installs the update bundle into the same slot again. Is it possible to force RAUC to install bundle into a specific slot (namely `rootfs.1`)?

jluebbe, 2020-07-21:
> On a new SOM I boot from external microSD and run a bash script that creates partitions on eMMC, installs bootloader and unpack `.tar.gz` image archive into both root partitions. After this primary [...]

This is explicitly out of scope for RAUC: https://rauc.readthedocs.io/en/latest/index.html#and-what-not

Nevertheless, I can see that having the stot status information from the beginning would be useful, but we've not considered that detail before. @ejoerns has a WIP branch (https://github.com/rauc/rauc/compare/master...ejoerns:topic/slot-status-concept) that should allow you to update both slots by running `rauc install` two times. Maybe you could take a look and we can discuss further?

> The problem is that RAUC cannot determine booted slot (because system is not booted neither from `rootfs.0` nor from `rootfs.1`):

I see that you fixed that by adding a slot for your MicroSD. You can pass rauc.external on the kernel command line, to explicitly tell it that all slots are inactive:
https://rauc.readthedocs.io/en/latest/integration.html#identification-via-kernel-commandline

ejoerns, 2020-10-24:
@hdmi87 Did the `rauc.external` hint help you to simplify the setup? Did you have a look at my patch series for a possible part of the solution for writing both rootfs.0 **and** rootfs.1?

diorcety, 2021-04-20:
@hdmi87 
> I tried to call rauc install with --override-boot-slot parameter , but this changes nothing.

I fall on the same issue. If you are using service the --override-boot-slot is useless cf:
https://github.com/rauc/rauc/blob/master/src/main.c#L1569

PR #730 (2021-05-21): src/main: limit --override-boot-slot argument to specific subcommands
For most commands, `--override-boot-slot` will not have any impact as this is only evaluated for the service (or 'install' command when RAUC is being compiled without 'service' support).

Fix this by moving the parameter from the global option list to the options list of 'service' (and 'install' / 'status').

Also fix the sharness tests that had this argument set without any effect.

Resolves #620
closed 2021-05-21 by jluebbe