rauc/rauc #621: Rezise option doesn't work, written size != image size

pridybailo-n, 2020-07-23:
Hi,

Trying to integrate rauc v1.2 in my custom Yocto build (PHYTEC phyCORE-AM57x SOM based board)
Here is my system.conf file, Im using uboot bootloader and A-B update scheme:

> [system]
compatible=tseb-am5728
bootloader=uboot

>[keyring]
path=/etc/rauc/ca.cert.pem

>[slot.rootfs.0]
device=/dev/mmcblk0p2
type=ext4
bootname=A
resize=true

>[slot.rootfs.1]
device=/dev/mmcblk0p3
type=ext4
bootname=B
resize=true

.wks file:
>part /boot --source bootimg-partition --ondisk mmcblk0 --fstype=vfat --label boot --active --align 4 --size 16
part / --source rootfs  --ondisk mmcblk0 --fstype=ext4 --label A --align 4
part --source rootfs --ondisk mmcblk0 --fstype=ext4 --label B --align 4

fdisk output:
>Disk /dev/mmcblk0: 29.8 GiB, 32010928128 bytes, 62521344 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xb31e1be3

>Device         Boot    Start      End  Sectors  Size Id Type
/dev/mmcblk0p1 *           8    32775    32768   16M  c W95 FAT32 (LBA)
/dev/mmcblk0p2         32776 11210009 11177234  5.3G 83 Linux
/dev/mmcblk0p3      11210016 22387249 11177234  5.3G 83 Linux

rauc info output:
>rauc-Message: Reading bundle: /home/root/tseb-system-update-tseb-am5728.raucb
rauc-Message: Verifying bundle... 
Compatible: 	'tseb-am5728'
Version:    	'1.0'
Description:	'tseb-system-update version 1.0-r0'
Build:      	'20200723132049'
Hooks:      	''
1 Image:
(1)	tseb-image-dev-tseb-am5728.ext4
	Slotclass: rootfs
	Checksum:  6fde1ebfc7c85b131a578c3d2cada077179553f9a03cfa56652163341416a1f5
	Size:      107122688
	Hooks:     
0 Files

rauc install execution log:
>installing
  0% Installing
  0% Determining slot states
 20% Determining slot states done.
 20% Checking bundle
 20% Verifying signature
 40% Verifying signature done.
 40% Checking bundle done.
 40% Loading manifest file
 60% Loading manifest file done.
 60% Determining target install group
 80% Determining target install group done.
 80% Updating slots
 80% Checking slot rootfs.1
 90% Checking slot rootfs.1 done.
 90% Copying image to rootfs.1
100% Copying image to rootfs.1 failed.
100% Updating slots failed.
100% Installing failed.
LastError: Installation error: Failed updating slot rootfs.1: Written size (2147483647) != image size (107122688)
Installing `/home/root/tseb-system-update-tseb-am5728.raucb` failed

I supposed that resize option in system.conf should solve this issue, but its not. Ext4 filesystem used in accordance with documentation, resize2fs util is actually installed:
>resize2fs 1.43 (17-May-2016)
Usage: resize2fs [-d debug_flags] [-f] [-F] [-M] [-P] [-p] device [-b|-s|new_size] [-z undo_file]

I tried to use different values in IMAGE_ROOTFS_SIZE,  IMAGE_ROOTFS_EXTRA_SPACE, IMAGE_OVERHEAD_FACTOR, wks --size, or not use them at all.
Spent few days on that and now decided to write this ticket.
Maybe you can help me with that? The main goal is to solve this error in any way, via resize option or just get equal image size.

Thanks!

pridybailo-n, 2020-07-24:
Solved.
The issue was caused by 32 bit type limitation in rauc v1.2. It means that we could not use size larger than 2Gb.

For me It works with the following configuration:
>IMAGE_ROOTFS_EXTRA_SPACE = "1048576"      # < 2Gb limit
IMAGE_OVERHEAD_FACTOR = "1"                          # Free to change if we want to increase partition size

.wks file:
>part /boot --source bootimg-partition --ondisk mmcblk0 --fstype=vfat --label boot --active --align 4 --size 16
part / --source rootfs --ondisk mmcblk0 --fstype=ext4 --label A --align 4
part --source rootfs --ondisk mmcblk0 --fstype=ext4 --label B --align 4

Also it seems that this problem was fixed in current release (now they are using 64bit data types instead of 32bit).

closed 2020-07-24 by pridybailo-n

jluebbe, 2020-07-24:
Thanks for the update, @pridybailo-n!