rauc/rauc #623: Option to mount dm-integrity protected partitions

ufechner7, 2020-08-05:
We tried to define a dm-integrity protected partition in system.conf , but this is resulting in an error when executing rauc status. 
Questions:
a. should the partition type dm-integrity be added?
b. is there a work around?

RAUC version 1.4

ejoerns, 2020-08-12:
@ufechner7 sounds to me that this is another incarnation of the behavior described in #624.

ufechner7, 2020-08-12:
Well, this is not a question, but a feature request. Using type raw for integrity protected partitions limits the features, that are available a lot. The feature request is to allow to mount integrity protected partitions correctly and to allow all use cases that are available for ext4 partitions. This could be implemented for example be adding a line 
pre_mount_commands=integritysetup open ...; second command
and 
post_unmount_commands=integritysetup close ..