rauc/rauc #624: rauc corrupts dm-verity protected partitions

ufechner7, 2020-08-12:
Sometimes rauc corrupts a dm-verity protected root file system. One way to see this error is to force a switch of the active slot with the command "rauc status mark-active other".

From my understanding dm-verity protected partitions must be written with dd only and never be mounted rw afterwards.

How can this be achieved?

jluebbe, 2020-08-12:
This is likely caused by rauc writing the slot.raucs status file to the slot. Do you have set the slot type to 'ext4'? For dm-verity you need to use type 'raw'.

ejoerns, 2020-08-12:
@ufechner7 To still gain the benefit of RAUC's slot/update status information, you could configure a global slot status file, for example in a common data partition:

``` ini
statusfile = /path/to/global/statu.file
```

Using `raw` over `ext4` in dm-verity use-cases might also be something to note in the docs (FAQ section?) as this will probably not be the last time someone runs into this.

What happens is, that for the not-mounted partition RAUC attempts to mount it on its own and thus also does not recognize any potential (readonly) mount option given for example in the inactive system's fstab. And it does so as the`ext4` slot type implies having a writable file system. The `raw` type tells rauc to not make such an assumption but to only dd images to it.

ufechner7, 2020-08-12:
Thanks for the quick replies! I will test your suggestions and update you if the issue is solved. :)

ufechner7, 2020-08-12:
Ok, tested it, and the solution works fine:
- use type raw for dm-verity protected partitions
- use a global status file on a non-protected ext4 partition

Remaining questions:
Is the status file required for RAUC to work correctly, or is it more a kind of log file that is nice to have, but not strictly required?

ejoerns, 2020-08-12:
@ufechner7 RAUC basically works without the status file but it is also more than a log file because there is some logic that can use information stored in it. And it will become of increasing importance in the future, despite the design goal is still to have RAUC working correctly in case of an unreadable status file.

ufechner7, 2020-08-15:
I think this can be closed as soon as the FAQ is updated.

PR #647 (2020-10-26): docs/faq: document hints for using RAUC with dm-verity-protected slots
Fixes #624
commit 284643b7 (2020-10-26): docs/faq: document hints for using RAUC with dm-verity-protected slots

Fixes #624
closed 2020-10-26 by jluebbe