rauc/rauc #633: bundle encryption

Cicicok, 2020-09-15:
I'm really surprised that nobody asked about encrypting the bundles. I found only [this mail conversation](https://www.mail-archive.com/rauc@pengutronix.de/msg00086.html), which is today more than 2 years old.

Is there some progress or new information in this field?

jluebbe, 2020-09-15:
As far as I know, the design which I suggested in that thread hasn't been implemented.
Do you have a specific use case in mind? How would you protect the decryption key on the device?

Cicicok, 2020-09-15:
I'm working on a master thesis, where a I should build a complex update infrastructure (cloud, recovery system, output on display, etc.) based on RAUC. One of the ideas or extensions was to encrypt the bundles. At first, I was thinking about encrypting the built bundle after its creation using openssl, but extending rauc is more preferable solution.

I like your suggested solution:
`[ squashfs hashed with dm-verity and optionally encrypted in dm-crypt compatible format ][ CMS encrypted and signed container for (algorithm settings, dm-verity root hash, payload key) ]`
Because I'm pretty sure that, you know this stuff better than me.

The ideal place to store and protect the key is to use TPM or some crypto element. That's the ultimate goal, but I would choose the same "approach" as with the signature. (Either store it in rootfs or use PKCS#11.) The list of the key storage options can be:

- in rootfs as plaintext (with plaintext It's not the same as with public key, but it's something)
- store and read it from crypto element
- integrate crypto element into decryption process (don't readout the key)

I know that all of this is easier to say then make it done, but sometimes the start (and overthinking) is the most difficult part of the development process.

jluebbe, 2020-09-15:
> I'm working on a master thesis, where a I should build a complex update infrastructure (cloud, recovery system, output on display, etc.) based on RAUC. One of the ideas or extensions was to encrypt [...]

Interesting! I'd be happy to discuss design trade-offs here (or in the #rauc IRC/Matrix channel). Ideally, I'd expect to see a PR in the end (or at least some prototype ;).

> I like your suggested solution:
> `[ squashfs hashed with dm-verity and optionally encrypted in dm-crypt compatible format ][ CMS encrypted and signed container for (algorithm settings, dm-verity root hash, payload key) ]`
> Because I'm pretty sure that, you know this stuff better than me.

One interesting option might be to encrypt for multiple recipient keys in the same CMS structure. If i'm not mistaken, that should be supported by OpenSSL. It would then depend on the use-case if one would have one decryption key per device or maybe per production batch. Of course that could also be added later.

When using many keys, it would probably be better to switch to ECC keys to save some space and increase performance. That shouldn't™ need any changes in RAUC, though.

> The ideal place to store and protect the key is to use TPM or some crypto element. That's the ultimate goal, but I would choose the same "approach" as with the signature. (Either store it in rootfs [...]

Aggred. I'd skip the second option though, as any crypto element also supports the third option. And then it would also be the same as the signing key handing.

> I know that all of this is easier to say then make it done, but sometimes the start (and overthinking) is the most difficult part of the development process.

:)

Cicicok, 2020-09-21:
dm-verity and dm-crypt are new to me, so at first I need to look into that. I think the way of implementation should be like this:

1. Add dm-verity into bundle creation and bundle mounting process.
2. Add dm-crypt into bundle creation and bundle mounting process.
3. Add new data into CMS container (dm-verity root hash, encryption key for dm-crypt, algorithm settings for dm-crypt?).
4. Encrypt CMS container using Openssl.
5. Sign CMS container using Openssl.

What do you think?

Cicicok, 2020-09-22:
After some googling, try-and-errors I think, that I understand the concept more than yesterday. The post before can be ignored. 

1. dm-verity
I've made a squashfs image, appended hash_device image on it and run veritysetup format with offset. Which gave me root hash. Now I have single file (update bundle) which can be opened/mounted and root hash + offset which need to be stored in the CMS container. The question is how to calculate maximum size of hash device, so we're not wasting space? Should rauc be calling `veritysetup` in order to open/close/format the image or is there some other method?

2. dm-crypt
I've created empty image and formatted it as LUKS container because it was more comfortable for me (plain dm-crypt w/ openssl will be used). After successfull opening of the container I've `dd`-ed the squashfs+hash_device image which I created previously for dm-verity. At this point the container can be closed and as single image file shipped to target device, where it will be decrypted and newly created device `/dev/mapper/decrypted_bundle`, opened using `veritysetup`. From this step goes into CMS container passphrase, cypher type, hash type, key size.

3. CMS container
The data collected from dm-verity and dm-crypt needs to be stored in some sort of structure. What it should be? (I'm coming from web world, mine the first idea is JSON, but it can't be the correct answer). This need to be singed and encrypted. Actually in `man cms` there is an example for this:
```
openssl cms -sign -in ml.txt -signer my.pem -text \
               | openssl cms -encrypt -out mail.msg \
               -from steve@openssl.org -to someone@somewhere \
               -subject "Signed and Encrypted message" -des3 user.pem
```

I saw in your email communication, which I referenced in the first post, talks about limitation of root permission. Can you please elaborate on that?

Now I can began with integration of this into rauc client ("easy-pease").

ejoerns, 2020-09-22:
@Cicicok @jluebbe is on vacation this week. He has some concrete ideas on how this could/should be implemented (at least the verity part) as we planned to do this for another project some time ago but did not realize then in this context.

Would it be ok for you to wait for him until next week? Or are you in a hurry with your project plan? ;)

It should be possible to do what the userspace tools do in RAUC directly, this would have the benefit of not having too much new dependencies apart from kernel support for these features.

Cicicok, 2020-09-23:
@ejoerns Thank you for the information. I'm currently on the beginning of my project, so I have enough time.

I'm looking forward to hear your ideas :)

Cicicok, 2020-11-06:
after few weeks I'm ready to resume the implementing. Are there some news about encryption?

jluebbe, 2020-11-09:
> after few weeks I'm ready to resume the implementing. Are there some news about encryption?

We've done some experiments in this area, but there is noting ready yet. I'd suggest to discuss possible appraches via IRC (#rauc on FreeNode) or [Matrix](https://app.element.io/#/room/#rauc:matrix.org) (my username there is shoragan).

cmorty, 2021-02-15:
@Cicicok; Any news?

hongkongkiwi, 2021-02-26:
Also keen to know updates on this, I know dm-verity support for signing has been implemented but it seems that doesn't yet support encryption. It would be nice if dm-crypto could be supported?

jluebbe, 2021-02-26:
@hongkongkiwi Yes, support for dm-crypt could be integrated along the same lines as the dm-verity metadata (key in Manifest and then encrypt the CMS structure).

rforro, 2021-08-20:
I'm constantly thinking about the encryption and I'm slowly leaning to a conclusion that in most cases it will bring nothing. Here are my reasons why it don't make sense:

- Linux is licensed under GPL, why encrypt if you have to provide source code.
- Unless the storage is encrypted as well (and probably Secure Boot enabled) attacker can simply read the memory after updating.
- Using multiple recipient keys when more than few devices are produced can became unmanageable
- While using OTA delivery the HTTPS connection is on its own encrypted. (people are lazy, nobody will update manually using usb, just force them to ota)

ejoerns, 2021-08-20:
> I'm constantly thinking about the encryption and I'm slowly leaning to a conclusion that in most cases it will bring nothing. Here are my reasons why it don't make sense:
> 
> * Linux is licensed under GPL, why encrypt if you have to provide source code.

The kernel is. And components. But this is of course not what you want to protect. You want to protect the IP of your application (logic).

> * Unless the storage is encrypted as well (and probably Secure Boot enabled) attacker can simply read the memory after updating.

Note that It is a different attack scenario if you can read the update from on-chip flash or from the transport medium.

> * Using multiple recipient keys when more than few devices are produced can became unmanageable

Proper key handling is always a challenge probably, yes ;)

> * While using OTA delivery the HTTPS connection is on its own encrypted. (people are lazy, nobody will update manually using usb, just force them to ota)

We see many scenarios where OTA is not (yet) wanted or actually possible. Especially in machines, industry etc. I agree that use cases become fewer, but still, there are some and some will remain probably. Also note that despite having a secure transport, you might want to upload artifacts to potential unsafe cloud servers, etc.

BrunoVernay, 2021-09-17:
Encryption is definitively useful. In industrial context, OTA is rarely an option. 
News chips like the Cortex M33 are able to store keys with protection for hardware attacks. (Maybe not as strong as CC certified Secure Elements, but more easily avalable.)

Jarsop, 2021-11-22:
I have another use case. If we have a hooks in the bundle which contain sensible information (about update procedure etc), maybe we want to encrypt the bundle. Another option is to encrypt the artifacts itself (image, hook etc).

What do you mean about that ?

ejoerns, 2021-12-16:
I just wanted to drop a note that I have started implementing bundle encryption support for RAUC in the context of a project.

It basically follows the above-mentioned general concept of using dm-crypt (on top of dm-verity) for symmetric encryption of the payload, but without the LUKS header which adds unneeded complexity for our use case where we have a single key and a very specific idea of how we encrypt.

The encryption of the CMS is planned to happen in a separate step, like `rauc encrypt`, and uses standard CMS encryption capabilities to encapsulate and encrypt the signature CMS for a number of recipients.

We've made pre-evaluations that showed that even a high number of individual recipients (1 to 10.000) should be possible to handle without too much time/size overhead.

ejoerns, 2022-03-10:
Note that this feature is being implemented in #863.

PR #863 (2022-03-16): Bundle Encryption Support
This adds support for encrypted bundles, by introducing a new bundle type `crypt`:

``` cfg
[bundle]
format=crypt
```

The implementation is based on the `verity` bundle format (using the  dm-verity device mapper).
It works by symmetrically encrypting the bundle payload (i.e. the SquashFS) and using the kernel device mapper *dm-crypt* to symmetrically decrypt this on the fly.
The symmetric decryption key is part of the manifest in the CMS struct, which itself is (asymmetrically) encrypted.

Fixes #633.
closed 2022-03-16 by jluebbe