rauc/rauc #635: "rauc status mark-active other" does not affect eMMC boot partitions

mschwan-phytec, 2020-09-30:
Marking the other slots as active with `rauc status mark-active other` does not switch the eMMC boot partitions accordingly in conjunction with a "boot-emmc" type slot. The eMMC boot partitions are only switched with a `rauc install`. This can be checked by reading the extcsd register values:
```
$ mmc extcsd read /dev/mmcblk2 | grep PARTITION_CONFIG
```
Is this intended behavior or a bug?

ejoerns, 2020-09-30:
As eMMC updates are not part of the global A+B scheme but only a method of atomically updating a non-redundant 'slot', this is not affected by 'status mark-active'.

There is no sense in coupling an eMMC bootpart with a distinct rootfs as we have the decision point later (in the bootloader) and eMMC bootpart switching does not allow to fall back. 

In a redundancy system, you will always have a component that performs the switching between the redundant entities. This component itself cannot be redundant and is thus decoupled from the A+B switching.
If you move the decision point to the extCSD register, then, despite this could be done in theory, you would be very very limited in what your redundancy system is capable of.

mschwan-phytec, 2020-09-30:
Then what difference does it make, that `rauc install` switches the boot partition but `rauc status mark-active` does not? If I understand you correctly, neither of them should change the extCSD register then.

There is the case where a newer kernel version may be not compatible with the currently used bootloader. In such a scenario, you want to make sure to use the correct combination of bootloader and kernel. Here, it would absolutely make sense to couple eMMC bootpart with rootfs to be able to fall back.

ejoerns, 2020-09-30:
~@mschwan-phytec rauc install only switches the extCSD if the bundle contains an update for the bootloader. It writes the image to the inactive bootpartition and switches the extCSD to this after writing succeeded. This way you can power-off your board while writing the bootloader without bricking your board. But that has nothing to do with redundancy boot, only with updating.~

If you couple eMMC with extCSD, how would you implement the fallback then?
Your updated well-fitting bootloader + kernel combination boots. Then there is an issue with rootfs and your systems stalls. The watchdog resets you and reboots. In a simplified version, you will end up in the fallback then meaning your new bootloader runs and detects that you have to fall back. How would you proceed here to ensure that you do not boot the old kernel with your new bootloader? If I'm not fully mistaken you would have to switch eMMC bootpart and reboot again. Is that what you were thinking about?

mschwan-phytec, 2020-09-30:
> [...] Then there is an issue with rootfs and your systems stalls. The watchdog resets you and reboots. In a simplified version, you will end up in the fallback then meaning your new bootloader runs and detects that you have to fall back. [...]

You would end up in the new version, no? But yes, the new bootloader runs.

> [...] How would you proceed here to ensure that you do not boot the old kernel with your new bootloader? If I'm not fully mistaken you would have to switch eMMC bootpart and reboot again. Is that what you were thinking about?

Correct. At least for U-Boot this could be done with `mmc partconf`.

jluebbe, 2023-10-24:
From RAUC's perspective, all slots before the A/B decision point are 'atomic' only so there cannot be any fallback. Accordingly, they are not part of the `rauc status mark-*` mechanism.

If a different behavior is required, this probably means a different slot type for the eMMC boot partitions with a separate slot instance per partition. Also it would likely require a new/custom bootloader backend which understand how fallback is implemented.

closed 2023-10-24 by jluebbe