rauc/rauc #639: Partition table corrupted after update using casync bundles with eMMC bootloader slot

mschwan-phytec, 2020-10-07:
When installing casync bundles and an eMMC bootloader slot is defined, the partition table becomes corrupted after an update.

*Without* the eMMC bootloader slot and the corresponding files in the bundle the system updates fine and the partition table is not being corrupted.

This is the partition layout I used. It is a simple A/B layout:
```
root@machine:~# fdisk -l
Disk /dev/mmcblk2: 15 GB, 15913189376 bytes, 31080448 sectors
485632 cylinders, 4 heads, 16 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Device       Boot StartCHS    EndCHS        StartLBA     EndLBA    Sectors  Size Id Type
/dev/mmcblk2p1    0,130,3     16,211,3          8192     270335     262144  128M  c Win95 FAT32 (LBA)
/dev/mmcblk2p2    16,211,4    277,232,19      270336    4464639    4194304 2048M 83 Linux
/dev/mmcblk2p3    277,232,20  294,58,20      4464640    4726783     262144  128M  c Win95 FAT32 (LBA)
/dev/mmcblk2p4    294,58,21   555,79,36      4726784    8921087    4194304 2048M 83 Linux
```
The `system.conf` of RAUC looks like the following. Note the bootloader slot `[slot.bootloader.0]` for updating the bootloader areas of the eMMC.
```
root@machine:~# cat /etc/rauc/system.conf
[system]
compatible=machine
bootloader=uboot
mountprefix=/mnt/rauc

[keyring]
path=ca.cert.pem

# Bootloader
[slot.bootloader.0]
device=/dev/mmcblk2
type=boot-emmc

# System A
[slot.rootfs.0]
device=/dev/mmcblk2p2
type=ext4
bootname=system0

[slot.boot.0]
device=/dev/mmcblk2p1
type=vfat
parent=rootfs.0

# System B
[slot.rootfs.1]
device=/dev/mmcblk2p4
type=ext4
bootname=system1

[slot.boot.1]
device=/dev/mmcblk2p3
type=vfat
parent=rootfs.1
```
The casync bundle and chunk store reside in `/tmp` for testing this issue:
```
root@machine:~# ll /tmp/
drwxrwxrwt    5 root     root         140 Oct  7 12:28 ./
drwxr-xr-x   18 root     root        4.0K Oct  6 13:59 ../
drwx------ 4109 19695    10513      80.3K Oct  7 12:29 casync-bundle.castr/
-rw-------    1 root     root      175.3K Oct  7 12:28 casync-bundle.raucb
```
Interestingly, the update itself proceeds just fine:
```
root@machine:~# rauc install /tmp/casync-bundle.raucb
Installing
  0% Determining slot states
 20% Determining slot states done.
 20% Checking bundle
[... 87 lines cut ...]
Oct 07 12:31:04 machine rauc[4328]: mkfs.fat: warning - lowercase labels might not work properly with DOS or Windows
Oct 07 12:31:04 machine rauc[4328]: mkfs.fat 4.1 (2017-01-24)
Oct 07 12:31:04 machine rauc[4328]: Mounting vfat slot /dev/mmcblk2p1
Oct 07 12:31:04 machine rauc[4328]: Extracting /run/rauc/bundle/boot.tar.gz.img.caidx to /run/rauc/boot.0
Oct 07 12:31:06 machine rauc[4328]: Unmounting vfat slot /dev/mmcblk2p1
Oct 07 12:31:07 machine rauc[4328]: installing /tmp/casync-bundle.raucb: Updating slot boot.0 status
Oct 07 12:31:07 machine rauc[4328]: Updating slot file /run/rauc/boot.0/slot.raucs
Oct 07 12:31:07 machine rauc[4328]: Checking image type for slot type: boot-emmc
Oct 07 12:31:07 machine rauc[4328]: installing /tmp/casync-bundle.raucb: Updating slot boot.0 done
Oct 07 12:31:07 machine rauc[4328]: Image detected as type: *.img.caibx
Oct 07 12:31:07 machine rauc[4328]: installing /tmp/casync-bundle.raucb: Checking slot bootloader.0
Oct 07 12:31:07 machine rauc[4328]: Slot needs to be updated with /run/rauc/bundle/emmc-boot.img.caibx
Oct 07 12:31:07 machine rauc[4328]: Updating /dev/mmcblk2 with /run/rauc/bundle/emmc-boot.img.caibx
Oct 07 12:31:07 machine rauc[4328]: installing /tmp/casync-bundle.raucb: Updating slot bootloader.0
Oct 07 12:31:07 machine rauc[4328]: Extracting /run/rauc/bundle/emmc-boot.img.caibx to /dev/mmcblk2
Oct 07 12:31:07 machine rauc[4328]: No seed slot available for bootloader
Oct 07 12:31:07 machine rauc[4328]: Marking target slot rootfs.0 as bootable...
Oct 07 12:31:07 machine rauc[4328]: installing /tmp/casync-bundle.raucb: Updating slot bootloader.0 status
Oct 07 12:31:07 machine rauc[4328]: installing /tmp/casync-bundle.raucb: Updating slot bootloader.0 done
Oct 07 12:31:07 machine rauc[4328]: Updating slot file /run/rauc/rootfs.0/slot.raucs
Oct 07 12:31:07 machine rauc[4328]: installing /tmp/casync-bundle.raucb: All slots updated
Oct 07 12:31:07 machine rauc[4328]: installing /tmp/casync-bundle.raucb: finished
Oct 07 12:31:07 machine rauc[4328]: installing `/tmp/casync-bundle.raucb` succeeded
```
Any ideas how I can further debug this issue? Has anyone else experienced this?

jluebbe, 2020-10-07:
The boot-emmc type should only touch the mmcblk2boot0/1 partitions, which are exposed as block devices separate from the normal mmcblk2 ("user data area"). In old kernels, there was a corruption issue with accessing these partitions, though (see the note in https://rauc.readthedocs.io/en/latest/advanced.html#update-emmc-boot-partitions). Does your kernel include the fix?

What's the content of the corrupted MBR? If it's not the issue above, maybe record an installation of a bundle containing only the bootloader via strace?

mschwan-phytec, 2020-10-07:
Thanks for the hints, @jluebbe!

The aforementioned patch is included in our kernel.

The content of the MBR is blank after an update. :open_mouth: 
```
root@machine:~# dd if=/dev/mmcblk2 bs=512 count=1 | hexdump -C
1+0 records in
1+0 records out
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000200
```

I will record an installation via strace and report back.

mschwan-phytec, 2020-10-13:
I recorded a bundle installation with only the bootloader with strace but I am not sure how this would help. How would one distinguish writes to the MBR from other writes e.g.?

Now, I tried something different and simply printed the pointer array of casync extract arguments in RAUC. This is the output for the bootloader slot:
```
casync extract --store /tmp/casync-bootloader-bundle.castr --seed-output=no /run/rauc/bundle/emmc-boot.img.caibx /dev/mmcblk2
```
`emmc-boot.img` is the bootloader image with a fixed amount of zeros prepended (because our used SoC expects the bootloader to be written at a specific offset). At first sight, everything seems to be right, except the last argument: Shouldn't casync extract to a boot partition `/dev/mmcblk2bootX` instead of the user area `/dev/mmcblk2`? This would at least explain why the MBR is overwritten with zeros (the zeros from the beginning of `emmc-boot.img`).

I looked at the source code a little more and found that `updatepairs[]` defines the following:
```c
RaucUpdatePair updatepairs[] = {
        ...
        {"*.img.caibx", "*", img_to_raw_handler}, /* fallback */
        ...
#if ENABLE_EMMC_BOOT_SUPPORT == 1
        {"*.img", "boot-emmc", img_to_boot_emmc_handler},
#endif
        ...
```
There is no case for a `"*.img.caibx"` as the source and `"boot-emmc"` as the destination. So `img_to_raw_handler` is used. The issue with this is that `img_to_raw_handler` does not account for eMMC boot partitions and writes to the user area instead.

I guess the solution would be to implement a new handler that properly writes `*.img.caibx` files to eMMC boot partitions?

jluebbe, 2020-10-13:
> There is no case for a `"*.img.caibx"` as the source and `"boot-emmc"` as the destination. So `img_to_raw_handler` is used. The issue with this is that `img_to_raw_handler` does not account for eMMC boot partitions and writes to the user area instead.
> 
> I guess the solution would be to implement a new handler that properly writes `*.img.caibx` files to eMMC boot partitions? Or we could change `img_to_boot_emmc_handler` to both accept `*.img` and `*.img.caibx` files.

Ah, yes. The combination of the boot-emmc (probably also boot-mbr/gpt-switch) is not supported for casync. It's not straight forward to introduce casync in all the handlers, as it doesn't support piping the output. So for these special cases, may need to extract into a temporary file first, and then use the real handler.

OnkelUlla, 2020-10-14:
> Ah, yes. The combination of the boot-emmc (probably also boot-mbr/gpt-switch) is not supported for casync. It's not straight forward to introduce casync in all the handlers, as it doesn't support [...]

As mentioned in https://github.com/systemd/casync/pull/227 supporting pipes should be a simple and straight forward extension analog to what already has been done there.

mschwan-phytec, 2020-10-15:
> Ah, yes. The combination of the boot-emmc (probably also boot-mbr/gpt-switch) is not supported for casync. It's not straight forward to introduce casync in all the handlers, as it doesn't support [...]

Not sure why you would want to pipe the output through a file first? At least for "boot-emmc" it would be easier to just use `write_image_to_dev()` which can distinguish between normal and casync images. This is also the way `img_to_{raw,fs}_handler()` are implemented. I tried this and it seems to work fine. "boot-mbr-switch" may be little bit trickier but could be done in a similar way. I do not have a setup for the latter to test though. What do you think?

jluebbe, 2020-10-15:
> Not sure why you would want to pipe the output through a file first? At least for "boot-emmc" it would be easier to just use `write_image_to_dev()` which can distinguish between normal and casync images.

Looking at the code again, the fallback to the `img_to_raw_handler()` should have been prevented by the `{"*", "boot-emmc", NULL},` entry. That doesn't work because the `{"*.img.caibx", "*", img_to_raw_handler},` is too early in the list. Could you check if moving that to the end of the list causes the proper `Unsupported image emmc-boot.img.caibx for slot type boot-emmc` error?

Yes, `write_image_to_dev()` looks simpler for this case, but: `casync_extract_image()` looks for an active member of the same slot class to use as a seed, which doesn't exist for `boot-*` slots in the way it does for normal slots. Without a seed device, there is no real benefit to using casync for these slots. So either we could disable casync for these (small) slots, or integrate it more deeply to allow seeding from the boot device.

> This is also the way `img_to_{raw,fs}_handler()` are implemented. I tried this and it seems to work fine. "boot-mbr-switch" may be little bit trickier but could be done in a similar way. I do not have a setup for the latter to test though. What do you think?

The additional complication there is that we need to write to the device at an offset, which casync can't do on it's own. I suspect that going via a temp file allows better code sharing between these three `boot-*` handlers, though.

mschwan-phytec, 2020-10-16:
> Looking at the code again, the fallback to the `img_to_raw_handler()` should have been prevented by the `{"*", "boot-emmc", NULL},` entry. That doesn't work because the `{"*.img.caibx", "*", [...]

Checked that and yes, moving it down in the list results in a proper error message. I can send a quick PR so that users at least see that casync is not yet supported for "boot-emmc" slots.

> Yes, `write_image_to_dev()` looks simpler for this case, but: `casync_extract_image()` looks for an active member of the same slot class to use as a seed, which doesn't exist for `boot-*` slots in the way it does for normal slots. Without a seed device, there is no real benefit to using casync for these slots. So either we could disable casync conversion for these (small) slots, or integrate it more deeply to allow seeding from the boot device.

I somewhat lean towards the latter and would like to integrate it that way, so casync uses the boot partitions for seeding. I think "boot-emmc" is also a little different from the other bootloader slot types in this regard, as the partitions are fixed and not swapped during the update process. Maybe it makes sense to have that reflected in the implementation.

But I can wait for systemd/casync#227 to get an answer. I am not yet settled down for anything.