rauc/rauc #641: Slow tar.bz2 decompression.

dev-0x7C6, 2020-10-15:
Hi,

I'm playing with `rauc` and it's a great tool.

But unfortunately it is very slow for me. I check what is going on and I found that the bottle neck is a  `tar.bz2` decompression. My `raucb` is about `500MiB` in size and decompression taking a long time, decompression rate is about `4~5MiB/s` on `rpi4` which is disappointing.

Even on my PC (with `Ryzen 3900X`) decompression rate of `tar.bz2` is `19 MiB/s`.

Is there a way to turn that compression and just store plain tar inside raucb? I'm using it with Yocto with `meta-rauc`, I'm only see there is possibile compression on squashfs - which maybe useful after storing uncompressed file (I'm using `bundle.bbclass`).

Cheers,
Bartłomiej

ejoerns, 2020-10-15:
@dev-0x7C6 decompression speed is not RAUC-specific. RAUC will only call ``tar`  command to extract the archive with the required decompression algorithm. Thus if a tar.bz2 image is to slow, simply use an algorithm with a quicker decompress speed, like gzip or xz. Or, as you noted, as the squashfs itself compresses, simply place an uncompressed tar archive in the bundle.
You also have of course the option to use a file system image instead, like ext4.

dev-0x7C6, 2020-10-15:
Ok I understand now, good design choice.

If you don't mind I need little more help.

I'm using [bundle.bbclass](https://github.com/rauc/meta-rauc/blob/master/classes/bundle.bbclass) to generate additional image bundles and `tar.bz2` is used by default. I'm unsure how to change it. Should I pass additional `BUNDLE_ARGS`-  I failed to find additional parameters that could help with this case. I searched in documentation https://rauc.readthedocs.io/en/latest/using.html#creating-bundles but there is no mentions about advanced parameters and such.

Inside `bundle.bbclass` I found that squash compression can be done by passing `--mksquashfs-args="-comp zstd -Xcompression-level 22"`, but missing part is how to pass uncompressed tar using `rauc bundle`

ejoerns, 2020-10-15:
> Inside bundle.bbclass I found that squash compression can be done by passing --mksquashfs-args="-comp zstd -Xcompression-level 22", but missing part is how to pass uncompressed tar using rauc bundle

You would need to configure your BSP to build a tar archive. This is what you control with `IMAGE_FSTYPES`

    IMAGE_FSTYPES = "tar"

This is something you should set in your machine.conf or image recipe.

Then for the bundle (that only takes the generated image and packs it), you need to set something like:

    RAUC_SLOT_rootfs[fstype] = "tar"

dev-0x7C6, 2020-10-16:
Ok thanks for help.

closed 2020-10-16 by dev-0x7C6