rauc/rauc #643: rauc spontaneously changes a partition

jkryszyn, 2020-10-16:
Hello,
from time to time (very rarely, there is no known way to reproduce it) my device boots from another partition. My guess is that the current partition is marked as bad because of unknown reason. How can I debug this problem and solve it? This is an inherited project which I am not yet familiar with and I have very little knowledge about rauc so please forgive me If there is no crucial info in this post. Maybe I could hint something with my `mark-good.service`:

```bash
[Unit]
Description=RAUC Good-marking Service
ConditionKernelCommandLine=|bootchooser.active
ConditionKernelCommandLine=|rauc.slot
[Service]
ExecStart=/usr/bin/rauc status mark-good
[Install]
WantedBy=multi-user.target
```

My current idea is to create a file indicating that the current partition is ok and create a script executed by `mark-good.service` which checks if the file exists and if yes than marks slot as good. During update this file would be deleted from the current partition and created on the updated partition. Is it ok?

ejoerns, 2020-10-16:
@jkryszyn If your board boots into the wrong slot/partition, then this should happen because the bootloader and the underlying boot selection mechanism decided to.

What bootloader do you use?

If the bootloader decides to boot the "wrong" partition, then I would assume either the boot selection mechanism has a flaw, the mark-good service is not executed as expected or the userspace and the bootloader write their status into different places.

For most implementations we recommended to default to 3 attempts to boot one slot. Thus if you can check if your system changes the slot after 3 (or n) boots, this could help to encircle the issue behind it.

jkryszyn, 2020-10-16:
I use u-boot. After some searching I found patches to files related to boot and there are 3 attempts.

jkryszyn, 2020-10-22:
@ejoerns ? 
One more question. I noticed another strange behavior. Let's assume that I have two versions of my software: 1.0 and 1.1. The latter differs from 1.0 with one file placed in ` /root` folder. Let's call this file `/root/test`. 1.1 has also a post-install handler which deletes `/root/test` after update.

1. I program my device from SD card with 1.0 - both slots have the same filesystem
2. I make an update to 1.1 using rauc. Current slot now is B (1.1). There is `/root/test`
3. I make an update to 1.0 using rauc. Current slot now is A (1.0). Post-install handler from 1.1 deleted `/root/test` on slot B.
4. I make an update to 1.1 using rauc. Current slot now is B (1.1) but there is no `/root/test`

Why `root/test` is missing in step 4?

ejoerns, 2020-10-22:
> @ejoerns ?

In my last message I wrote

> Thus if you can check if your system changes the slot after 3 (or n) boots, this could help to encircle the issue behind it.

Did you check that? Becaus it is quite hard to say what went wrong without knowing more details. In general  the U-Boot booting should work as far as I see.

Another option would be to have a dump of then env (before and) after the sudden slot change happened. This could show if there was any attempts counter decreased or not. It's sufficient to filter this on the interesting variables involved (`BOOT_x_LEFT`, `BOOT_ORDER`).

From the running system, a `rauc status` should also at lest tell you if it considers a slot as "good" or "bad" from the bootloader perspective.

> One more question. I noticed another strange behavior. Let's assume that I have two versions of my software: 1.0 and 1.1. The latter differs from 1.0 with one file placed in ` /root` folder. Let's [...]

The post-install hander deletes '/root/test` from the current system, not the just-installed one right?

> 1. I program my device from SD card with 1.0 - both slots have the same filesystem
> 2. I make an update to 1.1 using rauc. Current slot now is B (1.1). There is `/root/test`
> 3. I make an update to [...]

Which RAUC version is that? And is the additional file actually part of the update image or does it come from any hook?

> 
> Why `root/test` is missing in step 4?

My first quess is that it has something to do with the slot-skipping capability of RAUC, but this is disabled by default in more recent version and can only be the case if the image hashes of 1.0 and 1.1 are acutally the same.  In this case, RAUC won't re-write the images by default to save time.

jkryszyn, 2020-10-22:
> In my last message I wrote
>
>
> Did you check that? Becaus it is quite hard to say what went wrong without knowing more details. In general the U-Boot booting should work as far as I see.

I am unable to do this. As I mentioned, this happens very rarely and there is no way to reproduce it. I haven't seen this behavior with my own eyes. There are only statements of two users (among many) who described strange behaviour of their devices. The only explanation of what they observed is that rauc changed the slot. This is the problem - I am working in the dark, I have to find a solution of a problem which I can't repeat.
 
> Another option would be to have a dump of then env (before and) after the sudden slot change happened. This could show if there was any attempts counter decreased or not. It's sufficient to filter [...]

Ok, I will try to introduce this in the next update, although It can take some time until I get any results from the users.

> The post-install hander deletes '/root/test` from the current system, not the just-installed one right?

Yes, exactly. This handler works in step 3, after the update. It deletes the file from the current system.

> Which RAUC version is that? And is the additional file actually part of the update image or does it come from any hook?

rauc 1.1. Yes, this file is a part of the update image.
 
> My first quess is that it has something to do with the slot-skipping capability of RAUC, but this is disabled by default in more recent version and can only be the case if the image hashes of 1.0 and 1.1 are acutally the same. In this case, RAUC won't re-write the images by default to save time.

Ok, so this is my case. Can I do something about it (besides upgrading rauc)? More precisely: can I disable slot-skipping with next update using some post-install handler?

ejoerns, 2020-10-22:
> I am unable to do this. As I mentioned, this happens very rarely and there is no way to reproduce it. I haven't seen this behavior with my own eyes. There are only statements of two users (among [...]

So if you cannot reproduce this and it works well for you after having tested a number of update cycles this should actually be hard to debug, yet.
It could also be the respective devices were switched on and off too fast in a row so that the good-marking was never reached.

Is the fallback a major issue? Maybe you then want to disable fallback at all, or ensure both partitions are updated to the same state each time.

> Ok, so this is my case. Can I do something about it (besides upgrading rauc)? More precisely: can I disable slot-skipping with next update using some post-install handler?

The logic was actually changed right *after* the v1.1 release. Thus you best option is to ship a recent rauc version with your next update. The second best option ist to ship an updated system.conf with ``force-install-same=true`` set for each slot.

jkryszyn, 2020-10-23:
@ejoerns thank you very much! Your comments are very valuable. Indeed, the best option would be to update both partitions. This would solve a lot of compatibility issues I have to think of otherwise. I will try to do that.

For now I've made an update which contains modified system.conf file (as you stated) and this seems to work ok.

ejoerns, 2020-10-23:
Good to hear. For now, I consider this to be resolved. Feel free to reopen if that is not the case or you got any new findings from the field.

closed 2020-10-23 by ejoerns