rauc/rauc #651: How to invoke/replace an expired self-signed key?

fei4xu, 2020-10-28:
Hi,

I'm not an expert of the CA and public/private key related things, so I used the **self-signed key** as documented at (https://rauc.readthedocs.io/en/latest/advanced.html#single-key), things worked great before.

Now I realized that the key is expired and I can't do rauc update anymore. I think I should generate a new key/cert pair or something else.

What I have tried:
1. Generate a new self-signed key/cert pair again. 
This does not work and there is an error during `rauc install`. I think this is expected since it's clearly stated in [the document](https://rauc.readthedocs.io/en/latest/advanced.html#single-key) that I can't **replace expired certificates**.
2. Try to use current private key (key.pem) and generate a new public key with new expiration date. I use this command:
```sh
openssl req -x509 -newkey rsa:4096 -nodes -key dev.key.pem -out dev.cert.pem -days 365
```
I'm using the `-key dev.key.pem` parameter hoping that openssl can use existing private key to generate a new public key (again, I'm not an expert of openssl). But when I run the rauc update, same error happens.

Then I read the document again, it seems that the [Simple-CA](https://rauc.readthedocs.io/en/latest/advanced.html#simple-ca) is promising, but I don't know how to do it.

I'm using the yocto project with meta-rauc, so the last choice might be I update my system using a whole SD image update, but that's another story without rauc and I can't do it since SD image update is very difficult to do now to my device.

Any suggestions?

Thanks!

fei4xu, 2020-10-28:
My issue is very similar to this one: https://www.mail-archive.com/rauc@pengutronix.de/msg00160.html

But still I don't know the details of what I can do now.

fei4xu, 2020-10-28:
UPDATE:

If I generate a new key, and copy the key to `/etc/rauc/*.cert.pem` before I do `rauc instell`, the new key will just work even if it's self-signed.

Is this by-design?

I read the doc again:
> you can’t:
replace expired certificates **without updating the keyring**

The phrase of **without updating the keyring** seems to unclear to me, does it mean **if** I copy the new cert.pem to /etc/rauc, the expired certificates can be replaced?

ejoerns, 2020-10-28:
Hi @fei4xu 

> 1. Generate a new self-signed key/cert pair again.
>    This does not work and there is an error during `rauc install`. I think this is expected since it's clearly stated in [the document](https://rauc.readthedocs.io/en/latest/advanced.html#single-key) that I can't **replace expired certificates**.

yep, that won't work. That's always the tricky part of security: If you don't do it right it will be *securer* than you expected. :smirk:
An actually expired certificate on the target side will never allow you to update the target if expiration times are checked as we do in RAUC by default.

A similar request to what you attempt to do could be https://github.com/rauc/rauc/issues/353 which also describes a workaround using signing-time for checking (instead of system time).

> 2. Try to use current private key (key.pem) and generate a new public key with new expiration date. I use this command:

Yes, as this new public key is then fully unrelated to the public key stored in the keyring on the target.

> Then I read the document again, it seems that the [Simple-CA](https://rauc.readthedocs.io/en/latest/advanced.html#simple-ca) is promising, but I don't know how to do it.

The simple-ca docs should describe that, but it is just another method of generating keys. The base considerations required remain the same.

For these, please also have a look at our design checklist https://rauc.readthedocs.io/en/latest/checklist.html#security that gives some good hints, at least at what to think about :)

> I'm using the yocto project with meta-rauc, so the last choice might be I update my system using a whole SD image update, but that's another story without rauc and I can't do it since SD image update is very difficult to do now to my device.
> 
> Another option is to change the date/time back to do the rauc, but I'm not sure it's a long-term solution.

If you have the option, you could do as a short-term solution. For a long-term solution creating a certificate with an infinite lifetime would be the way to go. You can see this for example for the root cert in our test example script: https://github.com/rauc/rauc/blob/master/test/openssl-ca-create.sh. However, when caring about security this is only really useful with derived certs that own a shorter period. But if you do not need to care much about security, then using a self-signed one only is the way to go.
With a RAUC update, you can always fully replace the keyring as it is part of the system.

> Any suggestions?
> 
> Thanks!

Hope that helped

Regards, Enrico

closed 2020-10-31 by fei4xu