rauc/rauc #653: Can RAUC be used to update linux kernel?

fei4xu, 2020-11-06:
Hi,

I'd like to see if rauc is capable to update linux kernel?

My use case is like this:

Recently I need to add `CONFIG_TUN=y` to my linux kernel building, but our current rauc configuration seems to only support rootfs update, the changed linux kernel with `tun` module never updates.

I'd like to check if this is a configuration issue, how can I configure rauc to update new kernel?

ejoerns, 2020-11-06:
@fei4xu I assume you have a separate kernel partition? Do you have it redundant as the rootfs? Or where do you boot from exactly?

If this is not yet covered by the current configuration you could add another update step to roll our a modified configuration for example.

fei4xu, 2020-11-07:
I have a separate kernel partition, it's not redundant now but we can change the design.

So can I say rauc is partition based, if I design to have a redundant kernel partition, rauc is able to update it?
What else support do I need, for example, u-boot modification/configuration?

ejoerns, 2020-11-11:
> So can I say rauc is partition based, if I design to have a redundant kernel partition, rauc is able to update it?

Basically, yes.

The eMMC partitioning is a bit named as if it would be used for A+recovery but from the sizes mentioned it is actually appropriate for full A+B redundancy.

For the kernel slots you might probably want to add

``` ini
[slot.kernel.0]
device=/dev/mmcblk0p1
type=raw <-- depends on how you place the kernel here. I assume you do not have a file system here
parent=rootfs.0 <-- this links the two partitions together to one instance of the A/B scheme

[slot.kernel.1]
device=/dev/mmcblk0p2
type=raw
parent=rootfs.1 <-- this is linked to the other rootfs then
```

As a result of this, RAUC will always update rootfs.0 together with kernel.0 and rootfs.1 together with kernel.1

Your bundle then needs to be extended like:

```
RAUC_BUNDLE_SLOTS = "rootfs kernel"
RAUC_slot_kernel = "linux-yocto"
RAUC_slot_kernel[type] = "kernel"
```

closed 2020-11-11 by fei4xu