rauc/rauc #659: Is it possible to update storage device while booted from another?

draokweil, 2020-11-12:
Hi, I have the following setup:
Board contains both internal eMMC storage and microSD card slot (/dev/mmcblk1 and /dev/mmcblk0).
Also Board can boot separately from one or another device (just need to change physical switch position to specify what device you need). 
Both SD and eMMC have the same systems and partition scheme on it. Only rauc system.conf differs in 'device' field.
Using A-B-uboot partition scheme on each storage device. Devices can access each other in runtime.

Is it  possible for rauc to install image to eMMC device if I execute 'rauc install image_for_emmc' from operation system that booted from SD-card?

ejoerns, 2020-11-12:
What is the use case for that? factory bootstrapping? Development?

RAUC can potentially install on everything properly described as a slot in the system configuration.
If I got it right, then you have two independent storage media each prepared to hold a fully redundant system. So there will be no description for the slots of the other storage medium and thus RAUC cannot know what to update.

ejoerns, 2021-01-08:
@draokweil Any updates on this, or did this already sufficiently answer your question and we can close the issue?