rauc/rauc #665: is_region_free fails for MBR with less than 4 partition entries

MartinH76761, 2020-12-01:
In mbr.c function is_region_free() the code scans the MBR partition table to assure the bootloader region doesn't overlap with existing partitions. However, the upper loop boundary is hardcoded as 4 partition table entries. This fails when using less partitions, in my case 3 partitions.

> LastError: Installation error: Failed updating slot bootloader.0: Region start address 0x40000 is in area of partition 3 (0x0 - 0xffffffffffffffff)

The code needs some sort of detection for empty partition table entries.

MartinH76761, 2020-12-01:
I haven't had the time to investigate MBR specification how empty table entries are indicated / initialized, but here is some (untested!) snipped I am thinking about:

```
diff --git a/src/mbr.c b/src/mbr.c
--- a/src/mbr.c
+++ b/src/mbr.c
@@ -172,6 +172,9 @@ static gboolean is_region_free(guint64 region_start, guint64 region_size,
                p_end = (guint64)GUINT32_FROM_LE(partition_tbl[i].partition_size_le) * sector_size +
                        p_start - 1;
 
+               if (p_start == 0x0 && p_end == 0xFFFFFFFFFFFFFFFF)
+                       continue; // empty partition table entry
```

ejoerns, 2021-01-08:
@MartinH76761 Thanks for reporting this!

You are right we do not handle the case described.
One can relatively easy reproduce the issue by changing the test code to contain only 3 partitions:

``` diff
--- a/test/boot_switch.c
+++ b/test/boot_switch.c
@@ -424,7 +424,6 @@ int main(int argc, char *argv[])
                        "start=        2048, size=        6144, type=ef, bootable\n"
                        "start=       14336, size=       65536, type=83\n"
                        "start=       79872, size=           1, type=5\n"
-                       "start=       81920, size=       49152, type=82\n"
                ,
                .sfdisk_expect = R_QUOTE({
                        /* *INDENT-OFF* */
@@ -448,11 +447,6 @@ int main(int argc, char *argv[])
                                        "start" : 79872,
                                        "size" : 1,
                                        "type" : "5"
-                               },
-                               {
-                                       "start" : 81920,
-                                       "size" : 49152,
-                                       "type" : "82"
                                }
                                ]
                        }

```

and then run

``` shell
$ ./qemu-test shell
$ make test/boot_switch.test
$ ./test/boot_switch.test
```

The actual valid check for an empty partition should be checking for the number of sectors used (here `partition_size_le`) to be 0. This is at least what the fdisk code does: 

https://github.com/karelzak/util-linux/blob/master/libfdisk/src/dos.c#L158

Would you like to prepare a fix for this?

PR #972 (2022-09-26): src/mbr: skip empty partitions in is_region_free() check
Otherwise `partition_size_le` will be 0 and region calculation will be invalid and fail with something like:

> Failed updating slot bootloader.0: Region start address 0x40000 is in area of partition 3 (0x0 - 0xffffffffffffffff)

Checking for the partition size to be 0 is exactly what util-linux does in this case, too:

https://github.com/util-linux/util-linux/blob/364de8f4f5625d17094a4d6e9b5fb49cb19c20a4/libfdisk/src/dos.c#L158

To have test coverage for this, simply remove the fourth partition in corresponding test case.

Fixes: #665
commit 7b05b8fb (2022-09-26): src/mbr: skip empty partitions in is_region_free() check

Otherwise partition_size_le will be 0 and region calculation will be
invalid and fail with something like:

| Failed updating slot bootloader.0: Region start address 0x40000 is in area of partition 3 (0x0 - 0xffffffffffffffff)

Checking for the partition size to be 0 is exactly what util-linux does
in this case, too:

https://github.com/util-linux/util-linux/blob/364de8f4f5625d17094a4d6e9b5fb49cb19c20a4/libfdisk/src/dos.c#L158

To have test coverage for this, simply remove the fourth partition in
corresponding test case.

Fixes: #665
closed 2022-09-26 by jluebbe