rauc/rauc #667: RAUC Systemd service cannot find mount during bundle install

IvanSlaev, 2020-12-08:
Hello,
I have cross-compiled (not using any systems like Yocto, etc.) and installed RAUC myself on a custom board. Which was an opportunity as I have to supply all needed prebuild libraries myself too. Also I have used prebuild deb package of RAUC to install on the board. That was still not my problem.
I came to it as I started systemd service and tried to install update. The service could not found mount while starting rauc service from command line works just fine. That was the problem!
After a bit of debugging and headache I realised that started as systemd service, g_subprocess_newv cannot find mount as it is encoded in r_mount_full procedure (just adding mount to the command line):
g_ptr_array_add(args, g_strdup("mount"));
And I fixed it in two ways
1. supply the full path e.g.:
g_ptr_array_add(args, g_strdup("/bin/mount"));
2. or just supply the trivial PATH to the service file:
Environment=PATH=/home/someUser/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin

Aas RAUC does use g_subprocess_newv to call another tool, it is mandatory to add trivial PATH to systemd service and warn users to edit is if their one is not trivial.

ejoerns, 2021-01-07:
By default, systemd should already provide a default `PATH` in the units environments.
This is why this is not an issue by default (and hard-coding to `/bin/mount` will not work in all cases, too).

You can check this by using

```
# systemd-run env
Running as unit: run-rd7ae3b436b3a4c708a0c11af0413bb6b.service
# journalctl -u run-rd7ae3b436b3a4c708a0c11af0413bb6b.service
[...]
Jan 07 22:57:41 machine env[302]: PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
```

How does that look like in your case? Do you compile systemd with specific options concerning default env?

kamel-bouhara, 2021-02-22:
Hello,

I just reproduced this issue while trying to install a bundle I got the following message:

```
rauc-Message: 10:08:24.101: Formatting ubifs slot /dev/ubi0_2
(rauc:469): rauc-subprocess-DEBUG: 10:08:24.101: launching subprocess: mkfs.ubifs -y /dev/ubi0_2
(rauc:469): GLib-DEBUG: 10:08:24.102: posix_spawn avoided (fd close requested) (child_setup specified)
100% Copying image to rootfs.1 failed.
100% Updating slots failed.
(rauc:469): rauc-subprocess-DEBUG: 10:08:24.110: launching subprocess: umount /mnt/rauc/bundle
(rauc:469): GLib-DEBUG: 10:08:24.111: posix_spawn avoided (fd close requested) (child_setup specified)
100% Installing failed.

(rauc:469): rauc-WARNING **: 10:08:24.569: Installation error: Failed updating slot rootfs.1: failed to start mkfs.ubifs: Failed to execute child process ?mkfs.ubifs? (No such file or directory)
Installation error: Failed updating slot rootfs.1: failed to start mkfs.ubifs: Failed to execute child process ?mkfs.ubifs? (No such file or directory)
finished
```

I'm compiling rauc without systemd, it seems the absolute path isn't resolved in this case ?

FYI, The following patch will fix the issue :
```
diff --git a/src/update_handler.c b/src/update_handler.c
index ebadd97..1b38191 100644
--- a/src/update_handler.c
+++ b/src/update_handler.c
@@ -448,12 +448,13 @@ static gboolean ubifs_format_slot(RaucSlot *dest_slot, GError **error)
        gboolean res = FALSE;
        g_autoptr(GPtrArray) args = g_ptr_array_new_full(3, g_free);

-       g_ptr_array_add(args, g_strdup("mkfs.ubifs"));
+       g_ptr_array_add(args, g_strdup("/usr/sbin/mkfs.ubifs"));
```

Regards,
Kamel Bouhara

ejoerns, 2022-09-23:
> I'm compiling rauc without systemd, it seems the absolute path isn't resolved in this case ?

@kamel-bouhara RAUC cannot explicitly be compiled with or without systemd. You can use it with or without. In any case, the instance that runs RAUC must ensure to setup PATH properly. This is either your shell or, if run by D-Bus activation, either systemd or this helper: https://github.com/rauc/rauc/blob/master/data/rauc-service.sh.in

For the original question there was no further information provided, thus I will close this.

closed 2022-09-23 by ejoerns