rauc/rauc #670: Is it possible to report progress from a slot install hook script?

kgraefe, 2020-12-11:
I have a custom install hook script that extracts a tarball onto a partition which takes a minute or so. Is it possible to report progress  to my GUI through DBUS? I could not find anything in the documentation.

ejoerns, 2021-01-07:
@kgraefe we currently do not support fine-grained installation progress (i.e. per-slot-update progress) at all. Thus progress steps are incremented only when the next slot is updated, not to signal progress of file copying or tar extraction. Thus no, a hook cannot report progress at the moment.

What is the need to use a custom script for extraction (instead of using the default handler)?

Do you actually require progress (percentage) information or are your actually interested in getting some status or log / error messages from the hook script?

mbessler, 2021-01-07:
I'd like to second this request. In my case the latter (reporting status/progress and/or log/error messages) from the hook script would be nice.

kgraefe, 2021-01-07:
> @kgraefe we currently do not support fine-grained installation progress (i.e. per-slot-update progress) at all. Thus progress steps are incremented only when the next slot is updated, not to signal progress of file copying or tar extraction. Thus no, a hook cannot report progress at the moment.

I don't remember, tbh. Maybe I did not grasp all the built-in possibillities from the documentation. (It's a lot of documentation, which is great!)

> 
> What is the need to use a custom script for extraction (instead of using the default handler)?
> 
> Do you actually require progress (percentage) information or are your actually interested in getting some status or log / error messages from the hook script?

For me it would be nice to have progress percentage information. But it's only nice to have, no requirement. (And I know that would not very accurate for Tar extraction.)