rauc/rauc #671: Is casync support still experimental?

tortila, 2020-12-14:
Hi, this is not an issue but rather a question to the maintainers. 

I see that casync support was [added in v0.4](https://github.com/rauc/rauc/releases/tag/v0.4), and the documentation claims that [it is in experimental phase](https://rauc.readthedocs.io/en/latest/advanced.html#rauc-casync-support). Is it still the case with the most recent version, v1.4, after more than 2 years from the initial release?

ejoerns, 2021-01-07:
HI @tortila, sorry for the late reply (we were a bit busy with the 1.5 release the last weeks).

I would not call it 'experimental' anymore, no. We've seen different projects using or currently evaluating it. There are cases that are not fully covered by config options possibly, but basically, casync support with RAUC works as intended.

There are some current (fixed or fixable) limitations that one should be aware of like:

- https://github.com/systemd/casync/pull/227
- required chunks downloaded to local storage are not automatically removed at the moment

The only minor disclaimer I have to add is that from our experience, the development progress on casync itself has stalled a bit. But there are no known major issues for the most common use cases at the moment.
Potential alternatives that use the same chunk format etc. and provide either a larger set or a subset of features compared to casync are: [desync](https://github.com/folbricht/desync) (more features but in go) and [casync-nano](https://github.com/florolf/casync-nano) (aimed for embedded updating) (we did none of them evaluate, yet)

What would be your primary use case for casync? Streaming support? Saving disk space? Saving bundle size and bandwith?

om26er, 2021-01-19:
@ejoerns We are a bit concerned about the progress of casync as well. It seems priorities within RedHat changed and the project is no longer a "priority" etc.

It's also reassuring to know that there are some alternate projects out there that provide similar functionalities.

> What would be your primary use case for casync? Streaming support? Saving disk space? Saving bundle size and bandwith?

If I had to answer that: all.

tortila, 2021-01-20:
@ejoerns thanks for your insightful answer! I'm only evaluating different update systems for now. `casync` support in RAUC got my attention because in my case, the bandwidth on the device is limited. The streaming support could be of use as well, because the device is only online sporadically. 

By the way, I appreciate all the work that was put into RAUC's documentation - it's not only very useful for the tool itself, but is amazing in laying the fundamentals of updating of an embedded device for a novice in this field like me. Thank you!

ejoerns, 2021-02-09:
@om26er If you should plan to evaluate one of the alternatives, I would be interested in the results ;)

At least for streaming support: With the new 'verity' format where we do not need to download the entire bundle prior to verifying it, we have paved the way for adding streaming support into RAUC itself as well. (Delta updates would then be another later step maybe).

@tortila Thanks a lot. We've actually put some effort into the documentation, also with the aim to make people aware of the pitfalls to think about and to describe the base design ideas behind redundancy boot. But there are also still too many things (and will ever be) that deserve better documentation ;)

PR #957 (2022-09-29): Add casync tests and remove 'experimental' note from docs
The docs were saying casync support is still experimental. While casync was always basically working, this disclaimer was mainly added due to missing test coverage for casync-related code.

Add the missing tests and update the documentation.

Based on #956 
Fixes #671