rauc/rauc #678: Taking ownership of bundle file fails when file is placed on read-only mounted filesystem

mhei, 2021-01-06:
One usage scenario in our product is that the bundle can be placed on a USB pen drive which is mounted read-only. I tested rauc v1.5 and the update fails with the following message:
`failed to take ownership of bundle: failed to chmod bundle: Read-only file system`
Maybe this should be handled as special case?

jluebbe, 2021-01-06:
This was explicitly rejected as rauc cannot detect reliably it is mounted RW somewhere else or later at runtime.
 
Is the file owner or permissions the problem in your case? If it's only the permissions, you could use fmask=0022 as a mount option.

mhei, 2021-01-07:
Yes, I got it. And yes, our vfat mount options were complete stupid, but I've to blame myself... Sorry for the noise.

closed 2021-01-07 by mhei