rauc/rauc #679: casync bundle creation fails with "unable to find mounted device for bundle"

mschwan-phytec, 2021-01-06:
Since RAUC version 1.5 creating casync bundles no longer works. This is the relevant output when the `bundle.bbclass` executes the convert command to create the casync bundle:
```
rauc-Message: 13:21:55.950: Debug log domains: 'rauc'
(rauc:34224): rauc-DEBUG: 13:21:55.951: convert start
(rauc:34224): rauc-DEBUG: 13:21:55.951: system config not found, using default values
rauc-Message: 13:21:55.951: Failed to resolve realpath for '/dev/disk/by-uuid/be5faaeb-1680-4ca2-af65-f30f06dab903'
(rauc:34224): rauc-DEBUG: 13:21:55.951: input bundle: /home/user/documents/yocto/imx6ul/dunfell/build/tmp/work/machine/headless-bundle/1.0-r0/build/bundle.raucb
(rauc:34224): rauc-DEBUG: 13:21:55.951: output bundle: /home/user/documents/yocto/imx6ul/dunfell/build/tmp/work/machine/headless-bundle/1.0-r0/build/casync-bundle.raucb
rauc-Message: 13:21:55.951: Reading bundle: /home/user/documents/yocto/imx6ul/dunfell/build/tmp/work/machine/headless-bundle/1.0-r0/build/bundle.raucb
rauc-Message: 13:21:55.952: Verifying bundle signature...
(rauc:34224): rauc-DEBUG: 13:21:55.952: initial check_bundle_access failed with: unable to find mounted device for bundle
unable to find mounted device for bundle
WARNING: exit code 1 from a shell command.
```
I am looking at the code changes that version 1.5 brings but I do not really understand yet what the exact issue is with this one. AFAIK `check_bundle_access()` checks several parameters to prevent unintended write access to the bundle. Though I am unsure which device the error message is referring to and what is wrong with it.

jluebbe, 2021-01-06:
RAUC is trying to find the device where the bundle is located, to check if it is safe against concurrent modification. This may be too strict for the convert case...

Where is your `/home/user/documents/yocto/imx6ul/dunfell/build/tmp/work/machine/headless-bundle/1.0-r0/build/bundle.raucb` mounted from (according to mount)? Which filesystem is it? Does the /dev/disk/by-uuid/be5faaeb-1680-4ca2-af65-f30f06dab903 point to the correct block device?

jluebbe, 2021-01-07:
Do you get the same error when using `rauc info`? Does it disappear with `--no-verify`?

mschwan-phytec, 2021-01-07:
Ok, thanks @jluebbe for pointing me in the obvious direction. My build directory is located in a Docker container. Inside the container there is no access to `/dev/disk/by-uuid/be5faaeb-1680-4ca2-af65-f30f06dab903` which is a link to my build machine's root filesystem `/dev/sdb1`. `rauc info` however has the same behavior inside a container as it has outside of it: It fails without `--no-verify` because no keyring was given but runs fine with `--no-verify`. Inside a container `rauc info` does output that it cannot find said device, though it does not seem to have any negative effects.

The reason for building inside a container in this case is that my host distribution (Arch Linux) is not an officially supported system to build with in Yocto. There are also other scenarios where one might use containers (e.g. for automated builds).

Building "natively" (no container) works for the bundle, both the regular one as well as the casync bundle.

I am not sure how to proceed with this, as I consider the use-case of containers as quite usual. But this is also not solely an issue with RAUC.

mschwan-phytec, 2021-01-25:
Are there any new thoughts about this? Would it be possible to add an option that skips verification for `rauc convert`, or even `rauc bundle`? @jluebbe What do you think?

jluebbe, 2021-01-25:
This is related to the second part of #688. `rauc bundle` only verifies if you pass a keyring, but `rauc convert` is missing even the `--no-verify` option. As disabling signature verification completely seems a too large hammer, we could add a `--trust-environment` option (or `--no-exclusive-error`), which would reduce the error to a warning.

In the Yocto build, that could even be set by default, as we trust the environment there anyway.

sbaxter-izo, 2021-02-09:
I am experiencing the same issue with `rauc info` failing inside Docker without `--no-verify`, but succeeding when skipping verification. We run tests within our Yocto environment that ensure our bundles are signed properly by an internal python package that wraps `rauc`. Is there another way that we can still verify a signature while inside Docker so that we don't have to disable these tests until this bug is fixed?

Thanks!

ejoerns, 2021-09-10:
What I forgot to mention: Using the 'verity' format with casync and small rootfs sizes might be a problem. As we only handle the index files, the (compressed) squashfs will be quite small and might fall below the minimum size of 4096 bytes that is required for dm-verity to create a root hash.

closed 2022-01-26 by jluebbe

stsrc, 2022-03-10:
> In the Yocto build, that could even be set by default, as we trust the environment there anyway.

@jluebbe could you elaborate more about this statement, why we trust the environment of Yocto, if you remember? 
I have similar problem with "initial check_bundle_access failed with: unable to find mounted device for bundle" in yocto, patch with trust environment fixes it, but I have concerns about security.

jluebbe, 2022-03-10:
> > In the Yocto build, that could even be set by default, as we trust the environment there anyway.
> 
> @jluebbe could you elaborate more about this statement, why we trust the environment of Yocto, if you remember?

We have to ensure that the environment you're using for your build system is secure. If it wasn't, anything built in that environment could be compromised. It could even a modified version of RAUC that didn't perform any security checks.

So, if RAUC is called from a build system, any additional checks run by RAUC don't actually add any security.

This is different from the case where 'rauc convert' is run later on a different system and you want to check that the bundle you're converting is authentic.

stsrc, 2022-03-10:
ok, that solves my concerns. Thanks :)

PR #770 (2022-01-26): src: introduce --trust-environment option for 'rauc convert'
In some cases it might not be possible or wanted that RAUC (used as a host tool) performs its standard bundle access checks.

For example when running 'rauc convert' from within a (docker) container, RAUC might not be able to get the underlying device of the bundle's storage location. This way it cannot determine if the mount point is owend by root and will abort with

> initial check_bundle_access failed with: unable to find mounted device for bundle

While --no-verify potentially already works around this, it is a too big hammer fo [... cut]
commit 57f5adc4 (2022-01-26): src: introduce --trust-environment option for 'rauc convert'

In some cases it might not be possible or wanted that RAUC (used as a host
tool) performs its standard bundle access checks.

For example when running 'rauc convert' from within a (docker)
container, RAUC might not be able to get the underlying device of the
bundle's storage location. This way it cannot determine if the mount
point is owned by root and will abort with

| initial check_bundle_access failed with: unable to find mounted device for bundle

While --no-verify potentially already works around this, it is a too big
hammer f [... cut]