rauc/rauc #683: Handling the fallback when proper images are present in the partition

Noufalp90, 2021-01-13:
Hi,
I am facing an issue where, due to power shutdown my board got restarted multiple times and then it was observed that it has done partition A and partition B retry count and booting stuck at u-boot showing no retry left on the partition B. Is there any way i can overcome this issue ?

ejoerns, 2021-01-13:
As the scope of the question is not limited to meta-rauc, I've moved it to 'rauc'.

It should not be an issue if you retry booting in case of zero retries left.
Our example grub.conf should do this already: https://github.com/rauc/rauc/blob/master/contrib/grub.conf#L31
Do you use that?

Emantor, 2021-01-13:
> Our example grub.conf should do this already: https://github.com/rauc/rauc/blob/master/contrib/grub.conf#L31
> Do you use that?

We also have an example script for U-Boot [here](https://github.com/rauc/rauc/blob/master/contrib/uboot.sh
).

ejoerns, 2021-01-13:
Oh, I've linked and mentioned the wrong one, sorry ;) @Emantor thanks for fixing :+1:

Noufalp90, 2021-01-15:
@Emantor thank you for the insput. it does the resolved my issue

closed 2021-01-15 by Noufalp90