rauc/rauc #687: Updating single files

ArashJavan, 2021-01-20:
Hello everyone, 

I wondering if it is possible to just update a single file in my rootfs using rauc!
right now my **system.conf** -file looks like this:
```
[system]
compatible=My-Board
bootloader=uboot
 
[keyring]
path=/etc/rauc/demo.cert.pem

[slot.rootfs.0]
device=/dev/mmcblk0p4
type=ext4
bootname=A
 
[slot.rootfs.1]
device=/dev/mmcblk0p5
type=ext4
bootname=B
```

I first created a tar.xz file on my PC with following contenxt:
**rootfs.tar.xz**:
´´´
|- root
     |- test.log
´´´
As you can see I only want to update a single file, not the whole file system.

After that I created a **manifest file**:
```
[update]
compatible=My-Board
version=210120-162748

[image.rootfs]
sha256=7e553faab9325b5d67290f094f015a30f29dfea7fbd6cd5ff7e80d9402e93158
size=264
filename=rootfs.tar.xz
```

Then I copied the created 'my-update.raucb'-file on my board and tried to install it using:
´´´
rauc install ./my-update.raucb
´´´
But unfortunatly I get following error:

´´´
# rauc install ./my-update.raucb
installing
  0% Installing
  0% Determining slot states
 20% Determining slot states done.
 20% Checking bundle
 20% Verifying signature
 40% Verifying signature done.
 40% Checking bundle done.
 40% Loading manifest file
 60% Loading manifest file done.
 60% Determining target install group
 80% Determining target install group done.
 80% Updating slots
 80% Checking slot rootfs.0
 90% Checking slot rootfs.0 done.
 90% Copying image to rootfs.0
100% Copying image to rootfs.0 failed.
100% Updating slots failed.
100% Installing failed.
LastError: Installation error: Failed updating slot rootfs.0: failed to start mkfs.ext4: Failed to execute child process ?mkfs.ext4? (No such file or directory)
Installing `/root/./update-210120-162748.raucb` failed
´´´

Now, I would like to know if it is possible to update a single file using rauc?
Btw. I already have seen this issue: [497](https://github.com/rauc/rauc/issues/497#issue-507084158).

Thanks!

Emantor, 2021-01-20:
```
> LastError: Installation error: Failed updating slot rootfs.0: failed to start mkfs.ext4: Failed to execute child process ?mkfs.ext4? (No such file or directory)
> ```
> 
> Now, I would like to know if it is possible to update a single file using rauc?
> Btw. I already have seen this issue: [497](https://github.com/rauc/rauc/issues/497#issue-507084158).

As documented in the [required target tools](https://rauc.readthedocs.io/en/latest/integration.html?highlight=mkfs.ext4#required-target-tools) and output in the error message, you'll need `mkfs.ext4` on the device you are updating.

ArashJavan, 2021-01-21:
@Emantor thanks for you reply. I installed mkfs.ext4 (e2fsprogs) on the target board and did the same process again. 
This time rauc could update form the "my-update.raucb". 
The problem ist that rauc did not only update the file but the whole file system was overwritten, so everything was removed expect the only folder:
```
|- root
     |- test.log
```
And therefore when I change the slot to start from the updated slot, the system could not start sine everything was gone - no file system!

Is there anyway to only update just one file without removing everything else on the file system using rauc?

thanks

Emantor, 2021-01-21:
> Is there anyway to only update just one file without removing everything else on the file system using rauc?

No, RAUC is build as a slot-based updating system, where the slots always contain a whole linux system, which is updated in one go. If you are looking for an update system to update only single files, RAUC is not the correct tool for your use case. You might want to look into a file based updating system, i.e. something like a package manager.

jluebbe, 2021-01-21:
The intention is that a bundle always results in a known state for the newly installed system, which ensures that test results of one bundle stay valid independent of which state was on the target system before the update.

ArashJavan, 2021-01-21:
Thanks @jluebbe and @Emantor . I found out the only way to make file update using rauc, is just using "install-hooks" on the slot in manifest-file. This way I was able to update a file but unfortunately, this disables the whole rauc-machinery :(

closed 2021-01-21 by ArashJavan

gitgonewithwind, 2022-08-31:
@ArashJavan 
cc
@jluebbe and @Emantor 

HI ArashJavan,
I have the same duty to update single file ： for example at /home/myuser/test_executable_file .
can you tell me how to use the "install-hooks" on the slot in manifest-file. I have read the rauc Doc, but still not clear how to use the rauc (I have included it in my yocto image,but I do not know to use it)
can you completely post your  fsystem.conf  and  my-update.raucb    file and commands(eg: rauc install ./my-update.raucb ) . and one more question: where can I store the update file  test_executable_file  ?( in USB ? or on Internet somewhere?) I mean when I run the command (eg : rauc install ./my-update.raucb )where the comand can find the file to replace the old file ?
Thank you very much.

ejoerns, 2022-08-31:
@gitgonewithwind what is your motivation for updating a single file?

As noted before, the purpose of RAUC is primarily to bring the entire system into a well-defined state.
When you update only a single file, you have an arbitrary unreproducible state.

Do you update the base system with RAUC, too? Is this just an additional file you want to install or do you use RAUC solely for the installation of this file?

About where to put it for update: This is what RAUC bundles are. They contain update artifacts. Maybe invest the time and have a look at the documentation first. Also to see if RAUC actually fits your use case:

https://rauc.readthedocs.io/en/latest/basic.html#update-artifacts-bundles

gitgonewithwind, 2022-08-31:
Hi 
@ejoerns I have read the Doc ,but not completely understand it.  we  do not  want to update the entire system. we use yocto image.  the entire system rootfs is about 8G, it too big size. we can burn a new SD card to replace old  image on the SD card to boot the imx8 board.
so we only want to update our on key  executable(eg: /home/myuser/test_executable_file) and our node code in a folder (eg: /home/myuser/node).
we do not want to update uboot , kernal image, rootfs (so we only have one rootfs).
can you give me a detail example  how to update one files(or only a few files : /home/myuser/test_executable_file  /home/myuser/node) 
as you mentioned  that what RAUC bundles are . for my understanding in yocto ,  we can build image with  bitbake  fsl-fsl-image-validation-imx with .bb file fsl-image-validation-imx.bb

so I followed  the readme at here  https://github.com/rauc/meta-rauc/tree/sumo , I added rauc staff(as below) in it  and create a new .bb file: fsl-image-validation-imx-bundle-rauc.bb 
inherit bundle

RAUC_BUNDLE_COMPATIBLE ?= "imx8mqevk"

RAUC_BUNDLE_SLOTS ?= "rootfs"

RAUC_SLOT_rootfs ?= "fsl-image-validation-imx"

RAUC_KEY_FILE = "${COREBASE}/../meta-fsl-bsp-release/imx/meta-sdk/recipes-fsl/images/rauc-files/development-1.key.pem"
RAUC_CERT_FILE = "${COREBASE}/../meta-fsl-bsp-release/imx/meta-sdk/recipes-fsl/images/rauc-files/development-1.cert.pem"

and added other files system.conf(just add same thing like device=/home/myuser/test_executable_file)
--

so It can build a new image with:  bitbake fsl-image-validation-imx-bundle-rauc 
so I can burn a new image on SD card. and I can use it to boot my board as before and  run rauc and make sure it is running  after boot.
so I can directly to use it, where to update ??

Therefore I think I do not understand  how to use it.  can you give me some quick guide for it ? Thank you very much. 
let me read the Doc again.  I think my porting for yocto parts is correct, but I do no t know what is the next step .
 Should I need to change RAUC_BUNDLE_SLOTS ?= "rootf" to RAUC_BUNDLE_SLOTS ?= " /home/myuser/test_executable_file" ??

ejoerns, 2022-09-01:
> Hi @ejoerns I have read the Doc, but not completely understand it. we do not want to update the entire system. we use yocto image. the entire system rootfs is about 8G, it too big size. we can burn [...]

Reading this sounds to me like the blueprint of a system that RAUC is not designed for.
RAUC is primarily for atomic image-based updates of systems with redundant storage (A/B). None of these creteria apply.

It sounds challenging to me to think of not requiring updates of the base system in the field, but if you can exchange SD cards, it is probably not that critical and depends on your field of operation.

So the reason not to do it is to save time? Or bundle (and thus transmission) size?

On the other hand, the need to update the application independently from the rootfs is a topic that comes up more frequently.

> can you give me a detail example how to update one files(or only a few files : /home/myuser/test_executable_file /home/myuser/node) as you mentioned that what RAUC bundles are . for my understanding in yocto , we can build image with bitbake fsl-fsl-image-validation-imx with .bb file with fsl-image-validation-imx.bb

I really think you should not use this image as your product's image as it is built for a completely different intention. But basics of Yocto are out of scope here ;)

> so I followed the readme at here https://github.com/rauc/meta-rauc/tree/sumo , I added rauc staff(as below) in it and create a new .bb file: fsl-image-validation-imx-bundle-rauc.bb

I really hope that you accidentially chose the `sumo` version of the docu. If you are developing on a `sumo` BSP, which is technology short before the stone age, please consider moving to the present.

> ```
> inherit bundle
>
> RAUC_BUNDLE_COMPATIBLE ?= "imx8mqevk"
>
> RAUC_BUNDLE_SLOTS ?= "rootfs"
>
> RAUC_SLOT_rootfs ?= "fsl-image-validation-imx"
>
> RAUC_KEY_FILE = [...]

This is a bundle recipe for a meta-fsl-bsp-release-specific images. Neither the layer nor the imae should be used in this century.

> and added other files system.conf(just add same thing like `device=/home/myuser/test_executable_file`)
>
> so It can build a new image with: bitbake fsl-image-validation-imx-bundle-rauc so I can burn a [...]

No, the slot is the name of the slot. You can try using the file path as slot, it might work.

I would recommend using a distinct redundant application partition and update this with RAUC. Then you could replace the normal atomic bootloader switch by a custom backend that moves symlinks from your rootfs or something simliar.

But, since you probably do not need the capabilities of rauc (bootloader interaction, file system support, verification?) you are probably holding the wrong tool in your hands.

gitgonewithwind, 2022-09-01:
@ejoerns 
I would recommend using a distinct redundant application partition and update this with RAUC. Then you could replace the normal atomic bootloader switch by a custom backend that moves symlinks from your rootfs or something simliar.

you mean I can do like this:

change my  sd card partitions  without rauc(to boot my board)  to sd card  partitions with rauc:
*******                                         ********
 uboot                                         uboot
******                                        ***********
kernal      change it to               kernel
****                                            *****
rootfs                                         rootfs(wth rauc)
*****                                         ****
                                                  additional appfs (for me, files: test_executable_file node folder that link to rootfs(wth rauc)??
                                                 *****

right?

gitgonewithwind, 2022-09-01:
@ejoerns 
I would recommend using a distinct redundant application partition and update this with RAUC. Then you could replace the normal atomic bootloader switch by a custom backend that moves symlinks from your rootfs or something simliar.

you mean I can do like this:

change my  sd card partitions  without rauc(to boot my board)  to sd card  partitions with rauc:

partitions without rauc .....................partitiona with rauc

1. uboot        .........................................1. uboot

2. kernal    ..............change it to.......2. kernel

3. rootfs   ....................................... 3.rootfs(wth rauc)

 ..........................................................4. additional appfs 
............................................................. (for me, files: test_executable_file,node folder that link to rootfs(wth rauc)??

right?
So I only need to copy my updated  test_executable_file,node folder  to  partition 4 if I want to update them?

gitgonewithwind, 2022-09-20:
@ejoerns 

as you guide, I added an  redundant rootfs  successully. 
and my system.conf as below, and I put /etc/rauc/ca.cert.pem that generated in yocto meta-rauc. I can find it in my rootfs.
but I run rauc install or rauc info, it shows can not find the ::system library:fopen:No such file or directory:bss_file.c:175:fopen('/etc/rauc/ca.cert.pem ','r')    

: LastError: failed to load CA file '/etc/rauc/ca.cert.pem

did I generate wrong  ca.cert.pem file ?

Thank you.

raucbuc  install /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8m
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb: installing
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:   0% Installing
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:   0% Determining slot states
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:  20% Determining slot states done.
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:  20% Checking bundle
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:  20% Verifying signature
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:  40% Verifying signature failed.
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:  40% Checking bundle failed.
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb: 100% Installing failed.
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb: LastError: failed to load CA file '/etc/rauc/ca.cert.pem '
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb: idle
Installing `/run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb` failed
-sh-4.4# rauc info /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk
rauc-Message: Reading bundle: /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb
rauc-Message: Verifying bundle... 
281473440954144:error:02001002:system library:fopen:No such file or directory:bss_file.c:175:fopen('/etc/rauc/ca.cert.pem ','r')
281473440954144:error:2006D080:BIO routines:BIO_new_file:no such file:bss_file.c:182:
281473440954144:error:0B084002:x509 certificate routines:X509_load_cert_crl_file:system lib:by_file.c:253:
failed to load CA file '/etc/rauc/ca.cert.pem '

-sh-4.4# cat /etc/rauc/ca.cert.pem 
Certificate:
    Data:
        Version: 3 (0x2)
        Serial Number: 1 (0x1)
        Signature Algorithm: sha256WithRSAEncryption
        Issuer: O=Test Org, CN=Test Org rauc CA Development
        Validity
            Not Before: Jan  1 00:00:00 1970 GMT
            Not After : Dec 31 23:59:59 9999 GMT
        Subject: O=Test Org, CN=Test Org rauc CA Development
        Subject Public Key Info:
            Public Key Algorithm: rsaEncryption
                RSA Public-Key: (2048 bit)
                Modulus:
                    00:bc:36:f2:4f:d3:84:a6:81:8e:00:40:ae:10:d2:
                    59:78:b5:a7:89:97:70:23:6e:4b:89:66:e1:11:6c:
                    e8:c0:6a:27:37:fb:c2:aa:4c:3b:10:2c:98:8a:ea:
                    10:e2:72:bc:6f:dc:68:9b:63:b4:08:aa:26:7a:b0:
                    b7:5e:d5:82:e1:e7:88:46:b7:fc:1d:26:47:72:b1:
                    1d:54:f5:2d:87:b0:bc:b7:2c:ef:a8:28:2e:bd:94:
                    41:cf:62:89:a3:fe:d2:f5:08:b8:ee:86:ff:85:fd:
                    e9:a9:64:e0:54:34:95:bf:73:b3:dc:61:c0:4e:4e:
                    e2:b0:41:ca:de:0b:cb:52:66:c6:49:12:ba:e9:02:
                    fc:f6:e4:1d:2b:da:9a:06:58:86:f7:f0:41:0a:4a:
                    4e:55:74:19:5e:53:0f:db:56:a8:9e:8a:d6:bd:21:
                    94:5c:ef:18:11:04:e8:a7:09:fc:7b:2d:55:bb:29:
                    3b:86:e7:ea:a9:14:57:16:a1:1c:4e:1d:1b:44:a2:
                    39:f6:a2:ad:11:88:a4:81:79:e1:24:bc:fb:60:98:
                    9c:6d:50:b2:92:f7:46:07:46:d1:f7:65:4f:0c:25:
                    b4:8d:bc:97:0e:f7:4c:fb:83:51:71:37:01:38:ce:
                    f4:af:ea:97:b7:97:c0:48:83:1f:4d:8e:ba:44:6c:
                    da:8f
                Exponent: 65537 (0x10001)
        X509v3 extensions:
            X509v3 Subject Key Identifier: 
                78:15:43:91:42:AD:45:56:AD:BD:15:39:E6:0A:8B:9D:2B:26:18:15
[... 100 lines cut ...]
         1f:9c:d8:7b:72:91:93:7c:8d:a6:20:6a:88:16:cb:a5:2a:3c:
         bc:da:bc:ed:fc:56:90:9b:d1:53:f9:80:79:4f:06:a0:9d:b9:
         77:66:66:73:a0:71:83:f1:a7:b6:d2:a0:6b:fc:af:a8:3b:b9:
         f1:2a:5e:a5
-----BEGIN CERTIFICATE-----
MIIDhjCCAm6gAwIBAgIBATANBgkqhkiG9w0BAQsFADA6MREwDwYDVQQKDAhUZXN0
IE9yZzElMCMGA1UEAwwcVGVzdCBPcmcgcmF1YyBDQSBEZXZlbG9wbWVudDAgFw03
MDAxMDEwMDAwMDBaGA85OTk5MTIzMTIzNTk1OVowOjERMA8GA1UECgwIVGVzdCBP
cmcxJTAjBgNVBAMMHFRlc3QgT3JnIHJhdWMgQ0EgRGV2ZWxvcG1lbnQwggEiMA0G
CSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQC8NvJP04SmgY4AQK4Q0ll4taeJl3Aj
bkuJZuERbOjAaic3+8KqTDsQLJiK6hDicrxv3GibY7QIqiZ6sLde1YLh54hGt/wd
JkdysR1U9S2HsLy3LO+oKC69lEHPYomj/tL1CLjuhv+F/empZOBUNJW/c7PcYcBO
TuKwQcreC8tSZsZJErrpAvz25B0r2poGWIb38EEKSk5VdBleUw/bVqieita9IZRc
7xgRBOinCfx7LVW7KTuG5+qpFFcWoRxOHRtEojn2oq0RiKSBeeEkvPtgmJxtULKS
90YHRtH3ZU8MJbSNvJcO90z7g1FxNwE4zvSv6pe3l8BIgx9NjrpEbNqPAgMBAAGj
gZQwgZEwHQYDVR0OBBYEFHgVQ5FCrUVWrb0VOeYKi50rJhgVMGIGA1UdIwRbMFmA
FHgVQ5FCrUVWrb0VOeYKi50rJhgVoT6kPDA6MREwDwYDVQQKDAhUZXN0IE9yZzEl
MCMGA1UEAwwcVGVzdCBPcmcgcmF1YyBDQSBEZXZlbG9wbWVudIIBATAMBgNVHRME
BTADAQH/MA0GCSqGSIb3DQEBCwUAA4IBAQBFzRvOsOg3/zk6CmjhTpPVVEEkzoc3
L3wksCi6AEKi969nQnMVyUWLiVbp5aguQR7c25rdYTABrAYwE9RkQgn4Rfi1pDXc
IQW4i8aJ+dzOaTKpQ/UiihW1fJs4WALtEdHWZJ9VSrZpuM5uaodKD2I89CWylcJQ
Y35CGirMVUNCDDZBcz1wVukrACR0C45hcuufTdnxeYLnQixNv2qGE2xb+Rl5Puem
/Ndz3mvrPvf3BtBpKapuvzbKNyScL/hRGIP8htaMWrQfnNh7cpGTfI2mIGqIFsul
Kjy82rzt/FaQm9FT+YB5Twagnbl3ZmZzoHGD8ae20qBr/K+oO7nxKl6l
-----END CERTIFICATE-----

ejoerns, 2022-09-20:
Apparently, RAUC does not strip excess whitespace after file names.

Thus as a quick fix, pleas remove the space after your keyring path.

gitgonewithwind, 2022-09-20:
> Apparently, RAUC does not strip excess whitespace after file names.
> 
> Thus as a quick fix, pleas remove the space after your keyring path.

@ejoerns 
thank you so much for replying so soon.
got it! fix that and test more.

one more question , why  status for slot is bad?

rauc status

(rauc:4092): rauc-WARNING **: GDBus.Error:org.gtk.GDBus.UnmappedGError.Quark._g_2dio_2derror_2dquark.Code30: Failed getting primary slot: Failed getting primary slot: Failed to start fw_printenv: Failed)

Compatible:  My Example System
Variant:     
Booted from: /dev/mmcblk0p2
Activated:   (null) ((null))
slot states:
  rootfs.0: class=rootfs, device=/dev/mmcblk0p2, type=ext4, bootname=system0
      state=booted, description=, parent=(none), mountpoint=(none)
      boot status=bad
  rootfs.1: class=rootfs, device=/dev/mmcblk0p3, type=ext4, bootname=system1
      state=inactive, description=, parent=(none), mountpoint=(none)
      boot status=bad

gitgonewithwind, 2022-09-20:
@ejoerns 
fix the file name whitespace ,another error:
error: Compatible mismatch
how to do ?
thank you very much.

rauc info /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk
rauc-Message: Reading bundle: /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb
rauc-Message: Verifying bundle... 
Compatible:     'imx8mqevk'
Version:        '1.0'
Description:    'fsl-image-validation-imx-bundle-rauc version 1.0-r0'
Build:          '20220920155334'
Hooks:          ''
1 Image:
(1)     fsl-image-validation-imx-imx8mqevk.tar.bz2
        Slotclass: rootfs
        Checksum:  89a25a59788d8276b8bf0800e05714ad4af13eca0299b9b49e52c538f93db541
        Size:      1223237757
        Hooks:     
0 Files

Certificate Chain:
 0 Subject: /O=Test Org/CN=Test Org Development-1
   Issuer: /O=Test Org/CN=Test Org rauc CA Development
   SPKI sha256: EE:0C:61:9C:FB:94:46:4C:95:52:47:C9:F4:D6:EF:AC:9E:10:EB:C2:06:B5:BF:03:0B:FE:37:6A:29:F7:9A:E2
   Not Before: Jan  1 00:00:00 1970 GMT
   Not After:  Dec 31 23:59:59 9999 GMT
 1 Subject: /O=Test Org/CN=Test Org rauc CA Development
   Issuer: /O=Test Org/CN=Test Org rauc CA Development
   SPKI sha256: A0:82:5B:17:92:AB:5B:74:0E:DF:79:5C:B6:72:8B:A3:67:42:20:B1:3A:B5:DC:F8:89:3F:D5:C3:03:E9:D8:48
   Not Before: Jan  1 00:00:00 1970 GMT
   Not A rauc  install /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb: installing
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:   0% Installing
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:   0% Determining slot states
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:  20% Determining slot states done.
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:  20% Checking bundle
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:  20% Verifying signature
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:  40% Verifying signature done.
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:  40% Checking bundle done.
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:  40% Loading manifest file
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:  60% Loading manifest file done.
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:  60% Determining target install group
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:  80% Determining target install group done.
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb: 100% Installing failed.
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb: LastError: Installation error: Compatible mismatch
Installing `/run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb` failed

gitgonewithwind, 2022-09-20:
> Please properly format your questions and consider using GitHub [discussions](https://github.com/rauc/rauc/discussions) feature for new questions (and use quoting).
> 
> Also consider checking other questions first for a possible answer.
> 
> > (rauc:4092): rauc-WARNING **: GDBus.Error:org.gtk.GDBus.UnmappedGError.Quark._g_2dio_2derror_2dquark.Code30: Failed getting primary slot: Failed getting primary slot: Failed to start fw_printenv: Failed)
> 
> fw_printenv does not work, check if it is properly installed and if the config is set up correctly.
> 
> > error: Compatible mismatch
> 
> The compatible used in the system.conf differs from those in the bundle manifest.

@ejoerns 
got it.
let me check.