rauc/rauc #688: "bundle is stored on an unkown filesystem" casync bundle failure

om26er, 2021-01-21:
Our image is ZFS and it seems rauc doesn't like that. Here is the attached log. Running the same stuff on ext4 works fine. Does RAUC have any sort of filesystem allow list ? 
```
ERROR: mc:xavier-nx:xavier-nx-bundle-1.0-r0 do_bundle: Execution of '/home/cbbuilder/.yocto/cfx-hub/tmp-glibc/work/jetson_xavier_nx_devkit-oe-linux/xavier-nx-bundle/1.0-r0/temp/run.do_bundle.2399242' failed with exit code 1:
rauc-Message: 19:59:18.208: Debug log domains: 'rauc'
(rauc:2399245): rauc-DEBUG: 19:59:18.212: bundle start
(rauc:2399245): rauc-DEBUG: 19:59:18.212: system config not found, using default values
(rauc:2399245): rauc-DEBUG: 19:59:18.213: Resolved bootname /dev/mapper/matterhorn--vg-root to /dev/dm-0
(rauc:2399245): rauc-DEBUG: 19:59:18.213: input directory: /home/cbbuilder/.yocto/cfx-hub/tmp-glibc/work/jetson_xavier_nx_devkit-oe-linux/xavier-nx-bundle/1.0-r0/bundle
(rauc:2399245): rauc-DEBUG: 19:59:18.213: output bundle: /home/cbbuilder/.yocto/cfx-hub/tmp-glibc/work/jetson_xavier_nx_devkit-oe-linux/xavier-nx-bundle/1.0-r0/build/bundle.raucb
(rauc:2399245): rauc-DEBUG: 20:00:27.166: Payload size: 1656864768 bytes.
Creating bundle in 'plain' format
rauc-Message: 20:00:36.143: No keyring given, skipping signature verification
(rauc:2399245): rauc-DEBUG: 20:00:36.183: Signature offset: 1656864768 bytes.
(rauc:2399245): rauc-DEBUG: 20:00:36.183: Bundle size: 1656866260 bytes.
rauc-Message: 20:00:36.194: Debug log domains: 'rauc'
(rauc:2401001): rauc-DEBUG: 20:00:36.197: convert start
(rauc:2401001): rauc-DEBUG: 20:00:36.197: system config not found, using default values
(rauc:2401001): rauc-DEBUG: 20:00:36.197: Resolved bootname /dev/mapper/matterhorn--vg-root to /dev/dm-0
(rauc:2401001): rauc-DEBUG: 20:00:36.197: input bundle: /home/cbbuilder/.yocto/cfx-hub/tmp-glibc/work/jetson_xavier_nx_devkit-oe-linux/xavier-nx-bundle/1.0-r0/build/bundle.raucb
(rauc:2401001): rauc-DEBUG: 20:00:36.197: output bundle: /home/cbbuilder/.yocto/cfx-hub/tmp-glibc/work/jetson_xavier_nx_devkit-oe-linux/xavier-nx-bundle/1.0-r0/build/casync-bundle.raucb
rauc-Message: 20:00:36.198: Reading bundle: /home/cbbuilder/.yocto/cfx-hub/tmp-glibc/work/jetson_xavier_nx_devkit-oe-linux/xavier-nx-bundle/1.0-r0/build/bundle.raucb
rauc-Message: 20:00:36.200: Verifying bundle signature... 
(rauc:2401001): rauc-DEBUG: 20:00:36.200: initial check_bundle_access failed with: bundle is stored on an unkown filesystem (type=2fc12fc1)
bundle is stored on an unkown filesystem (type=2fc12fc1)
WARNING: exit code 1 from a shell command.
```

jluebbe, 2021-01-22:
I'm guessing you're trying to run rauc convert?

For plain bundles, RAUC since 1.5 checks if it can trust the filesystem to enforce permissions correctly:
https://github.com/rauc/rauc/blob/master/src/bundle.c#L1088
ZFS would probably belong the trusted list.

If possible, switch to the verity format. See:
https://rauc.readthedocs.io/en/latest/advanced.html#protection-against-concurrent-bundle-modification
https://rauc.readthedocs.io/en/latest/reference.html#sec-ref-formats

oberstet, 2021-01-23:
thanks for clarifying! fwiw, we patched away this stuff in our fork.

> reject unsafe filesystem types

so RAUC has an idea of "unsafe", and claims to know that? I pretty much doubt that. both. eg "NFS" is unsafe? depends: AWS EFS is fully Posix, as is Netapp, etc

anyways, in general, I am surprised that a tool like RAUC would even tell me about my filesystem if it is safe. I would welcome an option: "don't check, don't bother". user knows what he is doing. half-baked magic is a PITA ...

all of this checking, verity whatnot: ZFS has that built in. maybe I don't get what you actually mean here

https://rauc.readthedocs.io/en/latest/reference.html#verity-format

so using that does not run the broken filesystem test we run into?

om26er, 2021-01-23:
With patches rauc-native to support ZFS *and* verity bundle enabled, we now get https://github.com/rauc/rauc/blob/f81061cd5fb0c5b6bea4afeaf89db98f3375fe9d/src/bundle.c#L1154

```
| DEBUG: Executing shell function do_bundle
| rauc-Message: 13:33:51.585: Debug log domains: 'rauc'
| (rauc:304641): rauc-DEBUG: 13:33:51.588: bundle start
| (rauc:304641): rauc-DEBUG: 13:33:51.589: system config not found, using default values
| (rauc:304641): rauc-DEBUG: 13:33:51.589: Resolved bootname /dev/mapper/matterhorn--vg-root to /dev/dm-0
| (rauc:304641): rauc-DEBUG: 13:33:51.589: input directory: /home/cbbuilder/.yocto/cfx-hub/tmp-glibc/work/jetson_xavier_nx_devkit-oe-linux/xavier-nx-bundle/1.0-r0/bundle
| (rauc:304641): rauc-DEBUG: 13:33:51.589: output bundle: /home/cbbuilder/.yocto/cfx-hub/tmp-glibc/work/jetson_xavier_nx_devkit-oe-linux/xavier-nx-bundle/1.0-r0/build/bundle.raucb
| (rauc:304641): rauc-DEBUG: 13:35:00.449: Payload size: 1656549376 bytes.
| Creating bundle in 'verity' format
| (rauc:304641): rauc-DEBUG: 13:35:00.450: Hash creation SHA256, data blocks 404431.
| (rauc:304641): rauc-DEBUG: 13:35:00.450: Using 3 hash levels.
| (rauc:304641): rauc-DEBUG: 13:35:00.450: Data size: 1656549376 bytes.
| (rauc:304641): rauc-DEBUG: 13:35:00.450: Hashed size: 1669599232 bytes.
| rauc-Message: 13:35:07.541: No keyring given, skipping signature verification
| (rauc:304641): rauc-DEBUG: 13:35:07.541: Signature offset: 1669599232 bytes.
| (rauc:304641): rauc-DEBUG: 13:35:07.541: Bundle size: 1669601208 bytes.
| rauc-Message: 13:35:07.549: Debug log domains: 'rauc'
| (rauc:306283): rauc-DEBUG: 13:35:07.552: convert start
| (rauc:306283): rauc-DEBUG: 13:35:07.552: system config not found, using default values
| (rauc:306283): rauc-DEBUG: 13:35:07.553: Resolved bootname /dev/mapper/matterhorn--vg-root to /dev/dm-0
| (rauc:306283): rauc-DEBUG: 13:35:07.553: input bundle: /home/cbbuilder/.yocto/cfx-hub/tmp-glibc/work/jetson_xavier_nx_devkit-oe-linux/xavier-nx-bundle/1.0-r0/build/bundle.raucb
| (rauc:306283): rauc-DEBUG: 13:35:07.553: output bundle: /home/cbbuilder/.yocto/cfx-hub/tmp-glibc/work/jetson_xavier_nx_devkit-oe-linux/xavier-nx-bundle/1.0-r0/build/casync-bundle.raucb
| rauc-Message: 13:35:07.553: Reading bundle: /home/cbbuilder/.yocto/cfx-hub/tmp-glibc/work/jetson_xavier_nx_devkit-oe-linux/xavier-nx-bundle/1.0-r0/build/bundle.raucb
| rauc-Message: 13:35:07.555: Verifying bundle signature...
| rauc-Message: 13:35:07.558: Verified inline signature by 'O = Crossbar.io GmbH, CN = Crossbar.io GmbH Development-1'
| (rauc:306283): rauc-DEBUG: 13:35:07.558: Got 2 elements for trust chain
| rauc-Message: 13:35:07.559: Verifying bundle payload...
| Failed to create bundle: cannot check bundle payload without exclusive access: unable to find mounted device for bundle
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/home/cbbuilder/.yocto/cfx-hub/tmp-glibc/work/jetson_xavier_nx_devkit-oe-linux/xavier-nx-bundle/1.0-r0/temp/run.do_bundle.304638' failed with exit code 1:
| rauc-Message: 13:33:51.585: Debug log domains: 'rauc'
| (rauc:304641): rauc-DEBUG: 13:33:51.588: bundle start
| (rauc:304641): rauc-DEBUG: 13:33:51.589: system config not found, using default values
| (rauc:304641): rauc-DEBUG: 13:33:51.589: Resolved bootname /dev/mapper/matterhorn--vg-root to /dev/dm-0
| (rauc:304641): rauc-DEBUG: 13:33:51.589: input directory: /home/cbbuilder/.yocto/cfx-hub/tmp-glibc/work/jetson_xavier_nx_devkit-oe-linux/xavier-nx-bundle/1.0-r0/bundle
| (rauc:304641): rauc-DEBUG: 13:33:51.589: output bundle: /home/cbbuilder/.yocto/cfx-hub/tmp-glibc/work/jetson_xavier_nx_devkit-oe-linux/xavier-nx-bundle/1.0-r0/build/bundle.raucb
| (rauc:304641): rauc-DEBUG: 13:35:00.449: Payload size: 1656549376 bytes.
| Creating bundle in 'verity' format
| (rauc:304641): rauc-DEBUG: 13:35:00.450: Hash creation SHA256, data blocks 404431.
| (rauc:304641): rauc-DEBUG: 13:35:00.450: Using 3 hash levels.
| (rauc:304641): rauc-DEBUG: 13:35:00.450: Data size: 1656549376 bytes.
| (rauc:304641): rauc-DEBUG: 13:35:00.450: Hashed size: 1669599232 bytes.
| rauc-Message: 13:35:07.541: No keyring given, skipping signature verification
| (rauc:304641): rauc-DEBUG: 13:35:07.541: Signature offset: 1669599232 bytes.
| (rauc:304641): rauc-DEBUG: 13:35:07.541: Bundle size: 1669601208 bytes.
| rauc-Message: 13:35:07.549: Debug log domains: 'rauc'
| (rauc:306283): rauc-DEBUG: 13:35:07.552: convert start
| (rauc:306283): rauc-DEBUG: 13:35:07.552: system config not found, using default values
| (rauc:306283): rauc-DEBUG: 13:35:07.553: Resolved bootname /dev/mapper/matterhorn--vg-root to /dev/dm-0
| (rauc:306283): rauc-DEBUG: 13:35:07.553: input bundle: /home/cbbuilder/.yocto/cfx-hub/tmp-glibc/work/jetson_xavier_nx_devkit-oe-linux/xavier-nx-bundle/1.0-r0/build/bundle.raucb
| (rauc:306283): rauc-DEBUG: 13:35:07.553: output bundle: /home/cbbuilder/.yocto/cfx-hub/tmp-glibc/work/jetson_xavier_nx_devkit-oe-linux/xavier-nx-bundle/1.0-r0/build/casync-bundle.raucb
| rauc-Message: 13:35:07.553: Reading bundle: /home/cbbuilder/.yocto/cfx-hub/tmp-glibc/work/jetson_xavier_nx_devkit-oe-linux/xavier-nx-bundle/1.0-r0/build/bundle.raucb
| rauc-Message: 13:35:07.555: Verifying bundle signature...
| rauc-Message: 13:35:07.558: Verified inline signature by 'O = Crossbar.io GmbH, CN = Crossbar.io GmbH Development-1'
| (rauc:306283): rauc-DEBUG: 13:35:07.558: Got 2 elements for trust chain
| rauc-Message: 13:35:07.559: Verifying bundle payload...
| Failed to create bundle: cannot check bundle payload without exclusive access: unable to find mounted device for bundle
| WARNING: exit code 1 from a shell command.
```

jluebbe, 2021-01-25:
> > I'm guessing you're trying to run rauc convert?

Could you confirm that you get this during rauc convert (on the build host)? It looks that way from the logs, but I'd like to avoid guessing. I'll continue under that assumption.

> thanks for clarifying! fwiw, we patched away this stuff in our fork.

I hope you only removed it for your host-side build. Otherwise you've removed the mitigation of https://github.com/rauc/rauc/security/advisories/GHSA-cgf3-h62j-w9vv (for the old "plain format" bundles).

> > reject unsafe filesystem types
> 
> so RAUC has an idea of "unsafe", and claims to know that? I pretty much doubt that. both. eg "NFS" is unsafe? depends: AWS EFS is fully Posix, as is Netapp, etc

This is part of the TOCTOU mitigation for bundle installation on the target. We have a list of filesystems where we expect the kernel to protect against concurrent modification during installation. We also have a list where it is easy for an attacker to modify the bundle data during the TOCTOU window (FUSE and NFS). See the advisory for details.

As an out-of-tree filesystem, we missed ZFS and I agree that it should be in the local filesytem list (so considered safe).

> anyways, in general, I am surprised that a tool like RAUC would even tell me about my filesystem if it is safe. I would welcome an option: "don't check, don't bother". user knows what he is doing. half-baked magic is a PITA ... 

I agree, we should implement `--no-verify` for `rauc convert` (as for `info` and `resign`).

This is erring on the side of caution. Currently all signature checks use the same code path, which includes checking against possible concurrent modification when we can use dm-verity.

Perhaps doing this for `rauc convert` is too strict, but I wouldn't be surprised if someone is relying on the signature checks there.

> all of this checking, verity whatnot: ZFS has that built in. maybe I don't get what you actually mean here

As above, I agree that ZFS should be in the safe list.

> https://rauc.readthedocs.io/en/latest/reference.html#verity-format
> 
> so using that does not run the broken filesystem test we run into?

On the target, for 'rauc install', yes. It removes the need for the exclusive access check because we are able to use dm-verity to protect against modification.

As @om26er posted, during `rauc convert`, you trigger an follow-up check, where RAUC tries to check for non-root write permission on the block device. In your case, it is unable to find the block device using the mount table information. As I understand it, the data-checksum in ZFS is not intended to protect against malicious modification (at least by default), so I'm not convinced that it is sufficient. Please correct me if that's wrong.

Could you mail me the output of (mount and /proc/mount) so I could check if there's a good way to handle ZFS automatically?

If, for the host-side commands, it turns out that a reliable enough check is not possible, I'd consider reducing the error to a warning for rauc resign&convert when running as a normal user.

zonque, 2021-01-25:
#689 should fix that issue

om26er, 2021-01-25:
@zonque actually, I made a similar change a few days ago *and* with that change we are still hitting https://github.com/rauc/rauc/issues/688#issuecomment-766066546 -- Did you actually test that on ZFS ?

oberstet, 2021-01-25:
>  As I understand it, the data-checksum in ZFS is not intended to protect against malicious modification (at least by default), so I'm not convinced that it is sufficient. Please correct me if that's wrong.

I'm not entirely sure. ZFS has a Merkle tree of hashes over all blocks in a ZFS pool. if an attacker fiddles with bits at the raw block device level (circumventing ZFS hash tree updates normally occurring), then the bits in the block, and the hashes won't align. This will remain unnoticed until one of the following happens:

* the block is read, upon which ZFS will recompute the hashes of the block read plus all hashes up to root
* a ZFS scrubbing of the pool is initiated (which reads all blocks and resilvers)

Further, I don't think the hashing is authenticated (not a MAC) .. if that matters

om26er, 2021-01-25:
> Could you confirm that you get this during rauc convert (on the build host)? It looks that way from the logs, but I'd like to avoid guessing. I'll continue under that assumption.

This happens during the build, I am not sure at what step that happens because we don't run `rauc convert` manually, we inherit the `bundle` class from meta-rauc and export `RAUC_CASYNC_BUNDLE = "1"` in our bundle config.

> I hope you only removed it for your host-side build. Otherwise you've removed the mitigation of [GHSA-cgf3-h62j-w9vv](https://github.com/rauc/rauc/security/advisories/GHSA-cgf3-h62j-w9vv) (for the old "plain format" bundles).

Yeah, we only patched that for the host side tool (rauc-native recipe)

zonque, 2021-01-25:
> Did you actually test that on ZFS ?

Yes. But I'm not using casync but plain bundles in files, so YMMV.

om26er, 2021-01-25:
> Yes. But I'm not using casync but plain bundles in files, so YMMV.

right, we were not hitting that issue when using plain bundles on ZFS anyways ;-)

oberstet, 2021-01-25:
aah, ok, I guess we'll need to continue to use our fork for now then, as it also contains the change to make casync work.

----

rgd "Time-of-Check-Time-of-Use Vulnerabilities" (https://github.com/rauc/rauc/security/advisories/GHSA-cgf3-h62j-w9vv)

the steps to update are:

1. determine chunks to fetch
1. download chunks
1. verify chunks
1. if good, write chunks to partition

so an attacker must not be able to modify in between steps 3. and 4. ?

if so, ZFS on the device would allow an easy way out: create a read-only snapshot before step 3. and do the verify and update from the read-only snapshot ...

ejoerns, 2021-01-26:
> This happens during the build, I am not sure at what step that happens because we don't run `rauc convert` manually, we inherit the `bundle` class from meta-rauc and export `RAUC_CASYNC_BUNDLE = [...]

Setting `RAUC_CASYNC_BUNDLE = "1"` will let the bundle class call `casync convert`. The normal way to create a casync bundle is:
* creating a 'normal' bundle first
* calling `rauc convert` to convert this into a casync-Bundle + chunks store.

This is a separate command/step as the chunk store generation will often happen on the deployment server-side and not on the build machine. Generating the chunk store on the server-side allows for example to set a common location for the chunk store of all bundles and thus to deduplicate chunks on the server-side, too.
And, keeping the normal bundle, will potentially allow us to use it as a (fallback) alternative for network-less updates or as the original source for conversion into other formats.

> I agree, we should implement --no-verify for rauc convert (as for info and resign).

+1 as this is quite similar to resign actually and I hope we can trust the user probably knows what he's doing ;)
In general, the stricter protection focus should be on the target side.

> Could you mail me the output of (mount and /proc/mount) so I could check if there's a good way to handle ZFS automatically?

I guess this question remained open. Being able to determine the device correctly would at least help us for target use cases.

jluebbe, 2021-01-26:
> aah, ok, I guess we'll need to continue to use our fork for now then, as it also contains the change to make casync work.
> 
> rgd "Time-of-Check-Time-of-Use Vulnerabilities" ([GHSA-cgf3-h62j-w9vv](https://github.com/rauc/rauc/security/advisories/GHSA-cgf3-h62j-w9vv))
> 
> the steps to update are:
> 
> 1. determine chunks to fetch
> 2. download chunks
> 3. verify chunks
> 4. if good, write chunks to partition
> 
> so an attacker must not be able to modify in between steps 3. and 4. ?

You missed the steps before "1. determine chunks to fetch" which are done by RAUC and not casync in the case of a plain bundle:
a) hash squashfs and check signature
b) mount squashfs
c) read manifest and possibly execute hooks

casync is only involved when RAUC sees a slot which refers to a .caidx/.caibx.

If the attacker can modify the underlying filesystem or the bundle file directly (for example due to too open file/device permissions), the modified squashfs will be mounted, leading almost directly do code execution.

> if so, ZFS on the device would allow an easy way out: create a read-only snapshot before step 3. and do the verify and update from the read-only snapshot ...

If you use ZFS with cryptographic hashes, that would be a possible approach. As would using fsverity. But as these are not widely available, they are not usable as a mitigation for existing users of RAUC.
So we have the exclusive access checks for the plain bundle as a mitigation and the verity bundle format, which fixes the underlying issue without depending on a specific storage (it works on any FS, NAND and even over network mounts).

jluebbe, 2022-01-26:
Closed via #770.

closed 2022-01-26 by jluebbe