rauc/rauc #690: Are there any error codes?

krutztq, 2021-01-25:
Hi rauc Team,

I have  a question:
Are there any error codes or did I missed something? If not, are there any error codes planned?

I want read the latest error with my application and localize it for the GUI. My app has it's own localization workflow therefore a gettext solution would not fit my concept.

'rauc install' with an error always exits with error code 1, regardless of the error.
I found the dbus LastError property, but the comparison of strings is not robust enough for me.

I can handle the errors from the hook script well, but I have to handle them separately.
Here, too, it would be nice if you could get the exit code of the hook script from rauc itself.

Or is there another solution?

Thanks in advance

jluebbe, 2021-01-25:
Sorry, but there are currently no external error codes. My preferred approach would be to define error codes on the DBus level. @ejoerns took a quick look at that some time ago. I think that would allow us to make the error enums visible to DBus clients. Enrico, is that correct?

For the exit code of the hook script, I think we could add an optional details dictionary to a replacement of the LastError property.

krutztq, 2021-01-26:
Thank you for the quick reply. DBus level error codes would be great.
An additional LastErrorCode property (backwards compatible) with a small area for hook exit codes would be fine with me as well.

jluebbe, 2021-01-26:
As this feature is not on our roadmap so far, I can only offer to review any PRs which work towards this.

ejoerns, 2021-01-26:
@jluebbe To the best of my knowledge, there is no concept of error codes in dbus. What we have is a message type 'error' and an `ERROR_NAME` field [1].
What I had a look into is using explicit errors instead of automatically mapped ones. By default, a Glib error (GError) passed over D-Bus will result in the error MESSAGE being set to the error message of the GError, but the ERROR_NAME will be something like `org.gtk.GDBus.UnmappedGError.Quark._g_2dio_2derror_2dquark.Code30` which is a simple serialized form of the original glib error, but hard to parse for non-glib applications. Thus better use explicit error names, matching the interface naming convention, like: `de.pengutronix.rauc.Error.InavlidSiganture`.

In general, I really appreciate the idea to use error-specific exit codes. But, this would pretty sure be independent of the codes we currently use for error reporting as these have the concept of a (domain, code) tuple that we would have to map to a single code being returned.

This is probably also independent from the Glib mappings as these are for errors passed over dbus only and allow reflecting the code / domain tuble by using `..Error.<Code>.<Name> strings.`

[1] https://dbus.freedesktop.org/doc/dbus-specification.html

krutztq, 2021-01-26:
Using explicit error names like `de.pengutronix.rauc.Error.InvalidSignature` would also be fine, then you don't need an error code. 
For now I decided to parse the LastError property in my application, so that there is not too much logic in the hook script.

ChromaMaster, 2023-08-04:
Hello! I just wanted to ask if there are any plans for this for the near future. Thanks!

ejoerns, 2023-08-15:
@ChromaMaster Unfortunately no plans that I would know of. But I can only speak about 'our' projects or things that I hear from the community.