rauc/rauc #692: Installer fails if readonly slot is found

nickleverton, 2021-01-26:
When searching for an inactive slot to install into, the installer stops considering further slots when a readonly one is found, even if there are other inactive slots which could be used.

It would be preferable to ignore readonly slots and continue the search.

This is an example of a system.conf file where I came across the issue, using Rauc to implement a factory installer from SD card.  The installer examimed the readonly SDCard slots before the Factory ones and refused to continue.

```
[slot.uboot.0]
device=/dev/mmcblk0p1
parent=rootfs.0

[slot.rootfs.0]
device=/dev/mmcblk0p2
type=ext4
bootname=Factory

[slot.uboot.1]
device=/dev/mmcblk1p1
readonly=true
parent=rootfs.1

[slot.rootfs.1]
device=/dev/mmcblk1p2
type=ext4
readonly=true
bootname=SDCard
```

nickleverton, 2021-01-26:
Fix pasted inline (since I can't see how to attach patches)

```
diff --git a/src/install.c b/src/install.c
index 008de96..5ba9c13 100644
--- a/src/install.c
+++ b/src/install.c
@@ -259,6 +259,9 @@ static RaucSlot *select_inactive_slot_class_member(const gchar *rootclass)
                if (iterslot->state != ST_INACTIVE)
                        continue;

+               if (iterslot->readonly)
+                       continue;
+
                if (g_strcmp0(iterslot->sclass, rootclass) == 0) {
                        return iterslot;
                }
```

ejoerns, 2021-01-26:
@nickleverton a setup where one rootfs slot is read-only and the other is not violates the very base assumption of A/B updates that A and B are equal.

The semantical meaning of having two slots of class `rootfs` with instance numbers `0` and `1` is exactly that it is irrelevant if you update 0 or if you update 1.

A factory setup would instead require a setup like:

``` ini
[factory.0]
readonly=true
...

[rootfs.0]
...
```

Or, if you do not intend to update factory, anyway: Consider using [rauc.external](https://rauc.readthedocs.io/en/latest/integration.html#identification-via-kernel-commandline) in the kernel command line to indicate you explicitly run from a slot that is outside the A/B rootfs set.

Would that match your use-case?

nickleverton, 2021-01-26:
Hi,

Thanks for the quick answer but I think it's a bit of a red herring looking at this in an A/B or Factory/other context.  As far as I can see, the problem is that as soon as the installer comes across a readonly slot, it will abort without examining other slots, regardless of what image setup one has.  

One might say that this can be worked around by ordering the slots differently in the system.conf file, but I haven't seen that constraint mentioned in the docs (apologies if I missed it).  I think it would be better not to surprise rauc's users by having the slot ordering be so critical.

To reproduce:

Below is a Factory/A/B setup where the installer fails due to presence of a readonly slot.  This is the state that arises when I am originally booted from Factory slot, then install a bundle which goes into cluster A.  I then reboot into A and the following is the state of rauc and its variables at this point. 

In this state, I am unable to install a further update.  It ought to go into B but rauc fails with the message 
"Installation error: Target slot for class uboot of image imx-boot.img is readonly".  On tracing the code, this turns out to be because it comes across the Factory slot whilst building its table of target slots in <code>select_inactive_slot_class_member()</code>, and immediately returns NULL without checking for further potentially valid slots.

```
[slot.uboot.0]
device=/dev/mmcblk0p1
parent=rootfs.0
readonly=true

[slot.rootfs.0]
device=/dev/mmcblk0p2
type=ext4
bootname=Factory
readonly=true

[slot.uboot.1]
device=/dev/mmcblk0p4
parent=rootfs.1

[slot.rootfs.1]
device=/dev/mmcblk0p6
type=ext4
bootname=A

[slot.uboot.2]
device=/dev/mmcblk0p5
parent=rootfs.2

[slot.rootfs.2]
device=/dev/mmcblk0p7
type=ext4
bootname=B
```

```
BOOT_A_LEFT=2
BOOT_B_LEFT=3
BOOT_ORDER=A Factory B
```

```
# cat /proc/cmdline
console=ttymxc2,115200 root=/dev/mmcblk0p6 rootwait rw rauc.slot=A
```

```
# rauc status
=== System Info ===
Compatible: 
Variant:
Booted from: rootfs.1 (A)

=== Bootloader ===
Activated: rootfs.1 (A)

=== Slot States ===
x [rootfs.1] (/dev/mmcblk0p6, ext4, booted)
        bootname: A
        mounted: /
        boot status: good
    [uboot.1] (/dev/mmcblk0p4, raw, active)

o [rootfs.0] (/dev/mmcblk0p2, ext4, inactive)
        bootname: Factory
        boot status: good
    [uboot.0] (/dev/mmcblk0p1, raw, inactive)

o [rootfs.2] (/dev/mmcblk0p7, ext4, inactive)
        bootname: B
        boot status: good
    [uboot.2] (/dev/mmcblk0p5, raw, inactive)
```

```
# rauc -d install full-bundle.raucb
rauc-Message: 04:32:53.692: Debug log domains: 'rauc'
(rauc:810): rauc-DEBUG: 04:32:53.701: install started
(rauc:810): rauc-DEBUG: 04:32:53.701: input bundle: /home/root/full-bundle.raucb
(rauc:810): rauc-DEBUG: 04:32:53.714: Trying to contact rauc service
installing
  0% Installing
  0% Determining slot states
 20% Determining slot states done.
 20% Checking bundle
 20% Verifying signature
 40% Verifying signature done.
 40% Checking bundle done.
 40% Loading manifest file
 60% Loading manifest file done.
 60% Determining target install group
 80% Determining target install group done.
100% Installing failed.
LastError: Installation error: Target slot for class uboot of image imx-boot.img is readonly
Installing `/home/root/full-bundle.raucb` failed

```

ejoerns, 2021-01-26:
> Thanks for the quick answer but I think it's a bit of a red herring looking at this in an A/B or Factory/other context. As far as I can see, the problem is that as soon as the installer comes across [...]

The thing is that I would argue it is an invalid configuration having slots of the same class where some have the 'readonly' property set and some not.

From the perspective you have, I would agree that your fix is valid, and it would probably not be wrong when adding my assumption. But the readonly key was never really meant to be used like you use it. And I cannot yet guess a scenario that would benefit from having a single readonly slot in a slot class compared to having a different slot class instead. Would you see any disadvantage from using the approach I noted above? (which is basically the one form the docs: https://rauc.readthedocs.io/en/latest/examples.html?highlight=readonly#asymmetric-a-recovery-setup)

What I would propose instead is a consistency check that ensures all readonly properties of slots of the same class are equal.

I agree with you that the readonly property is not documented very well, and possibly lacks some description on what to use it nor for. And the behavior should be better defined or better checked.

ensc, 2023-12-01:
I am trying to implement a factory/recovery setup in parallel to normal A/B and face the same problem:

```
## factory slot
[slot.factory.0]
device=/dev/disk/by-path/platform-2190000.mmc-part1
type=raw
readonly=true

## system0 slots
[slot.rootfs.0]
device=/dev/disk/by-path/platform-2190000.mmc-part5
type=ext4
bootname=system0

## system1 slots
[slot.rootfs.1]
device=/dev/disk/by-path/platform-2190000.mmc-part6
type=ext4
bootname=system1
```

Running this with a bundle containing a `factory` class image results in

```
LastError: Installation error: Target slot for class factory of image fitimage-...factory.fit.img is readonly
```

My idea was that the bundle contains both the normal system and the factory/recovery image and the factory image is only written when forced to.  Are there (hidden) flags for 'rauc install' so that readonly images are written?  Or what else is the idea behind the `readonly` flag?

rauc-1.10.1

jluebbe, 2023-12-01:
The `readonly` flag lets rauc know about the slot, but it will never write to it. This is also the trigger for the Installation error you see.
As a installing a bundle is intended to result in a consistent system state, it doesn't really make sense to only install some of the images.