rauc/rauc #693: Signing bundle fails (buildroot rauc)

gjorgjipetkovski, 2021-01-27:
When using rauc, from buildroot produced SDK, to sign the bundle using USB token there is an error:

osboxes@ub-18-04:~/bundle/test-token$ /home/osboxes/source/SDK/arm-buildroot-linux-gnueabihf_sdk-buildroot/bin/rauc --debug --cert cert.pem --key 'pkcs11:token=Proj;object=rauc' bundle ../140/bundle update-7.2.raucb
Domains: 'rauc'
(rauc:28205): rauc-DEBUG: 10:25:49.865: bundle start
(rauc:28205): rauc-DEBUG: 10:25:49.865: system config not found, using default values
(rauc:28205): rauc-DEBUG: 10:25:49.865: Resolved bootname /dev/disk/by-uuid/63146394-e834-4c15-8549-523cabc44f58 to /dev/sda1
(rauc:28205): rauc-DEBUG: 10:25:49.865: input directory: ../140/bundle
(rauc:28205): rauc-DEBUG: 10:25:49.865: output bundle: update-7.2.raucb
139715416996592:error:02001002:system library:fopen:No such file or directory:bss_file.c:175:fopen('pkcs11:token=Proj;object=rauc','r')
139715416996592:error:2006D080:BIO routines:BIO_new_file:no such file:bss_file.c:182:
Failed to create bundle: failed signing bundle: failed to load key file 'pkcs11:token=Proj;object=rauc'

rauc version is quite old - 0.4

However, if local build of rauc (version 1.5.1.10-a145) is used, the signing using USB token goes well:

(rauc:28340): rauc-DEBUG: 10:43:25.486: Payload size: 47050752 bytes.
Creating bundle in 'plain' format
Enter PKCS#11 token PIN for Vauron:
rauc-Message: 10:45:30.734: No keyring given, skipping signature verification
(rauc:28340): rauc-DEBUG: 10:45:30.737: Signature offset: 47050752 bytes.
(rauc:28340): rauc-DEBUG: 10:45:30.737: Bundle size: 47052120 bytes.

Any hint would be appreciated.

jluebbe, 2021-01-27:
Support for PKCS#11 tokens was added in RAUC 1.0-rc1. As buildroot already contains a newer version, I'd suggest updating your buildroot.

gjorgjipetkovski, 2021-01-28:
Thanks. It works with updated buildroot (rauc 1.0-rc1)

closed 2021-01-28 by gjorgjipetkovski

ejoerns, 2021-01-28:
@gjorgjipetkovski Any reason to not use recent Buildroot releases? Or a recent rauc? Older software normally is not better compared to more recent ones and lacks important bugfixes, security fixes, new features etc. Additionally, rc1 is not even a RAUC release, it is only a 'release candidate' as the name implies.

gjorgjipetkovski, 2021-01-28:
@ejoerns Thanks for bringing it up.
Yes, I understand all of that. Actually it is an older project (hence older Buildroot) and this time we needed to do only a change in signing the bundle with pkcs#11 token. Updating the Buildroot is a future option if the client agrees the effort etc.
But anyhow, I agree that recent RAUC version is better option that older one, especially rc. Therefore, we will probably use locally built recent RAUC version for signing the bundle.
Do you see any potential issues if host RAUC version (or the one used for signing the bundle) differs from target RAUC version? (the target RAUC version is already 1.0-rc1)

ejoerns, 2021-01-29:
@gjorgjipetkovski newer versions of RAUC (on the host) provide some manifest conf options that older versions (on the target) are unable to parse. But if you do not explicitly use them, then the generated bundles will be compatible.
Despite we take care of not breaking compatibility whenever this is possible, it is always a good idea to test updates/migrations carefully before deploying them into the field.
Also note that versions before v1.5 are affected by [CVE-2020-25860](https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2020-25860) which is mainly relevant to the target (see [advisory](https://github.com/rauc/rauc/security/advisories/GHSA-cgf3-h62j-w9vv)).

gjorgjipetkovski, 2021-01-29:
Clear. Thanks @ejoerns .