rauc/rauc #696: Installing bundle: Failed splicing data when installing bundle

JW94, 2021-02-08:
Hello,

first of all thank you for your great work! I've been testing rauc for some days and it almost works flawlessly. Unfortunately when installing a bundle on my embedded system it fails 😟: 
```
root@pool-gateway:~# rauc -c /etc/rauc/system.conf info update-bundle-pool-gateway-20210208132423.raucb -d
(rauc:898): rauc-DEBUG: 15:46:31.607: install started
(rauc:898): rauc-DEBUG: 15:46:31.607: input bundle: /home/root/update-bundle-pool-gateway-20210208132423.raucb
(rauc:898): rauc-DEBUG: 15:46:31.752: Trying to contact rauc service
installing
  0% Installing
  0% Determining slot states
 20% Determining slot states done.
 20% Checking bundle
 20% Verifying signature
 40% Verifying signature done.
 40% Checking bundle done.
 40% Checking man[ 3340.341726] device-mapper: verity: sha256 using implementation "stm32-sha256"
ifest contents
 60% Checking manifest contents done.
 60% Determining target install group
 80% Determining target install group done.
 80% Updating slots
 80% Checking slot rootfs.1
 90% Checking slot rootfs.1 done.
 90% Copying image to rootfs.1
[ 3340.504328] device-mapper: verity: 7:0: data block 8 is corrupted
[ 3340.504338] device-mapper: verity: 7:0: data block 9 is corrupted
[ 3340.515370] SQUASHFS error: squashfs_read_data failed to read block 0x773a
[ 3340.521962] SQUASHFS error: Unable to read data cache entry [773a]
[ 3340.528800] SQUASHFS error: Unable to read page, block 773a, size 1cf8
[ 3340.534826] SQUASHFS error: Unable to read data cache entry [773a]
[ 3340.540844] SQUASHFS error: Unable to read page, block 773a, size 1cf8
[ 3340.547646] SQUASHFS error: Unable to read data cache entry [773a]
[ 3340.553544] SQUASHFS error: Unable to read page, block 773a, size 1cf8
[ 3340.560177] SQUASHFS error: Unable to read data cache entry [773a]
[ 3340.566408] SQUASHFS error: Unable to read page, block 773a, size 1cf8
[ 3340.572804] SQUASHFS error: Unable to read data cache entry [773a]
[ 3340.579044] SQUASHFS error: Unable to read page, block 773a, size 1cf8
100% Copying image to rootfs.1 failed.
100% Updating slots failed.
100% Installing failed.
LastError: Installation error: Failed updating slot rootfs.1: Failed splicing data: Error reading from file: Input/output error
Installing `/home/root/update-bundle-pool-gateway-20210208132423.raucb` failed
```

You can see these kernelmessages because I communicate over UART.

#### Config
```
[system]
compatible=stm32mp1
bootloader=uboot
mountprefix=/mnt
max-bundle-download-size=805306368
statusfile=/usr/local/raucstatus

[keyring]
path=/etc/rauc/ca.cert.pem

[slot.rootfs.0]
device=/dev/mmcblk2p4
type=ext4
bootname=A

[slot.rootfs.1]
device=/dev/mmcblk2p5
type=ext4
bootname=B
```
#### lsblk
```
NAME         MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk2      179:0    0  3.7G  0 disk
|-mmcblk2p1  179:1    0    2M  0 part
|-mmcblk2p2  179:2    0   64M  0 part
|-mmcblk2p3  179:3    0   64M  0 part
|-mmcblk2p4  179:4    0  512M  0 part /
|-mmcblk2p5  179:5    0  512M  0 part
`-mmcblk2p6  179:6    0  2.6G  0 part
mmcblk2boot0 179:16   0    2M  1 disk
mmcblk2boot1 179:32   0    2M  1 disk
 ``` 
#### Rauc info
```
(rauc:920): rauc-DEBUG: 16:02:01.849: Got 2 elements for trust chain
Compatible:     'stm32mp1'
Version:        '1.0'
Description:    'update-bundle version 1.0-r0'
Build:          '20210208132423'
Hooks:          ''
Bundle Format:  verity
  Verity Salt:  'b367aea7c188d5ba5c4557183799c3ef7e6d2996df8140ad5d05fb47ffafa668'
  Verity Hash:  '06c05ee3133a4264df8b998a1b790820039c83e23f392dbe1ed12590820c0505'
  Verity Size:  430080
1 Image:
(1)     hekate-core-pool-gateway.ext4
        Slotclass: rootfs
        Checksum:  82a6959e1726f3761f8734fd22d6e6f496a1f9c1d0b2f813651389d113bc4593
        Size:      256876544
        Hooks:
0 Files

Certificate Chain:
 0 Subject: O = Test Org, CN = Test Org Development-1
   Issuer: O = Test Org, CN = Test Org rauc CA Development
   SPKI sha256: 49:E1:BA:DD:4D:DE:49:02:2C:B5:26:A1:4E:56:BA:2D:DA:D4:16:E7:F7:3A:BB:F6:73:C2:B3:C6:08:E7:05:08
   Not Before: Jan  1 00:00:00 1970 GMT
   Not After:  Dec 31 23:59:59 9999 GMT
 1 Subject: O = Test Org, CN = Test Org rauc CA Development
   Issuer: O = Test Org, CN = Test Org rauc CA Development
   SPKI sha256: A3:29:62:CA:F3:6B:6D:6E:0A:61:CC:F3:6F:28:4C:78:B8:E6:84:5C:3A:C9:C8:4D:D2:A6:6B:B8:C5:8A:D4:7A
   Not Before: Jan  1 00:00:00 1970 GMT
   Not After:  Dec 31 23:59:59 9999 GMT
```

#### Rauc version
```
root@pool-gateway:~# rauc --version
rauc 1.5.1
```
I can't fix this error and don't know where the problem can be. Any help is appreciated!

ejoerns, 2021-02-09:
> root@pool-gateway:~# rauc -c /etc/rauc/system.conf info update-bundle-pool-gateway-20210208132423.raucb -d
> (rauc:898): rauc-DEBUG: 15:46:31.607: install started

Is that the unmodified console output? I would bet it does not install when calling 'info'.

> You can see these kernelmessages because I communicate over UART.

Yes, they are quite useful :)

Do you use dm-verity somewhere else in your system? Just to ensure that the verity error messages we see originate from the bundle and not from something else.

It looks a bit like RAUC overrides the bundle during installation. How do your mounts look like? Does /home/root originate from the rootfs device?

JW94, 2021-02-10:
Thanks for your fast response! 🙂

> Is that the unmodified console output? I would bet it does not install when calling 'info'.

Sorry my mistake, I copied the wrong line. I wanted to add the command afterwards and I took the wrong line in the terminal. This is the right one:
```
rauc -c /etc/rauc/system.conf install update-bundle-pool-gateway-20210208132423.raucb  -d
```

> Do you use dm-verity somewhere else in your system? Just to ensure that the verity error messages we see originate from the bundle and not from something else.

Yes it is an empty system which is only used for testing rauc 😃 I even run `killall rauc` before running rauc to ensure no other process is interfering.

> It looks a bit like RAUC overrides the bundle during installation. How do your mounts look like? Does /home/root originate from the rootfs device?

`/home/root` is on `rootfs.0` aka `SLOT A`. I tried to copy the bundle on partition `p6` which is a persistent partition for data of the application. But the error is the same. I'll attach the bundle and all my configs. Meanwhile I'll generate another bundle, maybe something went wrong while generating the first bundle in yocto.

[Rauc-Bundle](https://drive.google.com/file/d/14nJ17QV9s-v5scPdr5CxYYdGIRong6FC/view?usp=sharing)

JW94, 2021-02-11:
Unfortunately the other bundle didn't work aswell. It prints the same failure as before.

Then I tried to generate a bundle in `plain` bundle format and I can successfully update the rootfs partition! 🥳 
```
root@pool-gateway:~/test# rauc install -c /etc/rauc/system.conf hekate-core-bundle-pool-gateway-20210211125437.raucb -d
rauc-Message: 14:14:02.775: Debug log domains: 'rauc'
(rauc:1182): rauc-DEBUG: 14:14:02.789: install started
(rauc:1182): rauc-DEBUG: 14:14:02.789: input bundle: /home/root/test/hekate-core-bundle-pool-gateway-20210211125437.raucb
(rauc:1182): rauc-DEBUG: 14:14:02.940: Trying to contact rauc service
installing
  0% Installing
  0% Determining slot states
 20% Determining slot states done.
 20% Checking bundle
 20% Verifying signature
 40% Verifying signature done.
 40% Checking bundle done.
 40% Checking manifest contents
 60% Checking manifest contents done.
 60% Determining target install group
 80% Determining target install group done.
 80% Updating slots
 80% Checking slot rootfs.1
 90% Checking slot rootfs.1 done.
 90% Copying image to rootfs.1
100% Copying image to rootfs.1 done.
100% Updating slots done.
100% Installing done.
Installing `/home/root/test/hekate-core-bundle-pool-gateway-20210211125437.raucb` succeeded
```

In addition, I used the `tar.xz`  image type format and this works too. But when I switch the bundle to `verity` it doesn't work anymore. 

When I run `rauc install ...` the first time I get this message from the UART-Term: 
`[11187.392611] verity_handle_err: 8 callbacks suppressed`

I guess I will stay with `plain` as long as I can't solve the problem 😢

ejoerns, 2021-10-21:
I've seen similar issues on an i.mx6 device now.

Kernel is 5.10.73-rt54.

```
[16967.686213] device-mapper: verity: 7:0: data block 11515 is corrupted
[16967.686387] SQUASHFS error: Failed to read block 0x2cfb891: -5
[16967.686399] SQUASHFS error: Unable to read data cache entry [2cfb891]
[16967.686406] SQUASHFS error: Unable to read page, block 2cfb891, size 6ed4
[16967.686448] SQUASHFS error: Unable to read data cache entry [2cfb891]
[16967.686455] SQUASHFS error: Unable to read page, block 2cfb891, size 6ed4
[16967.686491] SQUASHFS error: Unable to read data cache entry [2cfb891]
```

As long as I do not have any other information I would assume this describes the same issue as noted above and set this issue to 'bug'.

With RAUC master I can reproduce this by calling `rauc mount` on the bundle and copying a file system image in it to any storage location.

@JW94 Do you still know the kernel version you used when you hit this issue?

JW94, 2021-10-21:
Ok, I will give it a try! Thank you!

deval-avive, 2023-01-12:
What was the resolution for this external bug? I am seeing this exact behavior on some (not all) devices that are running `rauc install` on encrypted rauc bundles. I am using a Microchip SAMA5.

ejoerns, 2023-01-12:
> ```
> [ 2981.070000] SQUASHFS error: zlib decompression failed, data probably corrupt
> [ 2981.070000] SQUASHFS error: squashfs_read_data failed to read block 0x7f9cfb
> [ 2981.070000] SQUASHFS error: Unable to read data cache entry [7f9cfb]
> ```

I don't see a verity device-mapper error here. Is this actually a verity bundle that's used here?

With verity, you should not detect corruption at the zlib level but should not be able to read any data instead.

deval-avive, 2023-01-13:
Appreciate the quick response!

Yes I see what you are saying. Are you suggesting that this is a filesystem corruption issue (where we are trying to install), because of this line? 

`[ 8145.810000] SQUASHFS error: zlib decompression failed, data probably corrupt
`

We are definitely creating a dm-verity rauc bundle, we have it enabled in our linux kernel. 
`
rauc encrypt --to="crypt_cert.pem" --keyring="cert.pem" ${BINARIES_DIR}/conventional_rauc_bundle.raucb ${BINARIES_DIR}/encrypted_rauc_bundle.raucb`
`

ejoerns, 2023-01-13:
Ok, thus it is even a `crypt` bundle that is used. This actually implies using verity, yes.

Are there any kernel messages before the `SQUAFSFS error` that could give a hint?

Which sha256 is your target using? It can be seen in the kernel log like:

    device-mapper: verity: sha256 using implementation "stm32-sha256"

ShortCipher, 2023-01-13:
Hi @ejoerns! 

Thank you so much for your quick responses, and also thanks for your work on RAUC!!!  It's really awesome!  I'm a coworker of @deval-avive, and we're trying to figure out this corruption error.  I don't see a `dmesg` with "verity" but it looks like we are using Atmel crypto:
```
[    0.340000] atmel_sha f0028000.sha: version: 0x510
[    0.340000] atmel_sha f0028000.sha: using dma0chan4 for DMA transfers
[    0.350000] atmel_sha f0028000.sha: Atmel SHA1/SHA256/SHA224/SHA384/SHA512
```

jluebbe, 2023-01-16:
We've seen bugs in the kernel drivers for other crypto accelerators, so perhaps you could disable the `atmel-sha` driver (`CRYPTO_DEV_ATMEL_SHA`) and use the software implementation instead. That should help you with finding the source of the error.