rauc/rauc #697: Stripping trailing spaces in compatible field

SijmenHuizenga, 2021-02-09:
Hi, I'm loving this project, especially the documentation <3 Thanks for doing this!

It seems trailing whitespaces in `compatible` field in the manifest.raucm are not stripped.
```
[update]
compatible=picocoma9x 
[...]
```
(note the space after `picocoma9x`). This results in:
```
# rauc info update-2015.04-1.raucb 
Compatible:     'picocoma9x '
Version:        '2015.04-1'
[...]
```

The 'Compatible' field shows the trailing space.

Now, it seems that in `/etc/rauc/system.conf` the trailing whispaces _are_ stripped. The following file:
```
[system]
compatible=picocoma9x 
[...]
```
results in the following status message:
```
# rauc status
=== System Info ===
Compatible:  picocoma9x
[...]
```
Note that there is no space behind `picocoma9x`.

Trying to apply this bundle to this system fails because of incompatibility. 

It would be nice to either strip spaces in both configuration files, or not strip spaces at either.

SijmenHuizenga, 2021-02-09:
:thinking: Maybe the space in `rauc status` is only stripped for visualization... not entirely sure...

jluebbe, 2021-02-09:
Without looking a the source: perhaps we only strip whitespace when reading the system config, and not when reading the manifest. I agree that it should be done in both cases.

ejoerns, 2021-02-09:
> Hi, I'm loving this project, especially the documentation <3 Thanks for doing this!

Thanks! Great to hear it was worth the effort writing it :smirk:

>  Maybe the space in `rauc status` is only stripped for visualization... not entirely sure...

I've made a quick example for trying to reproduce it. But it seems as if the status output contains the space as well (even if it is not shown that explicitly as we do not have quotation marks there):

```
=== System Info ===
Compatible:  Test Config 
[...]
```

The method used for reading in the string from the configuration file is basically the same in both manifest and system config case.

Maybe the output is modified in your terminal?

SijmenHuizenga, 2021-02-09:
Aaah yes, your remark about my terminal being the possible culprit lead me to do some more debugging....

```
# rauc status | tr " " "."
===.System.Info.===
Compatible:..picocoma9x.
Variant:.....
Booted.from:.rootfs.0.(A)
```

Conclusion: The space was there all along all along!

Looking back, I might have been editing the `system.conf` and expecting the changes to be picked up by rauc that was running as a service. Some hours after posting this ticket I discovered that I have to restart the service for it to re-read the conf file.

Thanks a lot for the quick response.

closed 2021-02-09 by SijmenHuizenga