rauc/rauc #699: Simply nor and nand writes to a single `flashcp` command

hellow554, 2021-02-16:
Currently for nor and nand flashes they get first erased completly:

https://github.com/rauc/rauc/blob/2182bcfe9df92db085570f19bbda3c84219efdae/src/update_handler.c#L1222

https://github.com/rauc/rauc/blob/2182bcfe9df92db085570f19bbda3c84219efdae/src/update_handler.c#L1265

afterwards the nor flash get's its data by using `flashcp` and the nand flash get written by `nandwrite`

https://github.com/rauc/rauc/blob/2182bcfe9df92db085570f19bbda3c84219efdae/src/update_handler.c#L1230

https://github.com/rauc/rauc/blob/2182bcfe9df92db085570f19bbda3c84219efdae/src/update_handler.c#L1273

My question is: why?

flashcp is recommended for all sorts of flash operations, because it does the erase first (optionally full flash erase with `-A`), writes the data and verifies it afterwards.

The only advantage `nandwrite` has is that you can specify `--pad` which will write unaligned data to flash.

I would love to see a simplified `img_to_nor_handler` and `img_to_nand_handler` function.

Is this feasible?

ejoerns, 2021-02-16:
> flashcp is recommended for all sorts of flash operations, because it does the erase first (optionally full flash erase with `-A`), writes the data and verifies it afterwards.

This is incorrect for NAND flash, as flashcp does not handle bad blocks, etc (also see [[1](https://www.spinics.net/lists/linux-mtd/msg05945.html)] or [[2](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=813995)]).

It normally is the right tool for NOR, yes.

> The only advantage `nandwrite` has is that you can specify `--pad` which will write unaligned data to flash.

Nope (as noted above).

> I would love to see a simplified `img_to_nor_handler` and `img_to_nand_handler` function.
> 
> Is this feasible?

The reason why we need to do a full erase of the NOR mtd device is that you have to ensure the written result is always reproducible content and no remnants from older writes are left.

for NOR we already user flashcp, for NAND we cannot (but lack verification). Thus the only remaining thing is that we do not use `-A` option for the full erase but our method instead. However, I think that is not really an issue.

Probably verification of written NAND content could be something to think about.

Does that answer your question? Would you agree with my explanation?

hellow554, 2021-02-16:
I totally agree with your explanation, thanks!
My part of the internet told me to use `flashcp` over `nandwrite`. Funny that your side of the internet tells otherwise. I think yours has the better arguments ;)

closed 2021-02-16 by hellow554