rauc/rauc #700: Passing image information to run_slot_hook

hellow554, 2021-02-16:
Currently, the image is set to `NULL` in `img_to_nand_handler` and other functions.

https://github.com/rauc/rauc/blob/2182bcfe9df92db085570f19bbda3c84219efdae/src/update_handler.c#L1256

My scenario is, that I want to update a device on my platform that access the nand partition, so I use the `img_to_nand_handler` to write it to `/dev/mtd2`.
After the image is copied I need to set the file size of the written image to a certain part of the devices eeprom, therefore I register a post-install hook and want to access the image name: [`$RAUC_IMAGE_NAME`](https://rauc.readthedocs.io/en/latest/reference.html?highlight=barebox#term-rauc-image-name), but that one won't be set without a valid Image pointer:

https://github.com/rauc/rauc/blob/2182bcfe9df92db085570f19bbda3c84219efdae/src/update_handler.c#L786-L790

Is there a reason why it is set to `NULL` instead of `image`? (and not only in `img_to_nand_handler`, but in almost all other update handlers as well)

ejoerns, 2021-02-16:
The `image` option in `run_slot_hook()` was originally introduced (https://github.com/rauc/rauc/commit/e1f78bb5fc25a7b7ced0d91f9aca3dafb51057ea) for the 'install-hook' that allows replacing the slot write with a custom script/binary. For this, access to the image information was mandatory.

For the other (pre-/post-hooks), we simply did not have any use case that required the image information, yet. Thus this was initially set to NULL for all pre- and post-hooks.

Despite I do not fully know/understand your use case yet, it probably is valid and one could think about adding this information to other hooks, too. Why do you need to access the name? Don't you actually want to access the image size then (image->checksum.size)? This is probably also quite interesting for install handlers, anyway.

hellow554, 2021-02-16:
I want to access the name, because the size is not available for the post-install hook, I use `stat` to get the size of the file, hence I need the filename.

I would also vote for an addictional field `RAUC_IMAGE_SIZE`, which uses `image->checksum.size` as you said above.
I would happily create a PR for this. Just tell me what you think.

ejoerns, 2021-02-16:
@hellow554 sounds good from my side.

> I want to access the name, because the size is not available for the post-install hook. So I use stat to get the size of the file, hence I need the filename.

Ok, but that would then be fixed too and you could directly use `RAUC_IMAGE_SIZE`, right?

hellow554, 2021-02-16:
If `RAUC_IMAGE_SIZE` would exist, I could use that instead of manually using `stat`, of course :)

PR #703 (2021-03-05): add RAUC_IMAGE_SIZE as hook variable
Closes #700
closed 2021-03-05 by jluebbe