rauc/rauc #704: Perform post-install actions with Dbus Api

hellow554, 2021-02-18:
I have the following use-case:

We, as a company, provide a basic platform bundle containing the kernel, devicetree and rootfs.
Our customer can modify certain aspects of the system, e.g. chaning the IP address, putting certain files onto the rootfs and some more.

Because we are lazy and money is always tight, we want to let the customer do its own changes, instead of us asking every time.

Therefore we want to provide a configuration file where they can put in all the information.
Afterwards that file along with the `.raucb` (bundled together into a zip along with all the other needed files by the customer) gets send to the device, unpacked by our updater and then the rauc update process gets started via the Dbus Api.

Now the part, where our updater needs to modify certain parts of the rootfs.

My idea was to use a post-install handler (in the `system.conf`) that calls the original service (e.g. via dbus) and notifies it that the installation is done and the customization can begin. In the end, the updater notifies the handler (which listens for changes via `busctl monitor`) that the process is done and pass him a status code, e.g. `0` for success, and `1` for failure.

Is this a feasible way to do? Is there a time limit (apart from the systems watchdog) how long a script can be executed? Is there a better way to do this without mounting manually the new partition (that's exaclty what I want to avoid, because handling all the edge cases sounds like a pita).

hellow554, 2021-03-22:
Okay, here's something that really is a problem for this:

https://github.com/rauc/rauc/blob/b350a9cf05eda2ace4937b025020fccd60a12584/data/rauc.service.in#L11

This prevents other processes from accessing raucs mounts, so we can't write directly into them. I'm not entirely sure how to circumvent this problem yet, but removing that line from the service file.

Does anybody have a different idea?

jluebbe, 2021-03-22:
The ``MountFlags=slave`` is intended to avoid other processes observing partially installed filesystems.

It sounds like a post-install **hook** might be a better approach. That will run at the correct time and in the same mount namespace. Maybe don't call the customer's script directly from the hook and add a `| tee /var/log/custom-update.log` or something similar to get the scripts's output.

jluebbe, 2021-03-22:
post-install hook are also a good