rauc/rauc #706: extract trough D-Bus

SijmenHuizenga, 2021-02-19:
Hi! The cli offers `rauc extract`. Would it be possible to expose this functionality through the DBus api?

jluebbe, 2021-02-19:
Why would you use 'rauc extract' on the target instead of 'rauc install'? 'extract' is mainly useful on the development host system.

SijmenHuizenga, 2021-02-19:
Yes, I am aware using `rauc install` is the best way to install an update. However, sometimes an update comes in that cannot be installed through rauc. The device already verifies all incoming updates using rauc. It would be nice to use the same interface for verification and metadata for these 'special' updates. The `extract` method seems like exactly what I need :smile:.

In my case this special update presents itself as an 'extra' application that for unrelated reasons rauc cannot get their own partition, and need to be installed using some custom logic.

jluebbe, 2021-02-19:
This case should be handle by install hooks. These are run with the bundle verified and mounted, as part of the normal installation flow.

SijmenHuizenga, 2021-02-23:
You are totally right, I could be using the install hooks. I have been holding off on that because it introduces a new component: the hook scripts. There is already a bunch of logic in the codebase that executes rauc. Having yet another place where code resides required to install updates is not desirable. I would rather have the logic of installing the bundle be part of the codebase that executes rauc. What I want:
```
App downloads bundle -> Rauc extract -> App installs files
```
What the hooks offer:
```
App downloads bundle -> Rauc install (using hooks) -> Install script
```

hellow554, 2021-03-01:
Shameless plug: I asked a question in https://github.com/rauc/rauc/issues/704 which hasn't been answered yet, sadly. Maybe it can inspire you and let rauc do the actual work and hook yourself into the installation process.

jluebbe, 2022-04-22:
As this feature request doesn't really fits RAUC's design (a bundle describes the full intended system state), I'm going to close this issue. Especially with streaming support, the application should no longer need to download the bundle itself.

Note that you don't have to move the logic of installing custom slots in to the hooks, as you can call any binary already on the system.

closed 2022-04-22 by jluebbe