rauc/rauc #707: programs in /sbin are not considered

hellow554, 2021-02-22:
The following error appears during an update:

```
LastError: Installation error: Failed updating slot bootloader.0: failed to start nandwrite: Failed to execute child process ?nandwrite? (No such file or directory)
```

```
# which nandwrite
/usr/sbin/nandwrite
# echo $PATH
/usr/local/bin:/usr/bin:/bin:/usr/local/sbin:/usr/sbin:/sbin
```

```
ln -s /usr/sbin/nandwrite  /usr/bin/nandwrite
```

```
rauc install bundle.raucb
100% Installing done.
```

🤔

ejoerns, 2021-02-22:
How do you start RAUC service? Is it started manually from command line, via init system (initv/sytsemd), or D-Bus activated? If it's D-Bus activated, does it use systemd or init for this?

For the different cases, the proper PATH should actually be set either by your shell, by systemd or by https://github.com/rauc/rauc/blob/master/data/rauc-service.sh.in.

hellow554, 2021-02-22:
based on busctl, I guess it's bus activated:

```
# busctl
NAME                      PID PROCESS         USER             CONNECTION    UNIT                        SESSION DESCRIPTION
de.pengutronix.rauc       190 rauc            root             :1.6          rauc.service                -       -
```

Any idea what the default path is for systemd to spawn a process? I never actually did that myself.
Wouldn't the proper way be to spawn a shell in the update process and let the shell decide the correct path?

ejoerns, 2021-02-22:
> Any idea what the default path is for systemd to spawn a process? I never actually did that myself.

https://man7.org/linux/man-pages/man5/systemd.exec.5.html lists this in the `$PATH` description. According to this, `/usr/sbin` should be included (as expected).

> Wouldn't the proper way be to spawn a shell in the update process and let the shell decide the correct path?

Systemd is designed for spawning processes. I do not see why we should use a shell here.

Does it work when starting `rauc service` manually from your shell?

What does

```
systemd-run -t /usr/bin/env
```

show for `PATH`?

hellow554, 2021-02-22:
> Does it work when starting rauc service manually from your shell?

yes (:

```
# systemd-run -t /usr/bin/env
LANG=C
PATH=/usr/local/bin:/usr/bin
INVOCATION_ID=dba12d1f9aa14bd7bb78fb849092ef2e
TERM=xterm-256color
```

hmmm, interesting. I'll try to investiage where that comes from.

hellow554, 2021-02-22:
I'm sorry for the noise, it is my fault.
The [`split_bin`](https://github.com/systemd/systemd/blob/84d32bf59bb1a13162aae08022d115290bc8eb91/meson_options.txt#L12-L13) option in systemd was active due to a misconfigured bitbake recipe.

Thanks for your help here!

closed 2021-02-22 by hellow554