rauc/rauc #709: Obsolete test manifest file?

hongkongkiwi, 2021-02-25:
Hi there,

In the manifest test file: https://github.com/rauc/rauc/blob/master/test/manifest.raucm

It makes reference to some undocumented features such as [file.<slot-class>] and [keyring].

I only found reference to keyring in the system configuration file and the file.<slot-class> option seems very useful but doesn't seem to be documented.

Bastian-Krause, 2021-02-25:
The `keyring` section is documented [here](https://rauc.readthedocs.io/en/latest/reference.html?highlight=keyring#system-configuration-file).

The `[file.*]` and `[image.*]` sections are not meant to be specified manually. They are written to the manifest by rauc when you call `rauc bundle`.

ejoerns, 2021-02-25:
> The `[file.*]` and `[image.*]` sections are not meant to be specified manually. They are written to the manifest by rauc when you call `rauc bundle`.

To be precise: The `[file.*]` section is actually obsolete. We should remove this from the example imho.
The `[image]` section is required to be in the manifest. This is indeed meant to be specified manually!
However, most build systems have their own abstractions to generate it.

hongkongkiwi, 2021-02-26:
Thanks for looking into this.

I actually really wanted to use the file functionality for updating our
apps directly, so sad to see that go, but I noted in a previous question
that it was obsolete so wanted to confirm as it was in the example manifest.

My context for the question writing shell scripts to generate the manifest
file which our CI system uses to bundle.

For the keyring section, I see that documentation only refers to the
“system config file”. Which is clear and well documented, I couldn’t see
any documentation about a keyring section in the bundle manifest file.
However that’s in the example test manifest I linked. I guess just a
mistake?

jluebbe, 2021-02-26:
> I actually really wanted to use the file functionality for updating our apps directly, so sad to see that go, but I noted in a previous question that it was obsolete so wanted to confirm as it was in the example manifest. 

The file functionality was completely separate from the bundle mechanism and used for updating filesystem images for loop-back-mounting, not for general file-based update. The overall approach of RAUC is based on ensuring that the whole system state is ensured after an update, which is much harder to do without the image-based appoach.

> For the keyring section, I see that documentation only refers to the “system config file”. Which is clear and well documented, I couldn’t see any documentation about a keyring section in the "bundle manifest file". However that’s in the example test manifest I linked. I guess just a mistake?

Yes, that's a mistake as well (left-over from initial experiments).

ejoerns, 2021-04-13:
I've addressed this now in #723 by removing the deprecated sections.

PR #723 (2021-04-13): Remove Long Deprecated Sections from Manifest Parsing.
This removes two manifest config options which were parsed but not actually used anywhere in the code.
This is unexpected and violates the concept that a manifest configuration/content actually has an impact on the installation.

Removed sections are:

* The `file.<slotclass>/<filename>` sections which originates from the long deprecated bundle-free network mode (not to be confused with the current network capabilities of RAUC) we had in the early days of RAUC.
* The `keyring` section that existed since the initial concept but actually nev [... cut]
closed 2021-04-13 by jluebbe