rauc/rauc #712: Receiving status updates without DBus

hongkongkiwi, 2021-03-17:
Hi there,

I'm working with RAUC on an embedded system. I prefer to keep the memory requirements low and don't want to include DBus. In our case we are using MQTT, so I would prefer to get the information into this bus.

Is it possible to have RAUC output information to a socket file instead of dbus? This way we could simply read the status from there. Specifically I would like to get status notifications without having a heavy dbus dependency.

My app can read a pipe or socket or file or something which can emit status updates from RAUC. Is there any alternative? Heck, I could even connect to a port if there were somehow status updates sent to that.

ejoerns, 2021-03-17:
@hongkongkiwi the only alternative for using D-Bus is getting status information via command line tool. Thus, no, we do not support status output to sockets.

The reason for using D-Bus here is that it is the de-facto standard for system IPC in almost all modern Linux systems. And nowadays storage is normally not **that much** limited, to not being able to include D-Bus at all.

Where does the wish to keep a super small system originate from? Are you actually that limited from the storage perspective?
Otherwise, the good wish to keep things small often leads to more complexity than required and prevents from using all proven standard mechanisms.

hongkongkiwi, 2021-05-28:
We are creating a small IOT system, dbus is quite heavy for this purpose as the system has less than 32mb of usable ram. Internally we use something similar to mqtt for internal/external messaging.

So I was hoping to easily bridge rauc to this bus. Perhaps in future it might be good to make a plugin interface so dbus and libdbus is not required and we could code in mqtt support? Just a thought.

jluebbe, 2023-05-11:
We don't have plans to implement this and have not seen other requests for this feature.

Currently, D-Bus support is deeply integrated with RAUC's service part, so adding another RPC method is not simple. You can build RAUC without D-Bus or service support, though and just use the CLI tool. Building on top of that with machine parseable progress messages might be the easiest way for your case.

I'm going to close this for now. Please reopen if you (or someone else) has plans to push this forward.

closed 2023-05-11 by jluebbe