rauc/rauc #713: signature codesign issue

hellow554, 2021-03-18:
Let's take the current test script from https://github.com/rauc/rauc/blob/master/test/openssl-ca-check.sh and extend it by these four lines:

```diff
diff --git a/test/openssl-ca-check.sh b/test/openssl-ca-check.sh
index cb5944e..8e03849 100755
--- a/test/openssl-ca-check.sh
+++ b/test/openssl-ca-check.sh
@@ -40,6 +40,10 @@ echo "Sign and check with Autobuilder-1"
 openssl cms -sign -in manifest -out manifest-a1.sig -signer dev/autobuilder-1.cert.pem -inkey dev/private/autobuilder-1.pem -outform DER -nosmimecap -binary -certfile dev/ca.cert.pem
 openssl cms -verify -in manifest-a1.sig -content manifest -inform DER -binary -crl_check -CAfile provisioning-ca.pem || echo FAILED

+echo "Sign and check with codeSign"
+openssl cms -sign -in manifest -out manifest-cs.sig -signer dev/xku-codeSigning.cert.pem -inkey dev/private/xku-codeSigning.pem -outform DER -nosmimecap -binary -certfile dev/ca.cert.pem
+openssl cms -verify -in manifest-cs.sig -content manifest -inform DER -binary -crl_check -CAfile provisioning-ca.pem || echo FAILED
+
 echo "Sign and check with Autobuilder-2 (revoked)"
 openssl cms -sign -in manifest -out manifest-a2.sig -signer dev/autobuilder-2.cert.pem -inkey dev/private/autobuilder-2.pem -outform DER -nosmimecap -binary -certfile dev/ca.cert.pem
 echo "  without CRL"
```

This will result in the error:

```
+ echo 'Sign and check with codeSign'
Sign and check with codeSign
+ openssl cms -sign -in manifest -out manifest-cs.sig -signer dev/xku-codeSigning.cert.pem -inkey dev/private/xku-codeSigning.pem -outform DER -nosmimecap -binary -certfile dev/ca.cert.pem
+ openssl cms -verify -in manifest-cs.sig -content manifest -inform DER -binary -crl_check -CAfile provisioning-ca.pem
Verification failure
139757941069120:error:2E099064:CMS routines:cms_signerinfo_verify_cert:certificate verify error:../crypto/cms/cms_smime.c:252:Verify error:unsupported certificate purpose
+ echo FAILED
FAILED
```

what is the reason that the xku-codeSigning certificate isn't checked in that script? If I see it corretly, I can't have the extendedKeyAttribute for now, because openssl will reject it?

My openssl version (on Ubuntu 20.04):

    OpenSSL 1.1.1f  31 Mar 2020

Emantor, 2021-03-18:
OpenSSL has no knowledge of the `xku-codeSigning` pupose and therefore rejects it. RAUC registers a custom handler which adds and respects this purpose, for more info see https://github.com/rauc/rauc/pull/570.

hellow554, 2021-03-18:
But I can observe the exact same behavoir (`unsupported certificate purpose`) if I use the meta-rauc layer and try to create and sign a bundle:

```
ERROR: my-bundle-1.0-r0 do_bundle: Execution of '/build/tmp/work/arm-linux-gnueabi/my-bundle/1.0-r0/temp/run.do_bundle.610' failed with exit code 1:
rauc-Message: 05:19:04.038: Debug log domains: 'rauc'
(rauc:613): rauc-DEBUG: 05:19:04.045: bundle start
(rauc:613): rauc-DEBUG: 05:19:04.045: system config not found, using default values
(rauc:613): rauc-DEBUG: 05:19:04.046: input directory: /build/tmp/work/arm-linux-gnueabi/my-bundle/1.0-r0/bundle
(rauc:613): rauc-DEBUG: 05:19:04.046: output bundle: /build/tmp/work/arm-linux-gnueabi/my-bundle/1.0-r0/build/bundle.raucb
(rauc:613): rauc-DEBUG: 05:19:04.946: Payload size: 53108736 bytes.
Creating bundle in 'verity' format
(rauc:613): rauc-DEBUG: 05:19:04.946: Hash creation SHA256, data blocks 12966.
(rauc:613): rauc-DEBUG: 05:19:04.946: Using 2 hash levels.
(rauc:613): rauc-DEBUG: 05:19:04.946: Data size: 53108736 bytes.
(rauc:613): rauc-DEBUG: 05:19:04.946: Hashed size: 53530624 bytes.
rauc-Message: 05:19:05.077: Keyring given, doing signature verification
Failed to create bundle: failed to sign manifest: signature verification failed: Verify error:unsupported certificate purpose
WARNING: exit code 1 from a shell command.
```

Emantor, 2021-03-18:
Do you have another purpose embedded in the certificate used for RAUC signing? Can you post the certificate attributes here? How did you create the certificate?

hellow554, 2021-03-18:
I created the certificates using openssl 🙈

I have one root, one intermediate and one "signing" certificate.
Here are the relevant sections for the certificates:

<details>
<summary>Root CA</summary>

```
[ root_ca_ext ]
keyUsage                = critical,keyCertSign,cRLSign
basicConstraints        = critical,CA:true
subjectKeyIdentifier    = hash
authorityKeyIdentifier  = keyid:always

[ signing_ca_ext ]
keyUsage                = critical,keyCertSign,cRLSign
basicConstraints        = critical,CA:true,pathlen:0
subjectKeyIdentifier    = hash
authorityKeyIdentifier  = keyid:always
```
</details>

<details>
<summary>Intermediate</summary>

```
[ codesign_ext ]
basicConstraints        = CA:false
keyUsage                = critical,digitalSignature
extendedKeyUsage        = critical,codeSigning
subjectKeyIdentifier    = hash
authorityKeyIdentifier  = keyid:always
```
</details>

<details>
<summary>Leaf</summary>

```
[ rauc_reqext ]
keyUsage                = critical,digitalSignature
extendedKeyUsage        = critical,codeSigning
subjectKeyIdentifier    = hash
```
</details>

Does that help or do you need anything else?

Emantor, 2021-03-18:
I assume `check-purpose=codesign` is also setup in your manifest. If so I am running out of things to check.

hellow554, 2021-03-18:
Just for clarification:

The system.config is know that to do an rauc update, right?

The manifest file is for creating such rauc update, right?

So how does the system config influences the bundle creation? Or is rauc checking the `bundle.raucb` against the system config **on the host** during the bitbake run?

jluebbe, 2021-03-18:
The resulting signature is checked on the host if you pass a keyring during bundle creation. In that case, you also need the matching system.conf on the host.

hellow554, 2021-03-18:
Ok, that could be the case then.
I haven't added that line yet, because I though it's only relevant for the target side.

I will check on this later, so give me a day or two.

jluebbe, 2021-03-18:
It might be useful to allow setting these options on the command line as well, avoiding the need for a systemd.conf on the development host.

ejoerns, 2021-03-18:
> It might be useful to allow setting these options on the command line as well, avoiding the need for a system.conf on the development host.

I'd agree. Personally, I would love to get rid of anything that smells like system.conf on the host side, anyway... :smirk:

jluebbe, 2022-08-19:
The remaining aspect here is the ability to set config options via the command line. We have #953 for that, so I'll close this issue.

closed 2022-08-19 by jluebbe