rauc/rauc #715: Support casync for NAND

wahwahe, 2021-03-25:
Using RAUC, Currently, only UBI volumes are handled for updating casync bundles.
Do you plan to handle also the update of the whole UBI image in the NAND flash memory with casync?

ejoerns, 2021-03-25:
@wahwahe NAND updating requires badblock handling, thus we currently use `nandwrite` command for this which takes an image as input. I'm not sure if this can be combined with casync in any way. `nandwrite` can also read from stdin which might be an option..

@OnkelUlla maybe you have some additional suggestions here?

OnkelUlla, 2021-03-26:
Taking a deeper look into the sources of ``nandwrite`` [here](http://git.infradead.org/mtd-utils.git/blob/v2.1.2:/nand-utils/nandwrite.c#l234), [here](http://git.infradead.org/mtd-utils.git/blob/v2.1.2:/nand-utils/nandwrite.c#l330) and [here](http://git.infradead.org/mtd-utils.git/blob/v2.1.2:/nand-utils/nandwrite.c#l411) it should just work™ when being fed the image via ``stdin`` instead of letting it open and read a regular file given on the commandline. This means that it should be enough to extend https://github.com/systemd/casync/pull/227 with ``S_ISFIFO(…)`` as already outlined in that pull request - given that this is okay on ``casync``'s side. Unfortunately nobody ever commented that pull request contentwise.

ejoerns, 2021-06-16:
I'll close this now as I consider the question to be answered basically. Feel free to reopen if further questions about this topic arise.

closed 2021-06-16 by ejoerns