rauc/rauc #717: Bundle signing issues

akaryakinx, 2021-03-30:
Hi, I am trying to resign a bundle, signed with a self-signed certificate, with a valid public certificate. 

./rauc resign -c system.conf --keyring=ca.crt.pem --intermediate=intermediate.pem --key=sign.key.pem --cert=sign.crt.pem casync-update-bundle.raucb casync-update-bundle-signed.raucb

rauc-Message: 17:36:42.174: Config option 'statusfile=<path>/per-slot' unset, falling back to per-slot status
rauc-Message: 17:36:42.174: Using per-slot statusfile
rauc-Message: 17:36:42.174: Reading bundle: casync-update-bundle.raucb
Reading bundle in 'verity' format
Creating bundle in 'verity' format
rauc-Message: 17:36:44.781: Keyring given, doing signature verification
Failed to resign bundle: failed to sign manifest: signature verification failed: Verify error:unable to get local issuer certificate

I use the config file because there is no way to specify the certificate directory on the command line. Specifying the config file works for bundle extraction.

[system]
compatible=machine
bootloader=uboot

[keyring]
directory=/etc/ssl/certs/

I think what happens is the initial verification of the old signature is successful, as well as signing with the new key. However, after signing, it tries to verify the *new* signature and it fails because 1) the keyring file provided on the command line is not valid for the new signature 2) the keyring directory from the config file is not used. I think the final verification step is unnecessary and if I replace the condition "if (keyring_path || keyring_dir)" with "if (0)" in cms_sign(), the process completes and the produced bundle seems to be valid.

I think there are several issues come to play.

1. It would be nice to take keyring directory from the command line. Requiring a config file for a command line tool is a bit annoying. There are also requires some parameters in the system section, which are not necessary for extraction/resigning/verification.
2. Parameter handling could be streamlined so that if there is signature-related ones in the config file, they are used.
3. Verification of the final signature could be made optional or produce a warning instead of terminating.
4. When resigning, there is a fundamental problem that two certificate stores are needed. I'm not sure how to better handle that.
5. Not fully related, but when I tried the --no-verify parameters, it would still require the keyring file, which seems unnecessary. Also, the final signature verification step seems optional, as suggested by the message ("Keyring given, doing signature verification"). But I could not find a way to run it without the keyring (it complains with "Cert, key and keyring files must be provided").

It's possible I'm getting things wrong and create bundles incorrectly. But if my concerns are valid, I could come up with fixes.

ejoerns, 2021-03-30:
> 1. It would be nice to take keyring directory from the command line. Requiring a config file for a command line tool is a bit annoying. There are also requires some parameters in the system section, which are not necessary for extraction/resigning/verification.

I personally agree with you that using a config file for the host tool part of RAUC is inconvenient and personally would highly like to avoid this in the future and better seperate host and target tool usage anyway.
So you really need both verification and a keyring directory for your use case here? I bet this could be added as an argument..

> 3. Verification of the final signature could be made optional or produce a warning instead of terminating.

This should be optional. Also note that we have two different arguments for the resign case `--keyring` and `--signing-keyring` where the latter is explicitly for post-siging verification while the first should be used for pre-resign verification only. Did you check that?

akaryakinx, 2021-03-31:
Thanks. I actually did not notice the `--signing-keyring` parameter, and it works for me. But there is no way to use a directory. And it seems there is no no to avoid this verification because there is always a keyring provided during resigning.

jluebbe, 2021-03-31:
We could check if the target of `--keyring` or `--signing-keyring` is a directory and then handle it accordingly.

I'd be interested why you use directories. Do you have a large number of independent PKI hierarchies?

akaryakinx, 2021-03-31:
This mainly came from testing of the resign procedure, without a particular use case. I didn't know about `--signing-keyring` and was trying to avoid the failure in the last verification step. So, supporting directories may not be needed after all.

jluebbe, 2022-08-19:
It looks like the issue was solved by using `--signing-keyring`. Please reopen, if that's not correct.

closed 2022-08-19 by jluebbe