rauc/rauc #718: rauc install fails woth dosfstools v4.2

bertouttier, 2021-04-06:
When using rauc install on a system with dosfstools v4.2 installed, the installation fails. This is because dosfstools v4.2 checks the vfat partition label name, and raises an error when it contains a '.' character. Since RAUC sets dots in vfat partition labels, the isntallation fails.

A workaround is to patch dosfstools in src/common.c and to allow dots in the label of the partition:
```
--- a/src/common.c
+++ b/src/common.c
@@ -350,7 +350,6 @@
             ret |= 0x02;
         if (doslabel[i] == 0x22 ||
             (doslabel[i] >= 0x2A && doslabel[i] <= 0x2C) ||
-            doslabel[i] == 0x2E ||
             doslabel[i] == 0x2F ||
             (doslabel[i] >= 0x3A && doslabel[i] <= 0x3F) ||
             (doslabel[i] >= 0x5B && doslabel[i] <= 0x5D) ||
```

What are your thoughts on fixing this?

jluebbe, 2021-04-06:
We use the slot name as the label by default, which contains a `.`. Is there a specific reason why `.` is no longer allowed?

ejoerns, 2021-04-06:
This was explicitly introduced with https://github.com/dosfstools/dosfstools/commit/40da1b2c0129124f70fc0ec74363c699a3a479f8, stating this is not allowed by FAT specification.

jluebbe, 2021-04-06:
So we could replace the `.` with `~`, `-` or `_`? Any preference?

bertouttier, 2021-04-06:
I guess `_` will do!

ejoerns, 2021-04-06:
The original document [[1]](http://download.microsoft.com/download/0/8/4/084c452b-b772-4fe5-89bb-a0cbf082286a/fatgen103.doc) says

> The following characters are not legal in any bytes of DIR_Name: 
    • Values less than 0x20 except for the special case of 0x05 in DIR_Name[0] described above.
    • 0x22, 0x2A, 0x2B, 0x2C, 0x2E, 0x2F, 0x3A, 0x3B, 0x3C, 0x3D, 0x3E, 0x3F, 0x5B, 0x5C, 0x5D, and 0x7C. 
> 
> Here are some examples of how a user-entered name maps into DIR_Name:
> 
> “foo.bar”            -> “FOO     BAR”
> “FOO.BAR”            -> “FOO     BAR”
> “Foo.Bar”            -> “FOO     BAR”
> “foo”                -> “FOO        “
> “foo.”               -> “FOO        “
> “PICKLE.A”           -> “PICKLE  A  “
> “prettybg.big”       -> “PRETTYBGBIG”
> “.big”               -> illegal, DIR_Name[0] cannot be 0x20

Anyway, not sure in how DIR_Name[0] and the volume label are equal.

Using "_" sounds safe, anyway.

Bastian-Krause, 2021-04-06:
> Anyway, not sure in how DIR_Name[0] and the volume label are equal.

The volume name (label) is actually stored as the name of a file with the ATTR_VOLUME_ID attribute:

> ATTR_VOLUME_ID 	There should only be one “file” on the volume that has this attribute set, and that file must be in the root directory. This name of this file is actually the label for the volume.

 [[1]](http://download.microsoft.com/download/0/8/4/084c452b-b772-4fe5-89bb-a0cbf082286a/fatgen103.doc)

ejoerns, 2021-04-06:
@Bastian-Krause Ok, missed the part that these rules are valid for each FAT entry (and thus also for the 'volume file'). Thx!

PR #725 (2021-04-16): src/update_handler: generate only valid vfat label names
Depending on the vfat implementation, vfat labels can be interpreted quite strictly, following the original specification.

This is why mkfs.vfat introduced more stricter sanity checks on label names with dosfstoolfs v4.2.

See also dosfstools commit 40da1b2c ("fatlabel: Implement proper checks when setting new label") for this:
https://github.com/dosfstools/dosfstools/commit/40da1b2c0129124f70fc0ec74363c699a3a479f8

Labels created so far by RAUC did not follow the FAT specification, too. This becomes actually more critical as recent mkf [... cut]
closed 2021-04-16 by ejoerns