rauc/rauc #722: DBus errors on non-systemd system

incaseoftrouble, 2021-04-10:
Hi,

I'm using rauc on a non-systemd raspberry (systemd would be overkill, I'm running < 10 processes).
The system is built with current buildroot LTS (2021.02.1), Linux 5.4 with muslc, `rauc --version > rauc 1.5.1`
Busybox takes care of the init (simple inittab file).
The system is booted from a squashfs image, so by default it is read-only. I mounted tmpfs on the usual paths (/run, /var/run, /tmp) and /var/lib/dbus (for the machine-id)

It might well be the case that I simply use dbus in the wrong way, I have no idea how exactly it works.
I started `dbus-daemon --system --nofork --nopidfile` (in inittab) and a dbus socket is created in `/run/dbus/system_bus_socket`

I observe the following outputs:
```
$ rauc -d status
rauc-Message: 00:40:09.186: Debug log domains: 'rauc'
(rauc:467): rauc-DEBUG: 00:40:09.195: status start
rauc-Message: 00:40:09.200: rauc status: error retrieving slot status via D-Bus: error creating proxy: Could not connect: No such file or directory
```
```
$ rauc -d service
rauc-Message: 00:41:34.931: Debug log domains: 'rauc'
(rauc:469): rauc-DEBUG: 00:41:34.940: service start
Connection to the system bus can't be made for de.pengutronix.rauc
```
I guess this could be attributed to missing environment variables informing rauc about the dbus?

If I try to nudge dbus to launch rauc, I get the following:
```
$ dbus-send --print-reply --system --type="method_call" --dest=de.pengutronix.rauc / de.pengutronix.rauc.Foobar
Error org.freedesktop.DBus.Error.Spawn.ChildExited: Launch helper exited with unknown return code 1
```
Syslog gives me
```
Jan  1 00:44:18 (none) daemon.info dbus-daemon[364]: [system] Activating service name='de.pengutronix.rauc' requested by ':1.33' (uid=0 pid=497 comm="dbus-send --print-reply --system --type=method_cal") (using servicehelper)
Jan  1 00:44:18 (none) daemon.info dbus-daemon[364]: [system] Activated service 'de.pengutronix.rauc' failed: Launch helper exited with unknown return code 1
```
I modified `/usr/libexec/rauc-service.sh` to write stdout and stderr into some files (with `G_MESSAGES_DEBUG="all"`), then running this above dbus-send again gives
```
STDOUT:
(rauc:553): rauc-DEBUG: 00:57:00.900: service start
STDERR:
rauc-Message: 00:57:00.892: Debug log domains: 'all'
Connection to the system bus can't be made for de.pengutronix.rauc
```

Sorry if this is a misconfiguration on my side, I simply couldn't find info on how to set rauc up with a "manual" dbus. It might also be the case that the error is on the side of buildroot (some directories not being created) or that muslc is causing this problem. Unfortunately, the error is completely cryptic to me and I'm kinda out of ideas.

incaseoftrouble, 2021-04-10:
Well, sent this one moment too soon it seems. `strace` revealed that rauc accesses `/var/run/dbus/system_bus_socket`. My dbus created the file in `/run/dbus/system_bus_socket`.

I have
```
$ grep -R /run/dbus/system_bus_socket /usr/share/
/usr/share/dbus-1/system.conf:  <listen>unix:path=/run/dbus/system_bus_socket</listen>
```
so it seems that the dbus-path is hardcoded for rauc.

After symlinking the files, `rauc status` works.

Suggestion: Make the error message nicer :-)

ejoerns, 2021-04-10:
I don't agree that systemd is overkill for your use case, except if you are really really storage limited, as you loose a lot of convenience and functionality when leaving it out, but surely will not have any negative effects when using it. But this just as a side note.

About your issues: The path you refer to is the hard-coded fallback in glib, which is selected when `DBUS_SYSTEM_BUS_ADDRESS` environment variable is not set. Also typically, legacy path `/var/run`  is a symlink to `/run` on most system integrations. I'd assume this is the default in Buildroot, too.

> Suggestion: Make the error message nicer :-)

Unfortunately, I don't know of any possibility to better distinguish between types of failures on this level as we only get a `NameLost` signal here. I agree this is quite generic for various possible error sources. Suggestions are welcome.

incaseoftrouble, 2021-04-10:
> I don't agree that systemd is overkill for your use case, except if you are really really storage limited, as you loose a lot of convenience and functionality when leaving it out, but surely will not have any negative effects when using it. But this just as a side note.

Well, enabling systemd in buildroot adds quite a bit of packages and indirections. `runsvdir` is (IMO) easier for such extremly simple systems (maybe 30 files).

(And I already have everything set up for runsvdir already, I'm migrating it to a more modern system :-) )

> About your issues: The path you refer to is the hard-coded fallback in glib, which is selected when `DBUS_SYSTEM_BUS_ADDRESS` environment variable is not set.

Ah! I see. I thought that this is openend by rauc directly. Then I guess there is no easy way to detect this issue on your side / its not an issue of rauc but rather `glib`. I guess you can close this issue, this problem is very specific (and indirectly caused by my misconfiguration, not symlinking `/var/run` to `/run`).

Just out of curiosity: Would you know how to run dbus/rauc appropriately without systemd? I guess you'd need to run dbus with `--print-addresss` and export the variables manually, right?

jluebbe, 2021-04-12:
You could also configure rauc with `--disable-service`, which removes the dbus dependancy, so you only have the `rauc install ...` CLI.

incaseoftrouble, 2021-04-12:
This would actually be my preferred approach, but `buildroot` is not aware of this (dbus is a hard dependency of rauc there).

Do you plan to support `--disable-service`? Then I'll submit an issue to buildroot and maybe work on making the dependency on dbus optional. Alternatively, I could also just add rauc as custom package I guess.

jluebbe, 2021-04-12:
`--disable-service` is already available in rauc and part of the CI tests, so it should work fine. That mode probably receives less real-world usage though, at least based on the systems we build/maintain for our customers.

incaseoftrouble, 2021-04-12:
Great, thanks. I'll just close this issue then and see how I proceed.

I'm planning to use the hawkbit-updater to deploy updates, so I guess I need DBus anyway. Thank you for your responses!

closed 2021-04-12 by incaseoftrouble