rauc/rauc #727: OTA Install Progress ?

om26er, 2021-04-17:
We want to show to the user that an OTA update is being installed. Currently I don't know if there is a way we could inquire any sort of "progress" from rauc. Also how would we be able to tell them the remaining time for the OTA update to install.

ejoerns, 2021-04-17:
Do you mean something like that: https://rauc.readthedocs.io/en/latest/using.html?processing-progress-data#processing-progress-data ?
Or could you be more precise about what you are missing there? In general, actual 'time' prediction is quite tricky or not possible at all.

om26er, 2021-06-16:
@ejoerns We are using Casync based updates and the progress is mostly
```
installing
  0% Installing
  0% Determining slot states
 20% Determining slot states done.
 20% Checking bundle
 20% Verifying signature
 40% Verifying signature done.
 40% Checking bundle done.
 40% Checking manifest contents
 60% Checking manifest contents done.
 60% Determining target install group
 80% Determining target install group done.
 80% Updating slots
 80% Checking slot rootfs.0
 90% Checking slot rootfs.0 done.
 90% Copying image to rootfs.0
```
In just two seconds and the remaining copying just never updates until the update is actually applied.

So I was looking for a bit more "detailed" progress something that likely shows the progress of the "copying" part, so that the users could know of the progress.

om26er, 2021-06-16:
Since the delta is calculated device side when using casync, I understand that it's an intensive task and takes a long time (20 minutes+ in some cases, depending on the size of the RootFS). So having a way through which the user could know of the progress would be helpful.

ejoerns, 2021-06-16:
I totally understand the motivation for your request. The problem is: How do we get this information out of casync? We can call it verbose to get a summary of seeding time at the end, but that is of course not what we want. 

I would not know of any existing option to get 'live' information about progress inside casync.

This issue applies to various 'copy' methods in RAUC. While we could get the progress of some, we cannot get for most. This is the main reason why there is still no 'detailed' progress implemented at all.

The follow-up question if we had detailed progress would than still be: How do we generate realistic percentage values from this? Because we normally do not know in advance how long a signature verification takes compared to copying the data to the system's slots. So this could still result in the annoying behavior that you have 90% of your installation done according to progress indication, but only 10% of the entire installation time.

The hard way to go could be to measure installation time and take this as a guess for the duration of the next installation. But, especially with casync, this may gracefully fail as time to copy data might vary significantly between installation.

If you have other good ideas on how to target this, please feel free to propose.

om26er, 2021-06-17:
I have been trying to experiment with `desync` and `casync-nano` since the development seems stagnant on `casync`.

Didn't get much far with `casync-nano` because it needs chunks to be sha256, so I will try to change rauc tomorrow to generate sha256 chunks (the `casync make` call really).

Desync seems to show progress on the command line and my initial testing suggests it's faster than `casync`. However the problem is that it doesn't support block device as seed.

om26er, 2021-06-23:
To have "real" progress we'd need to make some changes in Casync it seems but some adaptations would also need to be done on rauc's side.

* As we can't really predict the time it may take to calculate the hashes, that step would likely give something like "calculating delta"
* For the actual chunk download+installation it would be possible to use the same mechanism that `desync` uses.

jluebbe, 2023-05-11:
For raw and tar images, we now have fine-grained progress information thanks to #1093.

With regards to casync, there is currently no real activity, so we don't have concrete plans to work on this.