rauc/rauc #728: open_slot_device() fails on slot type boot-mbr-switch

MartinH76761, 2021-04-20:
> Issue ticket regarding mailing list entry:
> https://www.mail-archive.com/rauc@pengutronix.de/msg00206.html
 
When updating a bootloader slot of type boot-mbr-switch the function g_open() in open_slot_device() from update_handler.c fails:

```
90% Copying image to bootloader.0 failed.
[...]
LastError: Installation error: Failed updating slot bootloader.0: Opening output device /dev/mmcblk0 failed: Device or resource busy
```

In my opinion this is due to the O_EXCL flag in g_open(), because for type boot-mbr-switch the code tries to open the full device (/dev/mmcblk0) instead of an inactive partition (/dev/mmcblk0p1). I assume the device is busy because at leat the rootfs partition of the currently running system is mounted. For an associated system.conf see the documentation example 7.6.2.

I suggest that the O_EXCL flag should not be used on slot type boot-mbr-switch, because in this case it is legitimate to open a busy device for writing (compared to a "normal" slot update). Some testing confirms the functionality.

jluebbe, 2021-04-20:
Hmm, which version are you using? I don't see how open_slot_device() would be called from img_to_boot_mbr_switch_handler(). There is a call to g_open() in clear_boot_switch_partition(), but it's not using O_EXCL.

Could you check that the correct handler is used?

ejoerns, 2021-06-03:
@MartinH76761 do you have any further hints on this?

ejoerns, 2023-02-01:
@Hollingsworth-martin-bl thanks for sharing your patch!

I would still agree with @jluebbe that the code you refer to here cannot be called for ``boot-mbr-switch``. Could you describe if you have verified you actually ended up in the patched condition?

Also please note that `honister` is [EOL](https://wiki.yoctoproject.org/wiki/Releases) since October 2021. Thus now migrating to it sounds wrong.

Hollingsworth-martin-bl, 2023-02-02:
> I would still agree with @jluebbe that the code you refer to here cannot be called for `boot-mbr-switch`. Could you describe if you have verified you actually ended up in the patched condition?

I verified with a debug print that I end up in this case and this was the fix that actually reproducible solved the issue. However on the same SD card layout but with yocto honister and rauc v1.7 a quick test indicated that the patch is no longer needed. I did however not digg any deeper and agree that the issue is closed 👍 

> Also please note that `honister` is [EOL](https://wiki.yoctoproject.org/wiki/Releases) since October 2021. Thus now migrating to it sounds wrong.

Thanks for the heads up, tell that my SOC vendor ^^ Luckily the vendor has kirkstone in the pipeline for Q1/Q2 2023.

ejoerns, 2023-02-02:
@Hollingsworth-martin-bl Thanks for the explanation!

> Thanks for the heads up, tell that my SOC vendor ^^ Luckily the vendor has kirkstone in the pipeline for Q1/Q2 2023.

The great thing about open source is that you are not tied to what your vendor provides you. :wink: 
If your kernel/bootloader expertise is limited in your project, it might make sense to take the BSP layer from the vendor, but in most cases, it should be possible to build the rest of the software on a clean and recent (or LTS) poky. :rocket: