rauc/rauc #729: Unsupported certificate purpose when creating bundle

incaseoftrouble, 2021-04-20:
I'm creating a bundle using `rauc bundle --key=<key> --cert=<cert> --keyring=<ca> --intermediate=<intermediate> dir`. This fails with `Failed to create bundle: failed to sign manifest: signature verification failed: Verify error:unsupported certificate purpose`. 
Without `--keyring`, i.e. without verification on the host, bundle creation works. I suspect that an error occurs when parsing the keyring file

The certificate chain is valid:
```
$ openssl verify -CAfile <ca> -untrusted <intermediate> <cert>
<cert>: OK
```

I have a "normal" root -> intermediate -> signing setup, where
```
[ v3_root ]
subjectKeyIdentifier = hash
authorityKeyIdentifier = keyid:always,issuer
basicConstraints = critical, CA:true
keyUsage = critical, digitalSignature, cRLSign, keyCertSign

[ v3_intermediate ]
subjectKeyIdentifier = hash
authorityKeyIdentifier = keyid:always,issuer
basicConstraints = critical, CA:true, pathlen:0
keyUsage = critical, digitalSignature, cRLSign, keyCertSign

[ v3_codesign ]
subjectKeyIdentifier = hash
authorityKeyIdentifier = keyid:always,issuer:always
basicConstraints = CA:FALSE
extendedKeyUsage = critical, codeSigning
```
with the last according to the [docs](https://rauc.readthedocs.io/en/latest/advanced.html#certificate-key-usage-attributes)

This might be related to #713 . The problem also occurs when I run `rauc` with `-c system.conf` with a `[keyring] check-purpose=codesign`

My version:
```
$ rauc --version
rauc 1.5.1
```

incaseoftrouble, 2021-04-20:
Addition: Reading throught the docs again:

> Also, you can optionally use extendedKeyUsage attributes on intermediate CA certificates to limit which ones are allowed to issue code signing certificates.

So it might be the case that the root / intermediate certs also need the codeSigning usage?

jluebbe, 2021-04-20:
The keyUsage/extendedKeyUsage looks fine (the check is implemented in https://github.com/rauc/rauc/blob/master/src/signature.c#L21). I would have hoped that `-c system.conf` with the purpose should be enough for `rauc bundle`. You could make `setup_x509_store()` more verbose to verify that the correct purpose is configured.

incaseoftrouble, 2021-04-20:
More info: With `system.conf` having `check-purpose=any`, it works.

Running with `G_MESSAGES_DEBUG="all"` reveals the following:

intermediate:
```
                object: X509v3 Key Usage (2.5.29.15)
                critical: TRUE
                value:
                  0000 - 03 02 01 86                              ....

                object: X509v3 Extended Key Usage (2.5.29.37)
                critical: TRUE
                value:
                  0000 - 30 0a 06 08 2b 06 01 05-05 07 03 03      0...+.......
```
and sigining:
```
                object: X509v3 Basic Constraints (2.5.29.19)
                critical: BOOL ABSENT
                value:
                  0000 - 30 00                                    0.

                object: X509v3 Extended Key Usage (2.5.29.37)
                critical: TRUE
                value:
                  0000 - 30 0a 06 08 2b 06 01 05-05 07 03 03      0...+.......
```

incaseoftrouble, 2021-04-20:
Okay, I'm stupid :-) Apparently, I misconfigured something before. Apparently, it works with `check-purpose=codesign`.

But this toggle should be available on the command line IMO

jluebbe, 2021-04-20:
OK, thanks for testing this again. So we have two thing we should fix:

- [ ] rauc -c should not silently fall back to defaults
- [ ] we should add a way to pass config options via the command line

ejoerns, 2021-04-20:
> * [ ]  rauc -c should not silently fall back to defaults

Note that this is what #613 is for.

dwunderlich-s, 2022-07-28:
> OK, thanks for testing this again. So we have two things we should fix:
> 
> * [ ]  rauc -c should not silently fall back to defaults
> * [ ]  we should add a way to pass config options via the command line

Hi all, what is the status concerning the 2nd task "pass config options via the command line"? Will this be realized in near future? Thanks, Daniel

jluebbe, 2022-08-19:
> > * [ ]   we should add a way to pass config options via the command line
> 
> Hi all, what is the status concerning the 2nd task "pass config options via the command line"? Will this be realized in near future? Thanks, Daniel

We don't have concrete plans for this, but would be happy to discuss concrete design and take PRs.

jluebbe, 2022-08-19:
Closing in favor of #953, which is focused on the command line config aspect.

closed 2022-08-19 by jluebbe