rauc/rauc #731: failed to run casync extract: Failed to run synchronizer: Function not implemented

tte-paul, 2021-04-27:
Hi, currently I'm trying to experiment with casync support. The bundle can be converted from conventional format to casync format using the following cmd 
```
rauc convert --cert=<certfile> --key=<keyfile> --keyring=<keyring> conventional-bundle.raucb casync-bundle.raucb
```

 but during the installation process, it fails at **Failed to run synchronizer: Function not implemented**

```
-- Logs begin at Thu 2020-07-30 19:02:39 UTC, end at Tue 2021-04-27 18:58:52 UTC. --
Jul 30 19:02:57 creo systemd[1]: Starting Rauc Update Service...
Jul 30 19:02:58 creo rauc[206]: Config option 'statusfile=<path>/per-slot' unset, falling back to per-slot status
Jul 30 19:02:58 creo rauc[206]: Using per-slot statusfile
Jul 30 19:02:59 creo systemd[1]: Started Rauc Update Service.
Jul 30 19:03:23 creo rauc[206]: mounting slot /dev/mmcblk0p3
Jul 30 19:03:24 creo rauc[206]: Failed to load status file /run/rauc/rootfs.1/slot.raucs: No such file or directory
Jul 30 19:03:24 creo rauc[206]: mounting slot /dev/mmcblk0p6
Jul 30 19:03:24 creo rauc[206]: Failed to load status file /run/rauc/appfs.1/slot.raucs: No such file or directory
Jul 30 19:03:24 creo rauc[206]: Failed to load status file /slot.raucs: No such file or directory
Jul 30 19:03:24 creo rauc[206]: Failed to load status file /mnt/app/slot.raucs: No such file or directory
Apr 27 18:58:32 creo rauc[206]: input bundle: /mnt/usb/update-2021.04-01-casync.raucb
Apr 27 18:58:32 creo rauc[206]: Active slot bootname: A
Apr 27 18:58:32 creo rauc[206]: installing /mnt/usb/update-2021.04-01-casync.raucb: started
Apr 27 18:58:32 creo rauc[206]: installing /mnt/usb/update-2021.04-01-casync.raucb: Checking and mounting bundle...
Apr 27 18:58:32 creo rauc[206]: Reading bundle: /mnt/usb/update-2021.04-01-casync.raucb
Apr 27 18:58:32 creo rauc[206]: Verifying bundle signature... 
Apr 27 18:58:32 creo rauc[206]: Verified inline signature by 'O = rauc Inc., CN = rauc-demo'
Apr 27 18:58:32 creo rauc[206]: Mounting bundle '/mnt/usb/update-2021.04-01-casync.raucb' to '/run/rauc/bundle'
Apr 27 18:58:32 creo rauc[206]: Configured loop device '/dev/loop0' for 32768 bytes
Apr 27 18:58:33 creo rauc[206]: Configured dm-verity device '/dev/dm-0'
Apr 27 18:58:33 creo rauc[206]: Marking target slot rootfs.1 as non-bootable...
Apr 27 18:58:33 creo rauc[206]: installing /mnt/usb/update-2021.04-01-casync.raucb: Updating slots...
Apr 27 18:58:33 creo rauc[206]: Checking image type for slot type: ext4
Apr 27 18:58:33 creo rauc[206]: Image detected as type: *.ext4.caibx
Apr 27 18:58:33 creo rauc[206]: installing /mnt/usb/update-2021.04-01-casync.raucb: Checking slot rootfs.1
Apr 27 18:58:33 creo rauc[206]: Slot needs to be updated with /run/rauc/bundle/rootfs.ext4.caibx
Apr 27 18:58:33 creo rauc[206]: installing /mnt/usb/update-2021.04-01-casync.raucb: Updating slot rootfs.1
Apr 27 18:58:33 creo rauc[206]: Updating /dev/mmcblk0p3 with /run/rauc/bundle/rootfs.ext4.caibx
Apr 27 18:58:33 creo rauc[206]: Extracting /run/rauc/bundle/rootfs.ext4.caibx to /dev/mmcblk0p3
Apr 27 18:58:50 creo rauc[281]: Failed to run synchronizer: Function not implemented
Apr 27 18:58:52 creo rauc[206]: Installation error: Failed updating slot rootfs.1: failed to run casync extract: Child process exited with code 1
Apr 27 18:58:52 creo rauc[206]: installing /mnt/usb/update-2021.04-01-casync.raucb: Installation error: Failed updating slot rootfs.1: failed to run1
Apr 27 18:58:52 creo rauc[206]: installing /mnt/usb/update-2021.04-01-casync.raucb: finished
Apr 27 18:58:52 creo rauc[206]: installing `/mnt/usb/update-2021.04-01-casync.raucb` failed: 1
```

I found another issure https://github.com/rauc/rauc/issues/598 which has very similar problem, but it failed at **Failed to run synchronizer: Operation not permitted**.  Do you have any ideas about this issure? thanks a lot.

ejoerns, 2021-04-27:
The thing with casync is: Almost every error starts with "Failed to run synchronizer". Thus, this does not give a good hint, yet.

From the logs one can see that casync should attempt to unpack a blob index (.caibx) to an mmc/SD partition, which should be supported in general.

Do you use the master version of 'casync'? Or any (even) older one?

Could you start the rauc service with `G_MESSAGES_DEBUG="rauc rauc-subprocess"` set? This should print the exact casync command called in the service's log. And if possible, verify if you can reproduce the issue with calling casync manually (you can simply 'mount' the bundle containing the casync blob index for this (it's basically a squashfs))

tte-paul, 2021-04-28:
thanks @ejoerns, I'm using the commit 4ad9bcb94bc83ff36cfc65515107ea06a88c2dfc of casync with two patches which is from latest buildroot (https://github.com/buildroot/buildroot/tree/2021.02.x/package/casync). It is quite latest.

I've started the rauc service with "G_MESSAGES_DEBUG=rauc rauc-subprocess" set as you suggested, and the log as below:
```
# journalctl -u rauc.service  | tee rauc-003.log 
-- Logs begin at Thu 2020-07-30 19:02:39 UTC, end at Wed 2021-04-28 14:31:50 UTC. --
Jul 30 19:02:57 creo systemd[1]: Starting Rauc Update Service...
Jul 30 19:02:57 creo rauc[201]: service start
Jul 30 19:02:57 creo rauc[201]: No value for key "max-bundle-download-size" in [system] defined - using default value of 8388608 bytes.
Jul 30 19:02:57 creo rauc[201]: Config option 'statusfile=<path>/per-slot' unset, falling back to per-slot status
Jul 30 19:02:57 creo rauc[201]: Using per-slot statusfile
Jul 30 19:02:58 creo rauc[201]: name 'de.pengutronix.rauc' acquired
Jul 30 19:02:58 creo systemd[1]: Started Rauc Update Service.
Jul 30 19:10:52 creo rauc[201]: launching subprocess: fw_printenv BOOT_ORDER
Jul 30 19:10:53 creo rauc[201]: launching subprocess: fw_printenv BOOT_A_LEFT
Jul 30 19:10:53 creo rauc[201]: Found external mountpoint for slot rootfs.0 at /
Jul 30 19:10:53 creo rauc[201]: Found external mountpoint for slot appfs.0 at /mnt/app
Jul 30 19:10:53 creo rauc[201]: Found booted slot: rootfs.0 on /dev/mmcblk0p2
Jul 30 19:10:53 creo rauc[201]: launching subprocess: fw_printenv BOOT_ORDER
Jul 30 19:10:53 creo rauc[201]: launching subprocess: fw_printenv BOOT_ORDER
Jul 30 19:10:53 creo rauc[201]: launching subprocess: fw_printenv BOOT_A_LEFT
Jul 30 19:10:53 creo rauc[201]: mounting slot /dev/mmcblk0p3
Jul 30 19:10:53 creo rauc[201]: launching subprocess: mount -t ext4 /dev/mmcblk0p3 /run/rauc/rootfs.1
Jul 30 19:10:53 creo rauc[201]: launching subprocess: umount /run/rauc/rootfs.1
Jul 30 19:10:53 creo rauc[201]: mounting slot /dev/mmcblk0p6
Jul 30 19:10:53 creo rauc[201]: launching subprocess: mount -t ext4 /dev/mmcblk0p6 /run/rauc/appfs.1
Jul 30 19:10:53 creo rauc[201]: launching subprocess: umount /run/rauc/appfs.1
Jul 30 19:10:53 creo rauc[201]: Failed to load status file /slot.raucs: No such file or directory
Jul 30 19:10:53 creo rauc[201]: Failed to load status file /mnt/app/slot.raucs: No such file or directory
Apr 28 14:30:28 creo rauc[201]: launching subprocess: fw_printenv BOOT_ORDER
Apr 28 14:30:28 creo rauc[201]: launching subprocess: fw_printenv BOOT_A_LEFT
Apr 28 14:30:28 creo rauc[201]: Found external mountpoint for slot rootfs.0 at /
Apr 28 14:30:28 creo rauc[201]: Found external mountpoint for slot appfs.0 at /mnt/app
Apr 28 14:30:28 creo rauc[201]: Found booted slot: rootfs.0 on /dev/mmcblk0p2
Apr 28 14:30:28 creo rauc[201]: launching subprocess: fw_printenv BOOT_ORDER
Apr 28 14:30:29 creo rauc[201]: launching subprocess: fw_printenv BOOT_ORDER
Apr 28 14:30:29 creo rauc[201]: launching subprocess: fw_printenv BOOT_A_LEFT
Apr 28 14:30:29 creo rauc[201]: Found external mountpoint for slot rootfs.0 at /
Apr 28 14:30:29 creo rauc[201]: Found external mountpoint for slot appfs.0 at /mnt/app
Apr 28 14:30:29 creo rauc[201]: Found booted slot: rootfs.0 on /dev/mmcblk0p2
Apr 28 14:30:29 creo rauc[201]: launching subprocess: fw_setenv BOOT_A_LEFT 3
Apr 28 14:30:29 creo rauc[201]: rauc mark: marked slot rootfs.0 as good
Apr 28 14:31:29 creo rauc[201]: input bundle: /mnt/usb/update-2021.04-01-casync.raucb
Apr 28 14:31:29 creo rauc[201]: Active slot bootname: A
Apr 28 14:31:29 creo rauc[201]: thread started for /mnt/usb/update-2021.04-01-casync.raucb
Apr 28 14:31:29 creo rauc[201]: installing /mnt/usb/update-2021.04-01-casync.raucb: started
Apr 28 14:31:29 creo rauc[201]: Found external mountpoint for slot rootfs.0 at /
Apr 28 14:31:29 creo rauc[201]: Found external mountpoint for slot appfs.0 at /mnt/app
Apr 28 14:31:29 creo rauc[201]: Found booted slot: rootfs.0 on /dev/mmcblk0p2
Apr 28 14:31:29 creo rauc[201]: installing /mnt/usb/update-2021.04-01-casync.raucb: Checking and mounting bundle...
Apr 28 14:31:29 creo rauc[201]: Reading bundle: /mnt/usb/update-2021.04-01-casync.raucb
Apr 28 14:31:29 creo rauc[201]: Verifying bundle signature... 
Apr 28 14:31:30 creo rauc[201]: Verified inline signature by 'O = rauc Inc., CN = rauc-demo'
Apr 28 14:31:30 creo rauc[201]: Got 1 elements for trust chain
Apr 28 14:31:30 creo rauc[201]: Mounting bundle '/mnt/usb/update-2021.04-01-casync.raucb' to '/run/rauc/bundle'
Apr 28 14:31:30 creo rauc[201]: Configured loop device '/dev/loop0' for 32768 bytes
Apr 28 14:31:31 creo rauc[201]: Configured dm-verity device '/dev/dm-0'
Apr 28 14:31:31 creo rauc[201]: Checking slot: rootfs.1
Apr 28 14:31:31 creo rauc[201]:         Adding mapping: rootfs -> rootfs.1
Apr 28 14:31:31 creo rauc[201]: Checking slot: appfs.1
Apr 28 14:31:31 creo rauc[201]:         Adding mapping: appfs -> appfs.1
Apr 28 14:31:31 creo rauc[201]: Checking slot: rootfs.0
Apr 28 14:31:31 creo rauc[201]:         No mapping found
Apr 28 14:31:31 creo rauc[201]: Checking slot: appfs.0
Apr 28 14:31:31 creo rauc[201]:         No mapping found
Apr 28 14:31:31 creo rauc[201]: Using default installation handler
Apr 28 14:31:31 creo rauc[201]: Found image mapping: rootfs.ext4.caibx -> rootfs
Apr 28 14:31:31 creo rauc[201]: Found image mapping: appfs.ext4.caibx -> appfs
Apr 28 14:31:31 creo rauc[201]: Marking target slot rootfs.1 as non-bootable...
[... cut ...]
It indicates that it failed at `launching subprocess:casync extract --seed /dev/mmcblk0p2 --store /mnt/usb/update-2021.04-01-casync.castr --seed-output=no /run/rauc/bundle/rootfs.ext4.caibx /dev/mmcblk0p3`

Currently I put the `raucb` file and `castr` directory under usb sticker and mount it to /mnt/usb, and try to run cmd `rauc install /mnt/usb/update-2021.04-01-casync.raucb`
```
drwxr-xr-x  710 root     root         32768 Apr 27 18:50 update-2021.04-01-casync.castr
-rwxr-xr-x    1 root     root         35405 Apr 27 18:50 update-2021.04-01-casync.raucb
```

How should I mount the rootfs.ext4.caibx to /run/rauc/bundle/ from update-2021.04-01-casync.raucb?  Thanks.

nunojpg, 2022-02-06:
In my case (also while under Buildroot), it was the lack of libzstd support on casync (There is no package for Zstandard under buildroot ATM). I fixed it my calling rauc convert with "--casync-args --compression=gzip" (since I build buildroot with zlib).

jluebbe, 2022-08-19:
@tte-paul: Did @nunojpg's suggestion solve the issue for you? As this doesn't seem to be a RAUC issues, I'm going to close this for now.

closed 2022-08-19 by jluebbe